introlab/rtabmap_ros #85: stereo_vo performance is bad: Stereo.cpp:153::computeCorrespondences() A large number (667/934) of stereo correspondences are rejected! Optical flow may have failed, images are not calibrated or the background is too far (no disparity between the images)

Astrosun, 2016-06-13:
I tried Stereo Hand-Held Mapping tutorial with a bumblebee xb3, I followed the instructions and have calibrated the camera, the VO gets lost easily and shows:Stereo.cpp:153::computeCorrespondences() A large number (662/926) of stereo correspondences are rejected! Optical flow may have failed, images are not calibrated or the background is too far (no disparity between the images).

anyone can help? thax!

matlabbe, 2016-06-13:
Hi,

You can try to increase `Stereo/MaxDisparity` parameter to 128 pixels or more (default 64 pixels), Similar to [Zed launch example](http://wiki.ros.org/rtabmap_ros/Tutorials/StereoHandHeldMapping#Mapping_mode). 

You can also try `Stereo/OpticalFlow` to `false` to use a simple block matching approach.

cheers

Astrosun, 2016-07-05:
@matlabbe 
Hi matlabbe,

Thanks for your solution and sorry for the late reply since i had a deadline recently.The results are better than before but the vo still get lost sometimes.  i will try to figure it out maybe sth is wrong with my own program and will update you later~

Thanks a lot!

closed 2016-11-09 by matlabbe