introlab/rtabmap_ros #87: ZED warning after launch rtabmap : [ WARN] [1466283648.136178003]: Wait for a new image to proceed

willdzeng, 2016-06-18:
Hi Mathieu,

This is Di again, :)

Every time I run rtabmap, zed_wrapper_node starts to give warning about old_image.

I am wondering if this is happening because the computation power or can be fixed though rtabmap.

I am also gonna fire a issue in zed wrapper, to say what they think about it.

matlabbe, 2016-06-19:
Hi Di,

`rgbd_odometry` or `stereo_odometry` would use 100% of one core if the framerate is high (camera framerate > odometry framerate). This may influence other processes on the computer that may need to run very fast (like processing USB3 image stream). How much CPU % does `zed_wrapper_node` is using alone?

cheers

willdzeng, 2016-06-19:
around 5% to 10% CPU, 10% GPU for 1080P.
I just found out even with rviz showing the image it still have this warning.
I think its the zed_warpper's issue.

willdzeng, 2016-06-20:
Issue answered by zed_ros_wraper
stereolabs/zed-ros-wrapper#38

closed 2016-06-20 by willdzeng