introlab/rtabmap_ros #88: Source and Calculation files for rtabmap_ros

gabrielarpino, 2016-06-20:
Hi, I am trying to find the program files that do the SLAM calculations. I looked all over the src folder but couldnt find them. I would like to modify these for my project, something similar to this :  https://github.com/introlab/rtabmap/blob/master/corelib/src/BayesFilter.cpp

, but for rtabmap_ros and not just rtabmap. Thanks :)

matlabbe, 2016-06-20:
`rtabmap_ros` is a ROS wrapper of the `rtabmap` library. To modify the SLAM approach, you have to modify the library, in particular the `Rtabmap` class. For example the main SLAM function [Rtabmap::process()](https://github.com/introlab/rtabmap/blob/be13a9b967c7b47531cd3b0ca8bf45380eac3d76/corelib/src/Rtabmap.cpp#L801-L2605) is called from [here](https://github.com/introlab/rtabmap_ros/blob/cc0591c60aa8f8003505dcf6edf13e9ed3308c70/src/CoreWrapper.cpp#L1443) in `rtabmap_ros`.

cheers

gabrielarpino, 2016-06-20:
> Yes, and a make install. You would have to do a catkin_make again too if
> you changed the headers.
>
> cheers

Cool thanks :)

closed 2016-11-09 by matlabbe