introlab/rtabmap_ros #89: Assign initial proj_map size

willdzeng, 2016-06-21:
Hey Mathieu,

Is there a way to assign the initial size of proj_map?
Move_base needs the goal to be set inside the costmap. But the proj_map is initially small without previous mapping, so I can't assign goal point to ask the robot to explore the area.

Thanks~

willdzeng, 2016-06-21:
Also this would cause rviz give warning about

>  Update area outside of original map area.

[image]

matlabbe, 2016-06-21:
Hi,

You can use `grid_size` parameter. Example initializing a 50x50 meters map:
```xml
<node pkg="rtabmap_ros" type="rtabmap" name="rtabmap">
...
<param name="grid_size" type="double" value="50"/>
...
</node>
```

The planning approach used in RTAB-Map is used to handle memory management effects on the map over time. It is triggered when sending a goal to `/rtabmap/goal` topic instead of `/move_base_simple/goal` topic. The goal must be inside a radius of `RGBD/LocalRadius` (default 10 meters) from the map's graph to be accepted. The planning is topological. The Dijkstra approach is used to find a path in the map's graph from the current node to the goal. Waypoints are then sent to [move_base](http://wiki.ros.org/move_base) through actionlib.

It is recommended to use `/rtabmap/goal` if `RGBD/OptimizeFromGraphEnd` is `true` (default `false`) as goals will "follow" the map when it is corrected after a loop closure. If `RGBD/OptimizeFromGraphEnd` is `false`, sending goals to `/rtabmap/goal` or `/move_base_simple/goal` will have the same effect.

cheers,
Mathieu

willdzeng, 2016-06-22:
Got it, grid_size works

closed 2016-06-22 by willdzeng