introlab/rtabmap_ros #90: build rtabmap_ros with linkage of rtabmap form /usr/local/lib

willdzeng, 2016-06-22:
Deal with ROS sometimes need to delete /devel directory in workspace very often.
Maybe change the linkage of rtabmap to /usr/local/lib to make re-compile of rtabmap_ros easier?

matlabbe, 2016-06-22:
rtabmap can be installed in `/usr/local` (default install directory), catkin_make will still be able to find it.

matlabbe, 2016-06-22:
Missclick on "close and comment"... In https://github.com/introlab/rtabmap_ros#build-from-source, I recommended to install in `catkin_ws/devel` but it is optional.

matlabbe, 2016-06-22:
I updated the [step 2](https://github.com/introlab/rtabmap_ros#build-from-source) or the install instructions.

willdzeng, 2016-06-23:
I see, I guess the reason I failed to compile with /user/local is I didn't delete the old build folder.
Now I this is fixed.

closed 2016-06-23 by willdzeng