introlab/rtabmap_ros #91: Installation problem

Annaiyan, 2016-06-23:
Hi,
I'm trying to install the rtabmap_ros in my workspace.I have followed the installation procedure but i'm getting the following error,
No rule to make target /opt/ros/indigo/lib/x86_64-linux-gnu/librtabmap_core.so', needed by/home/aru/Workspace/catkin/devel/lib/librtabmap_ros.so'

i'm using ros-indigo in ubuntu 14.04 trusty

looking forward for the support,
regards,
Arun

matlabbe, 2016-06-23:
Is that during `catkin_make`? Is the file "/opt/ros/indigo/lib/x86_64-linux-gnu/librtabmap_core.so" exist?

Annaiyan, 2016-06-23:
yes during the catkin_make.

matlabbe, 2016-06-23:
is /opt/ros/indigo/lib/librtabmap_core.so exist?

Annaiyan, 2016-06-23:
No,there is no such file in /opt/ros/indigo/lib

matlabbe, 2016-06-23:
The problem is that `rtabmap_ros` is trying to link with `ros-indigo-rtabmap` instead of the one your built locally and installed in your `catkin_ws/devel` space. Try remove `CMakeCache.txt` from `catkin_ws/build` directory, then `catkin_make` again.

cheers

Annaiyan, 2016-06-24:
yes,it works now !
Many thanks for the support.

closed 2016-06-24 by Annaiyan

hgfriver, 2021-03-06:
I have faced the same problem, but this solution really works!!!!!!!!!!!!!!!!!!!!!!!
Thank you sooooooooo much :)

gowtham120, 2022-07-26:
> The problem is that `rtabmap_ros` is trying to link with `ros-indigo-rtabmap` (which seems not installed anymore) instead of the one built locally and installed in your `catkin_ws/devel` space. Try remove `CMakeCache.txt` from `catkin_ws/build` directory, then `catkin_make` again.
> 
> cheers

works for me

thank you..