introlab/rtabmap_ros #92: stereo mapping error

Annaiyan, 2016-06-24:
I have sucessfully installed the rtabmap_ros and compiled without error (catkin_make).
Now i'm trying to launch the stereo_mapping.launch file.I'm getting the following error,

> ERROR: cannot launch node of type [rtabmap_ros/stereo_odometry]: can't locate node [stereo_odometry] in package [rtabmap_ros]
> ERROR: cannot launch node of type [rtabmap_ros/rtabmap]: can't locate node [rtabmap] in package [rtabmap_ros]
> process[rviz-3]: started with pid [3256]
> process[points_xyzrgb-4]: started with pid [3257]
> [ERROR] [1466775545.132798371]: Failed to load nodelet [/points_xyzrgb] of type [rtabmap_ros/point_cloud_xyzrgb] even after refreshing the cache: Could not find library corresponding to plugin rtabmap_ros/point_cloud_xyzrgb. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
> [ERROR] [1466775545.132885779]: The error before refreshing the cache was: Could not find library corresponding to plugin rtabmap_ros/point_cloud_xyzrgb. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
> 

i tried this [ introlab/rtabmap_ros#7](url) but still i'm getting the same error.
any idea for solve this issue?

cheers,
-Arun

willdzeng, 2016-06-24:
> i tried this introlab/rtabmap_ros#7 <http://url> but still i'm getting

Seems like your ROS couldn't find the node.
you should try
1. delete devel/ build/ folder in your rtabmap_ros workspace.
2. build rtabmap_ros again
3. put the rtabmap_ros's working space source setup.bash at the last line
of ~/.bashrc
4. then open a new terminal
5. do ' rospack list'
6. launch again

Annaiyan, 2016-06-25:
i tried the above steps but still i'm getiing the same error.
when i do rospack list, i can see the package like below

> rtabmap /home/arun/Workspace/catkin/devel/share/rtabmap
> rtabmap_ros /home/arun/Workspace/catkin/src/rtabmap_ros
>
My workspace is already sourced like below,

> source /opt/ros/indigo/setup.bash
> source /home/arun/Workspace/catkin/devel/setup.bash
 couldn't find the exact reason for this issue!

matlabbe, 2016-06-25:
Did you do catkin_make? and that it has built everything with success.

Annaiyan, 2016-06-25:
yes,i did the catkin_make sucessfully

matlabbe, 2016-06-25:
Another reason would be that nodes created are not executable, for some reasons. Can you do:
```bash
mathieu@mathieu:~/catkin_ws/devel/lib/rtabmap_ros$ ls -l
total 41716
-rwxrwxr-x 1 mathieu mathieu  1015916 Jun 22 19:41 camera
-rwxrwxr-x 1 mathieu mathieu   704790 Jun 22 19:44 data_player
-rwxrwxr-x 1 mathieu mathieu   914571 Jun 22 19:44 map_assembler
-rwxrwxr-x 1 mathieu mathieu  1125792 Jun 22 19:45 map_optimizer
-rwxrwxr-x 1 mathieu mathieu   242607 Jun 22 19:45 odom_msg_to_tf
-rwxrwxr-x 1 mathieu mathieu  2327478 Jun 22 19:45 rgbd_odometry
-rwxrwxr-x 1 mathieu mathieu 14783674 Jun 22 19:47 rtabmap
-rwxrwxr-x 1 mathieu mathieu 19741129 Jun 22 19:48 rtabmapviz
-rwxrwxr-x 1 mathieu mathieu  1838346 Jun 22 19:48 stereo_odometry
```

Annaiyan, 2016-06-25:
I did the ls -l and the got the following

> arun@arun-Latitude-E5440:~/Workspace/catkin/devel/lib/rtabmap_ros$ ls -l
> total 2688
> -rwxrwxr-x 1 arun arun  706308 Jun 25 13:15 data_player
> -rwxrwxr-x 1 arun arun  916089 Jun 25 13:15 map_assembler
> -rwxrwxr-x 1 arun arun 1125752 Jun 25 13:15 map_optimizer

matlabbe, 2016-06-25:
some nodes are missing, what is the output of catkin_make?

Annaiyan, 2016-06-25:
i re installed the rtabmap ros and did the catkin_make again.Below are the catkin_make output
```
[ 98%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/nodelets/point_cloud_xyzrgb.cpp.o
[ 98%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/nodelets/point_cloud_xyz.cpp.o
[ 98%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/nodelets/disparity_to_depth.cpp.o
[ 98%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/nodelets/obstacles_detection.cpp.o
[ 98%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/nodelets/point_cloud_aggregator.cpp.o
[ 98%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/MsgConversion.cpp.o
[ 98%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/OdometryROS.cpp.o
[ 98%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/MapsManager.cpp.o
[ 98%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/rviz/MapCloudDisplay.cpp.o
[ 98%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/rviz/MapGraphDisplay.cpp.o
[ 98%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/rviz/InfoDisplay.cpp.o
[ 98%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/rviz/OrbitOrientedViewController.cpp.o
[ 98%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/rviz/moc_MapCloudDisplay.cxx.o
[ 98%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/rviz/moc_MapGraphDisplay.cxx.o
[ 98%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/rviz/moc_InfoDisplay.cxx.o
[ 98%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/rviz/moc_OrbitOrientedViewController.cxx.o
Linking CXX shared library ../devel/lib/librtabmap_ros.so
[ 98%] Built target rtabmap_ros
Scanning dependencies of target odom_msg_to_tf
Scanning dependencies of target data_player
Scanning dependencies of target map_assembler
Scanning dependencies of target map_optimizer
[ 98%] [ 98%] Building CXX object rtabmap_ros/CMakeFiles/odom_msg_to_tf.dir/src/OdomMsgToTFNode.cpp.o
Building CXX object rtabmap_ros/CMakeFiles/map_optimizer.dir/src/MapOptimizerNode.cpp.o
[ 98%] Building CXX object rtabmap_ros/CMakeFiles/data_player.dir/src/DbPlayerNode.cpp.o
[100%] Building CXX object rtabmap_ros/CMakeFiles/map_assembler.dir/src/MapAssemblerNode.cpp.o
Linking CXX executable ../devel/lib/rtabmap_ros/odom_msg_to_tf
Linking CXX executable ../devel/lib/rtabmap_ros/map_optimizer
Linking CXX executable ../devel/lib/rtabmap_ros/data_player
[100%] Built target odom_msg_to_tf
Scanning dependencies of target rgbd_odometry
Linking CXX executable ../devel/lib/rtabmap_ros/map_assembler
[100%] Building CXX object rtabmap_ros/CMakeFiles/rgbd_odometry.dir/src/RGBDOdometryNode.cpp.o
[100%] Built target map_optimizer
Scanning dependencies of target rtabmap
[100%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap.dir/src/CoreNode.cpp.o
[100%] Built target data_player
Scanning dependencies of target rtabmapviz
[100%] Building CXX object rtabmap_ros/CMakeFiles/rtabmapviz.dir/src/GuiNode.cpp.o
[100%] Built target map_assembler
Scanning dependencies of target stereo_odometry
[100%] Building CXX object rtabmap_ros/CMakeFiles/stereo_odometry.dir/src/StereoOdometryNode.cpp.o
Linking CXX executable ../devel/lib/rtabmap_ros/rgbd_odometry
[100%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap.dir/src/CoreWrapper.cpp.o
[100%] Building CXX object rtabmap_ros/CMakeFiles/rtabmapviz.dir/src/GuiWrapper.cpp.o
[100%] Built target rgbd_odometry
Linking CXX executable ../devel/lib/rtabmap_ros/stereo_odometry
[100%] Built target stereo_odometry
Linking CXX executable ../devel/lib/rtabmap_ros/rtabmap
[100%] Built target rtabmap
[100%] Building CXX object rtabmap_ros/CMakeFiles/rtabmapviz.dir/src/PreferencesDialogROS.cpp.o
Linking CXX executable ../devel/lib/rtabmap_ros/rtabmapviz
[100%] Built target rtabmapviz

```

matlabbe, 2016-06-25:
The nodes are created (but where?!). It is strange, on my computer catkin_make uses absolute linking paths, not relative:
```
Linking CXX shared library /home/mathieu/catkin_ws/devel/lib/librtabmap_ros.so
[ 91%] Built target rtabmap_ros
Linking CXX executable /home/mathieu/catkin_ws/devel/lib/rtabmap_ros/map_assembler
Linking CXX executable /home/mathieu/catkin_ws/devel/lib/rtabmap_ros/data_player
[ 92%] Built target data_player
Linking CXX executable /home/mathieu/catkin_ws/devel/lib/rtabmap_ros/map_optimizer
[ 92%] Built target map_assembler
Linking CXX executable /home/mathieu/catkin_ws/devel/lib/rtabmap_ros/odom_msg_to_tf
[ 93%] Built target odom_msg_to_tf
Linking CXX executable /home/mathieu/catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry
[ 94%] Built target map_optimizer
Linking CXX executable /home/mathieu/catkin_ws/devel/lib/rtabmap_ros/rtabmap
[ 95%] Built target rgbd_odometry
Linking CXX executable /home/mathieu/catkin_ws/devel/lib/rtabmap_ros/rtabmapviz
[ 96%] Built target rtabmap
Linking CXX executable /home/mathieu/catkin_ws/devel/lib/rtabmap_ros/stereo_odometry
[ 97%] Built target stereo_odometry
[100%] Built target rtabmapviz
```

What is your system? Where is `../devel/lib/rtabmap_ros/rtabmap`?

Annaiyan, 2016-06-25:
Inside the path 

> /devel/lib/rtabmap_ros 
there is no rtabmap 
i have only data_player, map_assembler, map_optimizer

matlabbe, 2016-06-25:
no, the standalone libraries should be built outside the ros workspace. Can you try to locate on your computer where the rtabmapviz file is actually created (because it seems created somewhere `[100%] Built target rtabmapviz`)?

Annaiyan, 2016-06-25:
I found all the executables camera,data_player,map_assembler,map_optimizer,odom_msg_to_tf,
rgbd_odometry,rtabmap,rtabmapviz,stereo_dodometry inside the build folder 
the exact path **/home/arun/Workspace/catkin/build/devel/lib**

matlabbe, 2016-06-25:
There is the problem. For some reasons, they are not created in `/home/arun/Workspace/catkin/devel/lib`. Try to clean your build directory and re-build, hope this will fix the wrong paths. By default catkin_make should create executables in `/home/arun/Workspace/catkin/devel/lib`.
```bash
$ rm -rf /home/arun/Workspace/catkin/build
$ cd /home/arun/Workspace/catkin
$ catkin_make
```

Annaiyan, 2016-06-25:
super! it works now. i can launch the file without error.
Many thanks for the support!

closed 2016-06-25 by Annaiyan