introlab/rtabmap_ros #93: transformation error

Annaiyan, 2016-06-27:
I'm trying to make octomap with stereo camera.I have launched like this,

>  ROS_NAMESPACE=/stereo_camera rosrun ueye stereo
which will publish 

> /stereo_camera/camera/parameter_descriptions
> /stereo_camera/camera/parameter_updates
> /stereo_camera/left/camera_info
> /stereo_camera/left/image_raw
> /stereo_camera/right/camera_info
> /stereo_camera/right/image_raw
after i have launched the demo_stereo_outdoor.launch file with some modification like `below,
<launch>  
   
   
   <arg name="rviz" default="true" />
   <arg name="rtabmapviz" default="false" />    
   <param name="use_sim_time" type="bool" value="false"/>   
   
   <group ns="/stereo_camera" >
      <node pkg="nodelet" type="nodelet" name="stereo_nodelet"  args="manager"/>   
      <node pkg="stereo_image_proc" type="stereo_image_proc" name="stereo_image_proc">                  
      </node>
   </group>
              
   <group ns="rtabmap">      
         
      <node pkg="rtabmap_ros" type="stereo_odometry" name="stereo_odometry" output="screen">
         <remap from="left/image_rect"       to="/stereo_camera/left/image_rect"/>
         <remap from="right/image_rect"      to="/stereo_camera/right/image_rect"/>
         <remap from="left/camera_info"      to="/stereo_camera/left/camera_info"/>
         <remap from="right/camera_info"     to="/stereo_camera/right/camera_info"/>
         <remap from="odom"                  to="/stereo_odometry"/>

         <param name="frame_id"      type="string" value="base_footprint"/>
         <param name="odom_frame_id" type="string" value="odom"/>

         <param name="Odom/Strategy"      type="string" value="0"/> 
         <param name="Vis/EstimationType" type="string" value="0"/> 
         <param name="Vis/MaxDepth"       type="string" value="10"/>
         <param name="Vis/MinInliers"     type="string" value="10"/>
         <param name="Odom/FillInfoData"  type="string" value="$(arg rtabmapviz)"/> 
         <param name="GFTT/MinDistance"   type="string" value="10"/>
         <param name="GFTT/QualityLevel"  type="string" value="0.00001"/> 
      </node>
   
      
      <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="--delete_db_on_start">
         <param name="frame_id"         type="string" value="base_footprint"/>
         <param name="subscribe_stereo" type="bool" value="true"/>
         <param name="subscribe_depth"  type="bool" value="false"/>

         <remap from="left/image_rect"   to="/stereo_camera/left/image_rect_color"/>
         <remap from="right/image_rect"  to="/stereo_camera/right/image_rect"/>
         <remap from="left/camera_info"  to="/stereo_camera/left/camera_info"/>
         <remap from="right/camera_info" to="/stereo_camera/right/camera_info"/>

         <remap from="odom" to="/stereo_odometry"/>

         <param name="queue_size" type="int" value="30"/>

         
         <param name="Rtabmap/TimeThr"                   type="string" value="700"/>
         <param name="Kp/MaxFeatures"                    type="string" value="200"/>
         <param name="Kp/MaxDepth"                       type="string" value="10"/>
         <param name="Kp/DetectorStrategy"               type="string" value="0"/>   
         <param name="SURF/HessianThreshold"             type="string" value="1000"/>
         <param name="Vis/EstimationType"                type="string" value="0"/>   
         <param name="RGBD/LoopClosureReextractFeatures" type="string" value="true"/>
         <param name="Vis/MaxDepth"                      type="string" value="10"/>
      </node>
 
      <node if="$(arg rtabmapviz)" pkg="rtabmap_ros" type="rtabmapviz" name="rtabmapviz" args="-d $(find rtabmap_ros)/launch/config/rgbd_gui.ini" output="screen">
         <param name="subscribe_stereo"    type="bool" value="true"/>
         <param name="subscribe_odom_info" type="bool" value="true"/>
         <param name="queue_size"          type="int" value="10"/>
         <param name="frame_id"            type="string" value="base_footprint"/>
         
         <remap from="left/image_rect"  to="/stereo_camera/left/image_rect_color"/>
         <remap from="right/image_rect"  to="/stereo_camera/right/image_rect"/>
         <remap from="left/camera_info"  to="/stereo_camera/left/camera_info"/>
         <remap from="right/camera_info" to="/stereo_camera/right/camera_info"/>
         <remap from="odom_info"         to="odom_info"/>
         <remap from="odom"              to="/stereo_odometry"/>
         <remap from="mapData"           to="mapData"/>
      </node>        
   </group>
  
    
   <node if="$(arg rviz)" pkg="rviz" type="rviz" name="rviz" args="-d $(find rtabmap_ros)/launch/config/demo_stereo_outdoor.rviz"/>

`</launch>
`also i do the transformation from camera frame id which is /left using rosrun tf static_publisher 0 0 0 0 0 0 base_footprint left 100.
But the transformation doesn't work.I'm getting the following error

> odometry: Could not get transform from base_footprint to /left (stamp=1467028492.328837) after 0.100000 seconds ("wait_for_transform_duration"=0.100000)!

 

> The stereo baseline (-0.223329) should be positive (baseline=-Tx/fx). We assume a horizontal left/right stereo setup where the Tx (or P(0,3)) is negative in the right camera info msg.

I have executed the tutorial with rosbag,it works fine (i can see the octomap if i launch octomap_server_node).

why the transformation is not working?
Am i missing anything in the launch file?

Thanks in advance,

-Arun

matlabbe, 2016-06-27:
Hi,

It looks like that published images by `ueye stereo` are not synchronized, the [left and right images have different timestamps](https://bitbucket.org/kmhallen/ueye/src/91d7a3ce87fc594e2be02ccfc168145b21172c15/src/StereoNode.cpp?at=default&fileviewer=file-view-default#StereoNode.cpp-475:505). You should set `approx_sync` to `true` for `stereo_odometry` node and `stereo_approx_sync` to `true` for `rtabmap` node.

The message:
<raw>
The stereo baseline (-0.223329) should be positive (baseline=-Tx/fx). We assume a horizontal left/right stereo setup where the Tx (or P(0,3)) is negative in the right camera info msg.
</raw>
tells that there is a problem with the calibration. Did you use this [tutorial](http://wiki.ros.org/camera_calibration/Tutorials/StereoCalibration) to calibrate the cameras?

cheers

Annaiyan, 2016-06-27:
I have added the approx_sync and stereo_approx_sync.

> <param name="approx_sync"           type="bool"   value="true"/>
> <param name="stereo_approx_sync"   type="bool"   value="true"/>

Usually i calibrate using the ros tutorial ( as mentioned above).Using the stereo_image_proc node i can get the 3D point cloud without problem.
is there any other option?

matlabbe, 2016-06-27:
Can you show your camera info values for the left and right images?
```
$ rostopic echo /stereo_camera/left/camera_info
$ rostopic echo /stereo_camera/right/camera_info
```

Another approach to use stereo cameras is to generate a depth image from the disparity image using [disparity_to_depth](http://wiki.ros.org/rtabmap_ros#rtabmap_ros.2BAC8-disparity_to_depth) nodelet, then uses the RGB-D interface instead of the stereo one.

cheers

Annaiyan, 2016-06-27:
here it is
/stereo_camera/left/camera_info
```
header: 
  seq: 474833
  stamp: 
    secs: 1467043998
    nsecs: 230242079
  frame_id: /left
height: 480
width: 752
distortion_model: plumb_bob
D: [0.05017381750693655, -0.08624079495140854, 0.0003068409695639882, 0.000677080628844562, 0.0]
K: [552.3732902673913, 0.0, 363.51933244136427, 0.0, 552.8141476458964, 219.7955026697695, 0.0, 0.0, 1.0]
R: [0.9999220342320602, -0.009443783893361217, -0.008169479970851845, 0.009284006356935394, 0.9997690937204566, -0.019379537335577894, 0.008350609749176471, 0.01930218089108497, 0.9997788221050019]
P: [581.5208662261348, 0.0, 365.81403732299805, 0.0, 0.0, 581.5208662261348, 239.09959602355957, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
``
/stereo_camera/right/camera_info
header: 
  seq: 474426
  stamp: 
    secs: 1467043998
    nsecs: 230242079
  frame_id: /left
height: 480
width: 752
distortion_model: plumb_bob
D: [0.027953063640482917, -0.05402691085378825, 0.0011373858995405864, -0.000966441379761347, 0.0]
K: [550.918504266175, 0.0, 360.00669342670585, 0.0, 551.7017610593424, 257.6691044303571, 0.0, 0.0, 1.0]
R: [0.9999733547444588, 0.006297571443670056, -0.003691936487088545, -0.00622498631824261, 0.999793336895366, 0.019352856248285973, 0.0038130494948946446, -0.01932935833236583, 0.9998058994424911]
P: [581.5208662261348, 0.0, 365.81403732299805, 129.71830781337002, 0.0, 581.5208662261348, 239.09959602355957, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False

```

matlabbe, 2016-06-27:
The problem is the value `P(0,3)` of your right camera info. The baseline is computed as `-(129.71830781337002) / 581.5208662261348 = -0.223067331 m`. If you look at the end of the [stereo calibration tutorial](http://wiki.ros.org/camera_calibration/Tutorials/StereoCalibration), `P(0,3)` is negative (`-28.144333`) but in your camera info it is positive (`129.71830781337002`). 

Try changing `129.71830781337002` to `-129.71830781337002`.

Annaiyan, 2016-06-27:
Thanks.Now i recalibrated everything and i'm getting odometry with some issues like below
```
[ INFO] [1467054653.697316955]: Setting RTAB-Map parameter "Vis/EstimationType"="0"
[ INFO] [1467054653.726432501]: Setting RTAB-Map parameter "Vis/MaxDepth"="1"
[ INFO] [1467054654.071127887]: Odom: quality=0, std dev=99.995003m, update time=0.063993s
[ INFO] [1467054654.121876485]: Odom: quality=11, std dev=0.020356m, update time=0.049977s
[ INFO] [1467054654.173280014]: Odom: quality=17, std dev=0.014539m, update time=0.043536s
[ INFO] [1467054654.223536290]: Odom: quality=16, std dev=0.019778m, update time=0.049253s
[ INFO] [1467054654.268868384]: RTAB-Map detection rate = 1.000000 Hz
[ INFO] [1467054654.268997151]: rtabmap: Deleted database "/home/arun/.ros/rtabmap.db" (--delete_db_on_start is set).
[ INFO] [1467054654.269031774]: rtabmap: Using database from "/home/arun/.ros/rtabmap.db".
[ INFO] [1467054654.271035826]: Odom: quality=14, std dev=0.028999m, update time=0.046470s
[ INFO] [1467054654.313184043]: Odom: quality=16, std dev=0.023155m, update time=0.041281s
[ INFO] [1467054654.359204439]: Odom: quality=14, std dev=0.009752m, update time=0.041020s
[ INFO] [1467054654.398840472]: Odom: quality=14, std dev=0.011443m, update time=0.039002s
[ INFO] [1467054654.451967357]: Odom: quality=15, std dev=0.031477m, update time=0.039444s
[ INFO] [1467054654.492283385]: Odom: quality=16, std dev=0.025391m, update time=0.039273s
[ INFO] [1467054654.530239445]: Odom: quality=12, std dev=0.012461m, update time=0.037265s
[ INFO] [1467054654.583888750]: Odom: quality=17, std dev=0.016441m, update time=0.041357s
[ INFO] [1467054654.625708717]: Odom: quality=18, std dev=0.015693m, update time=0.041027s
[ INFO] [1467054654.676791801]: Odom: quality=15, std dev=0.016882m, update time=0.038835s
[ INFO] [1467054654.719520518]: Odom: quality=18, std dev=0.020226m, update time=0.042037s
[ INFO] [1467054654.760887959]: Odom: quality=16, std dev=0.016314m, update time=0.040078s
[ INFO] [1467054654.809393854]: Odom: quality=14, std dev=0.022733m, update time=0.040156s
[ INFO] [1467054654.856345375]: Odom: quality=16, std dev=0.023703m, update time=0.046404s
[ INFO] [1467054654.914724522]: Odom: quality=14, std dev=0.011320m, update time=0.049886s
[ INFO] [1467054654.976287389]: Odom: quality=14, std dev=0.010644m, update time=0.044504s
[ INFO] [1467054655.016699142]: Odom: quality=17, std dev=0.010323m, update time=0.039612s
[ INFO] [1467054655.066117703]: Odom: quality=17, std dev=0.020779m, update time=0.041174s
[ INFO] [1467054655.110992495]: Odom: quality=17, std dev=0.016390m, update time=0.044235s
[ INFO] [1467054655.160107977]: Odom: quality=17, std dev=0.015417m, update time=0.038484s
[ INFO] [1467054655.200852430]: Odom: quality=17, std dev=0.030231m, update time=0.040016s
[ INFO] [1467054655.241847829]: Odom: quality=14, std dev=0.026533m, update time=0.040324s
[ INFO] [1467054655.295069766]: Odom: quality=20, std dev=0.017492m, update time=0.039770s
[ INFO] [1467054655.338030379]: Odom: quality=17, std dev=0.007165m, update time=0.042317s
[ INFO] [1467054655.386629369]: Odom: quality=11, std dev=0.009635m, update time=0.038489s
[ INFO] [1467054655.428521460]: Odom: quality=15, std dev=0.009078m, update time=0.041288s
[ INFO] [1467054655.472807101]: Odom: quality=13, std dev=0.014498m, update time=0.043521s
[ INFO] [1467054655.515550895]: Odom: quality=15, std dev=0.018776m, update time=0.041928s
[ INFO] [1467054655.560227380]: Odom: quality=13, std dev=0.015446m, update time=0.043144s
[ INFO] [1467054655.612367275]: Odom: quality=12, std dev=0.009700m, update time=0.040275s
[ INFO] [1467054655.655533520]: Odom: quality=17, std dev=0.028580m, update time=0.042503s
[ INFO] [1467054655.704713057]: Odom: quality=13, std dev=0.015337m, update time=0.038351s
[ INFO] [1467054655.749618678]: Odom: quality=16, std dev=0.012425m, update time=0.038523s
[ INFO] [1467054655.804051927]: Odom: quality=18, std dev=0.015588m, update time=0.039208s
[ INFO] [1467054655.845154413]: Odom: quality=18, std dev=0.013728m, update time=0.040445s
[ INFO] [1467054655.882920175]: Odom: quality=15, std dev=0.017924m, update time=0.037197s
[ INFO] [1467054655.937366661]: Odom: quality=18, std dev=0.022027m, update time=0.043683s
[ INFO] [1467054655.986245536]: Odom: quality=13, std dev=0.009694m, update time=0.047539s
[ INFO] [1467054656.034900441]: Odom: quality=16, std dev=0.028309m, update time=0.043290s
[ INFO] [1467054656.079428348]: Odom: quality=12, std dev=0.010504m, update time=0.043038s
[ INFO] [1467054656.116566945]: Odom: quality=12, std dev=0.019397m, update time=0.035818s
[ INFO] [1467054656.161437519]: Odom: quality=16, std dev=0.028516m, update time=0.040246s
[ INFO] [1467054656.205534810]: Odom: quality=19, std dev=0.020458m, update time=0.043471s
[ INFO] [1467054656.245216964]: Odom: quality=17, std dev=0.011987m, update time=0.038104s
[ INFO] [1467054656.293505502]: Odom: quality=15, std dev=0.012832m, update time=0.042431s
[ INFO] [1467054656.332222283]: Odom: quality=12, std dev=0.010305m, update time=0.038095s
[ INFO] [1467054656.384933044]: Odom: quality=18, std dev=0.019010m, update time=0.039515s
[ INFO] [1467054656.423436072]: Odom: quality=15, std dev=0.014154m, update time=0.037803s
[ INFO] [1467054656.464105234]: Odom: quality=15, std dev=0.009422m, update time=0.039664s
[ INFO] [1467054656.512199088]: Odom: quality=16, std dev=0.030319m, update time=0.037724s
[... cut ...] amera_info odom:=/stereo_odometry __name:=stereo_odometry __log:=/home/arun/.ros/log/50bce738-3c95-11e6-9609-ecf4bb684ae6/rtabmap-stereo_odometry-4.log].
log file: /home/arun/.ros/log/50bce738-3c95-11e6-9609-ecf4bb684ae6/rtabmap-stereo_odometry-4*.log
[ WARN] [1467054658.266325082]: rtabmap: Could not get transform from base_footprint to /left after 0.200000 seconds (for stamp=1467054657.806996)!

How to get rid of this error and get the odometry as well as map?

many thanks of support.
```

```

matlabbe, 2016-06-27:
It is like a new frame is received with a timestamp smaller than the previously processed frame. Looking again the [ueye code](https://bitbucket.org/kmhallen/ueye/src/91d7a3ce87fc594e2be02ccfc168145b21172c15/src/StereoNode.cpp?at=default&fileviewer=file-view-default#StereoNode.cpp-475:505), it seems they set the same timestamp between the left and right images, but not sure if the published frames are correctly ordered.

If you are building rtabmap from source, I [just added](https://github.com/introlab/rtabmap/commit/756e1768789b36012e9985d5d580713bb312e4ca) more info in the assert condition so we will see if `dt` is really negative. If so, it will mean that frames received are not ordered.

To debug, you could show the timestamps of the published left image and see if there is a problem with the stamps when stereo_odometry asserts:
```
$ rostopic echo /stereo_camera/left/image_rect_color/header/stamp
```

Annaiyan, 2016-06-28:
Here is the output of 
`rostopic echo /stereo_camera/left/image_rect_color/header/stamp`
```
secs: 1467103469
nsecs: 264570371
---
secs: 1467103469
nsecs: 264570371
---
secs: 1467103469
nsecs: 264570371
```

`rostopic echo /stereo_camera/right/image_rect_color/header/stamp`
```
secs: 1467103469
nsecs: 264570371
---
secs: 1467103469
nsecs: 264570371
---
secs: 1467103469
nsecs: 264570371
---

```
do you recommend to upgrade the standalone library (rtabmap) and try the same?

Annaiyan, 2016-06-28:
Now, i changed my settings like distance between the camera,framerate,exposure time and i recalibrated again.Also upgraded the standalone libraries. i can see the odometry for slow movements and still needs to check if the camera is moving constantly.
But i'm not getting the map data in rviz.
since there is no tf for /map to /left (camera).I tried like this
`rosrun tf static_transform_publisher 0 0 0 0 0 0 map left 100`
It is not working and i'm not getting any map data.

I need the /rtabmap/cloud_map in rviz.
any suggestions ?

matlabbe, 2016-06-28:
/map -> /odom will be published by rtabmap (once it can get odometry computed), and odometry will publish /odom -> /base_footprint (once it can compute odometry). I think `dt` is not negative, it is just 0 because the left timestamps can be identical for some time. Odometry assume always that new frame has a timstamp higher than the previous frame, otherwise odometry cannot be computed (velocity could not be estimated).

You don't need to get latest rtabmap source, I think it is the timestamp the problem. In [ueye code](https://bitbucket.org/kmhallen/ueye/src/91d7a3ce87fc594e2be02ccfc168145b21172c15/src/StereoNode.cpp?at=default&fileviewer=file-view-default#StereoNode.cpp-475:505), instead of:
```cpp
// Timestamp and publish the image. Called by the streaming thread.
void StereoNode::publishImageL(const char *frame, size_t size)
{
  ros::Time stamp = ros::Time::now();
  boost::lock_guard<boost::mutex> lock(mutex_);
  l_stamp_ = stamp;
  double diff = (l_stamp_ - r_stamp_).toSec();
  if ((diff >= 0) && (diff < 0.02)) {
    l_msg_camera_info_.header.stamp = r_msg_camera_info_.header.stamp;
  } else {
    l_msg_camera_info_.header.stamp = l_stamp_;
  }
  sensor_msgs::CameraInfoPtr info;
  sensor_msgs::ImagePtr img = processFrame(frame, size, l_cam_, info, l_msg_camera_info_);
  l_pub_stream_.publish(img, info);
}
void StereoNode::publishImageR(const char *frame, size_t size)
{
  ros::Time stamp = ros::Time::now();
  boost::lock_guard<boost::mutex> lock(mutex_);
  r_stamp_ = stamp;
  double diff = (r_stamp_ - l_stamp_).toSec();
  if ((diff >= 0) && (diff < 0.02)) {
    r_msg_camera_info_.header.stamp = l_msg_camera_info_.header.stamp;
  } else {
    r_msg_camera_info_.header.stamp = r_stamp_;
  }
  sensor_msgs::CameraInfoPtr info;
  sensor_msgs::ImagePtr img = processFrame(frame, size, r_cam_, info, r_msg_camera_info_);
  r_pub_stream_.publish(img, info);
}
```

do:
```cpp
// Timestamp and publish the image. Called by the streaming thread.
void StereoNode::publishImageL(const char *frame, size_t size)
{
  l_msg_camera_info_.header.stamp = ros::Time::now();
  sensor_msgs::CameraInfoPtr info;
  sensor_msgs::ImagePtr img = processFrame(frame, size, l_cam_, info, l_msg_camera_info_);
  l_pub_stream_.publish(img, info);
}
void StereoNode::publishImageR(const char *frame, size_t size)
{
  r_msg_camera_info_.header.stamp = ros::Time::now();
  sensor_msgs::CameraInfoPtr info;
  sensor_msgs::ImagePtr img = processFrame(frame, size, r_cam_, info, r_msg_camera_info_);
  r_pub_stream_.publish(img, info);
} 
```
then make sure you set approximate time parameters as in the related [post](https://github.com/introlab/rtabmap_ros/issues/93#issuecomment-228753202) above.

cheers

Annaiyan, 2016-06-29:
I have managed to include the above code as recommended and updated the ueye driver as well.
Now i'm getting different warning and there is no data in the /rtabmap/cloud_map (i think it it is due to odometry)
```
[ WARN] (2016-06-29 17:24:51.856) Stereo.cpp:159::computeCorrespondences() A large number (799/908) of stereo correspondences are rejected! Optical flow may have failed, images are not calibrated, the background is too far (no disparity between the images) or maximum disparity may be too small (64).
[ WARN] (2016-06-29 17:24:51.856) OdometryF2M.cpp:227::computeTransform() Registration failed: "Not enough 3D features in images (old=10, new=6, min=10)"
For you info,i'm getting 3D point cloud from stereo_imag_proc node.

looking for your suggestion on this issue?

```

matlabbe, 2016-06-29:
Many stereo correspondences are rejected. You can try to increase `Stereo/MaxDisparity` parameter to `128` or more (if your baseline is large and that image resolution is high):
```xml
<param name="Stereo/MaxDisparity"     type="string" value="128"/>
```

You can also look if the disparity image looks good (many valid disparity values) with 
```bash
$ rosrun image_view disparity_view image:=/stereo_camera/disparity
```
Low number of valid disparity means a poor calibration or that there is not a lot visual texture in the environment.

Annaiyan, 2016-06-29:
Now i increased the maximum disparity value to 250 and i can see some odometry if i publish the tf like this
`rosrun tf static publisher 0 0 0 0 0 0 left world 100`
since the rviz which is in the launch file is crashing. i'm launching the rviz seperately and add the topic to view.
Also the same warning 
`[ WARN] (2016-06-29 17:54:58.230) Stereo.cpp:159::computeCorrespondences() A large number (834/969) of stereo correspondences are rejected! Optical flow may have failed, images are not calibrated, the background is too far (no disparity between the images) or maximum disparity may be too small (250).
`
even if i get few odometry i'm not getting the map data .
Do i need perfect odometry to compute the map?

thanks in advance

matlabbe, 2016-06-29:
If visual odometry is lost most of the time, the map cannot be computed. Can you record a small rosbag (~10 sec) so I can see better what is the problem (834/969 rejected is very high)?
```
$ rosbag record -o out.bag /stereo_camera/left/image_rect_color /stereo_camera/right/image_rect /stereo_camera/left/camera_info /stereo_camera/right/camera_info
$ rosbag compress out.bag
```

Annaiyan, 2016-06-29:
Thanks.
Here is the link for the rosbag [https://www.dropbox.com/s/vgcetvg1qwy89ve/out_2016-06-29-19-42-23.bag.tar.gz?dl=0](url)
Let me know if there is a issue in downloading the file

matlabbe, 2016-06-29:
[image]

The right and left cameras are inverted (the right should be left and the left should be right). That also explained why you had a negative baseline earlier. Switch the cameras and calibrate again.

cheers

Annaiyan, 2016-06-29:
I will try that.May i know, how you predicted that it is inverted?

matlabbe, 2016-06-29:
See the alignment of the images above. The pole should be more on left in the right image than in the left image. It is like the left camera is on the right of the right camera.

willdzeng, 2016-06-29:
The right image shows more kitchen table. :))

Annaiyan, 2016-06-29:
many thanks for the input.
super,now i'm getting the odometry without any error but still i'm not getting the map. As you mentioned before,if odometry works map should get computed!
Here is the link for the new bag file [https://www.dropbox.com/s/mcr9hhdk9nrynwx/out_2016-06-29-21-36-20.bag?dl=0](url)
once the odometry is computed, how can i check the tf is get published between odom and map?

matlabbe, 2016-06-29:
I can get a map with your rosbag by doing this (I didn't try your launch file):
```bash
$ roscore
$ rosparam set use_sim_time true
$ roslaunch rtabmap_ros stereo_mapping.launch rtabmap_args:="--delete_db_on_start" frame_id:="/base_link" approximate_sync:=true
$ rosrun tf static_transform_publisher 0 0 0 -1.5707963267948966 0 -1.5707963267948966 /base_link /left 100
$ rosbag play --clock out_2016-06-29-21-36-20.bag
```
At the middle of the bag there are some lags, like no images for 3-4 seconds. I am more wondering about the synchronization between the left and right images, which is very important to get good stereo matches when the cameras are moving. Cameras with hardware synchronization is a must. Just found there is also this package [ueye_cam](http://wiki.ros.org/ueye_cam) (see `Camera Synchronization` section) instead of [ueye](http://wiki.ros.org/ueye) that could be used to get your images with exact time synchronization.

It will also work better in environments with more visual features.

[image]

Annaiyan, 2016-06-30:
Thanks for checking it.
Yes it is working (only the bag file)  if i launch like the above post.But if i try to do with the same camera online ,i can't see the map in rviz.My goal here is to build clear octomap if i get the rtabmap/cloud_map.
For  the camera synchronisation,i'm working on the ueye_cam and post it soon if it works better.

still the rtabmap/cloud_map is empty?

matlabbe, 2016-06-30:
I cannot test with your cameras online, but normally if they publish like in the bag, I can get `/rtabmap/cloud_map`.
[image]

Annaiyan, 2016-07-03:
Thanks for your reply.
Now i'm started to work with hardware synchronized camera (not ueye it's a different camera).I tried the same steps as i did before.
```
> $ roscore
> $ rosparam set use_sim_time true
> $ roslaunch rtabmap_ros stereo_mapping.launch rtabmap_args:="--delete_db_on_start" frame_id:="/base_link" approximate_sync:=true
> $ rosrun tf static_transform_publisher 0 0 0 -1.5707963267948966 0 -1.5707963267948966 /base_link /guidance 100

but the tranformation doesn't work for the stereo_mapping launch file.But if i try to see the point cloud by using stereo_image_proc , the tranformation works fine and i can see the point cloud.

Here is my error for the transformation 

`[ WARN] [1467551980.547786284]: odometry: Could not get transform from /base_link to guidance (stamp=1465573533.032289) after 0.800000 seconds ("wait_for_transform_duration"=0.800000)!`

the frame id of the stereo camera is **guidance.**

how should i change the tf to get rid of this error?
thanks in advance!

matlabbe, 2016-07-04:
If you are not using a rosbag, you should not use `$ rosparam set use_sim_time true`.

Annaiyan, 2016-07-04:
thanks.Now i have recorded the bag file from a stereo camera (mounted in uav).While the camera is moving the cloud map is empty.
```
[ INFO] [1467659622.045898401, 1467652863.260609451]: Odom: quality=170, std dev=1.364105m, update time=0.041069s
[ INFO] [1467659622.120729341, 1467652863.331062206]: Odom: quality=149, std dev=1.133597m, update time=0.038144s
[ INFO] [1467659622.145212579, 1467652863.361247473]: Assembled 2 clouds (0.000019s)
[ WARN] [1467659622.145280406, 1467652863.361247473]: Cloud map is empty! (poses=3 clouds=3)
[ INFO] [1467659622.145345195, 1467652863.361247473]: rtabmap: Rate=1.00s, Limit=0.000s, RTAB-Map=0.0882s, Pub=0.0107s (local map=2, WM=2)
[ INFO] [1467659622.211668887, 1467652863.421619197]: Odom: quality=155, std dev=1.040201m, update time=0.038620s
[ INFO] [1467659622.342078317, 1467652863.556092405]: Odom: quality=155, std dev=0.890746m, update time=0.033265s
[ INFO] [1467659622.412716411, 1467652863.626597014]: Odom: quality=122, std dev=1.272623m, update time=0.033387s
[ INFO] [1467659622.514678972, 1467652863.727368814]: Odom: quality=154, std dev=1.448603m, update time=0.034120s
[ INFO] [1467659622.598679928, 1467652863.808768872]: Odom: quality=127, std dev=1.042128m, update time=0.035336s
[ INFO] [1467659622.715054746, 1467652863.929598165]: Odom: quality=118, std dev=1.023070m, update time=0.032933s
[ INFO] [1467659622.802255766, 1467652864.020225706]: Odom: quality=102, std dev=1.148970m, update time=0.039027s
[ INFO] [1467659623.005045859, 1467652864.224433359]: Odom: quality=116, std dev=1.193853m, update time=0.037626s
[ INFO] [1467659623.105956901, 1467652864.325255528]: Odom: quality=120, std dev=1.288931m, update time=0.036536s
[ INFO] [1467659623.134209404, 1467652864.345368830]: Assembled 3 clouds (0.000019s)
[ WARN] [1467659623.134273114, 1467652864.345368830]: Cloud map is empty! (poses=4 clouds=4)
[ INFO] [1467659623.134334368, 1467652864.345368830]: rtabmap: Rate=1.00s, Limit=0.000s, RTAB-Map=0.0180s, Pub=0.0099s (local map=3, WM=3)
[ INFO] [1467659623.199646709, 1467652864.415861723]: Odom: quality=128, std dev=1.599514m, update time=0.035568s
[ INFO] [1467659623.319696852, 1467652864.536814635]: Odom: quality=127, std dev=1.423485m, update time=0.039420s
[ INFO] [1467659623.408674207, 1467652864.619876489]: Odom: quality=112, std dev=1.550732m, update time=0.040889s
[ INFO] [1467659623.520839663, 1467652864.730610114]: Odom: quality=104, std dev=0.968545m, update time=0.040821s
[ INFO] [1467659623.600450217, 1467652864.811111879]: Odom: quality=106, std dev=1.034437m, update time=0.037473s
[ INFO] [1467659623.709551578, 1467652864.921817566]: Odom: quality=76, std dev=0.863984m, update time=0.041906s
[ INFO] [1467659623.850835779, 1467652865.064981025]: Odom: quality=94, std dev=0.881865m, update time=0.032997s
[ INFO] [1467659623.903378127, 1467652865.115334435]: Odom: quality=63, std dev=0.748003m, update time=0.039902s
[ INFO] [1467659624.023172057, 1467652865.236111382]: Odom: quality=66, std dev=0.816649m, update time=0.041160s
[ INFO] [1467659624.200653544, 1467652865.414596257]: Odom: quality=84, std dev=0.798265m, update time=0.042256s
[ INFO] [1467659624.232398218, 1467652865.444786689]: Odom: quality=101, std dev=0.605279m, update time=0.030987s
[ INFO] [1467659624.244533713, 1467652865.454844358]: Assembled 4 clouds (0.000020s)
[ WARN] [1467659624.244596096, 1467652865.454844358]: Cloud map is empty! (poses=5 clouds=5)
[ INFO] [1467659624.244681671, 1467652865.454844358]: rtabmap: Rate=1.00s, Limit=0.000s, RTAB-Map=0.0338s, Pub=0.0099s (local map=4, WM=4)
[ INFO] [1467659624.316388582, 1467652865.535381243]: Odom: quality=92, std dev=0.874340m, update time=0.036140s
[ INFO] [1467659624.399926144, 1467652865.617555037]: Odom: quality=79, std dev=0.707562m, update time=0.037467s
[ INFO] [1467659624.518466184, 1467652865.728405571]: Odom: quality=102, std dev=1.209603m, update time=0.041690s
[ INFO] [1467659624.624102608, 1467652865.839203273]: Odom: quality=84, std dev=0.665889m, update time=0.044298s
[ INFO] [1467659624.746105059, 1467652865.957200903]: Odom: quality=93, std dev=0.868095m, update time=0.046197s
[ INFO] [1467659624.830858461, 1467652866.047866444]: Odom: quality=101, std dev=0.828130m, update time=0.051257s
[ INFO] [1467659624.917341305, 1467652866.128480763]: Odom: quality=86, std dev=0.751460m, update time=0.037921s
[ INFO] [1467659625.080653508, 1467652866.296789110]: Odom: quality=114, std dev=0.803259m, update time=0.040939s
[ INFO] [1467659625.126824368, 1467652866.339491445]: Odom: quality=75, std dev=0.563665m, update time=0.045479s
[ INFO] [1467659625.218027740, 1467652866.430033649]: Odom: quality=105, std dev=0.747339m, update time=0.040121s
[ INFO] [1467659625.252317888, 1467652866.461795784]: Assembled 5 clouds (0.000029s)
[ WARN] [1467659625.252472847, 1467652866.471854989]: Cloud map is empty! (poses=6 clouds=6)
[ INFO] [1467659625.252588667, 1467652866.471854989]: rtabmap: Rate=1.00s, Limit=0.000s, RTAB-Map=0.0235s, Pub=0.0106s (local map=5, WM=5)
[ INFO] [1467659625.346207997, 1467652866.562026900]: Odom: quality=67, std dev=0.676600m, update time=0.041305s
[ INFO] [1467659625.410005032, 1467652866.622392231]: Odom: quality=97, std dev=1.043739m, update time=0.043944s
[ INFO] [1467659625.526205519, 1467652866.743174230]: Odom: quality=77, std dev=1.079922m, update time=0.048562s
[ INFO] [1467659625.615313843, 1467652866.833709830]: Odom: quality=94, std dev=1.584768m, update time=0.051891s
[... cut ...]
[ INFO] [1467659626.274201673, 1467652867.487347532]: Odom: quality=76, std dev=0.694279m, update time=0.033993s
[ INFO] [1467659626.319580009, 1467652867.537662077]: Odom: quality=80, std dev=0.889817m, update time=0.041932s
[ INFO] [1467659626.349803069, 1467652867.568016497]: Assembled 6 clouds (0.000024s)
[ WARN] [1467659626.349871290, 1467652867.568016497]: Cloud map is empty! (poses=7 clouds=7)

```
i cannot get the cloud map, even if the camera is hardware synchronised.what could be the issue to get the cloud map empty?

matlabbe, 2016-07-04:
A standard deviation of ~1 meter is very large. Is the camera calibration in mm, cm or meters? It should be in meters. You can try to set `cloud_max_depth` to 0 (inf instead of 4 meters) to see if you have a very large cloud.

Annaiyan, 2016-07-17:
The issues are gettting solved.Now i'm getting good odometry and the main parameters which helped to get the perfect odometry are   (made opticalflow:false)
/MinDisparity, /MaxDisparity, /WinWidth, /WinHeight, /SpeckleRange, /SpeckleWindowSize.

the rtabmap/cloud_map constructing good map,if it is static and for very slow movements.But i'm having problem while it is moving constantly.

is there anyother parameter can i  tweak to get the good map and odometry while moving?

thanks in advance!

matlabbe, 2016-07-18:
Do you have a screenshot/video of the problem?

Annaiyan, 2016-07-18:
here is the video link, 
https://www.dropbox.com/s/e4dp6l7zmgom3td/rtabmap.mp4?dl=0
 Where i'm getting good odometry but there is no map at all.Here is my launch file
```

<launch>  
  <!-- Choose visualization -->
  <arg name="rtabmapviz"              default="false" /> 
  <arg name="rviz"                    default="true" /> 
  
  <!-- Localization-only mode -->
  <arg name="localization"            default="false"/>
  
  <!-- Corresponding config files -->
  <arg name="rtabmapviz_cfg"          default="-d $(find rtabmap_ros)/launch/config/rgbd_gui.ini" />
  <arg name="rviz_cfg"                default="-d $(find rtabmap_ros)/launch/config/rgbd.rviz" />
  
  <arg name="frame_id"                default="left"/>     <!-- Fixed frame id, you may set "base_link" or "base_footprint" if they are published -->
  <arg name="time_threshold"          default="0"/>             <!-- (ms) If not 0 ms, memory management is used to keep processing time on this fixed limit. -->
  <arg name="optimize_from_last_node" default="false"/>         <!-- Optimize the map from the last node. Should be true on multi-session mapping and when time threshold is set -->
  <arg name="database_path"           default="~/.ros/rtabmap.db"/>
  <arg name="rtabmap_args"            default=""/>              <!-- delete_db_on_start, udebug -->
  <arg name="launch_prefix"           default=""/>
  
  <arg name="stereo_namespace"        default="/stereo_camera"/>
  <arg name="left_image_topic"        default="$(arg stereo_namespace)/left/image_rect_color" />
  <arg name="right_image_topic"       default="$(arg stereo_namespace)/right/image_rect" />      <!-- using grayscale image for efficiency -->
  <arg name="left_camera_info_topic"  default="$(arg stereo_namespace)/left/camera_info" />
  <arg name="right_camera_info_topic" default="$(arg stereo_namespace)/right/camera_info" />
  <arg name="approximate_sync"        default="true"/>         <!-- if timestamps of the stereo images are not synchronized -->
  <arg name="compressed"              default="false"/>
  <arg name="convert_depth_to_mm"     default="true"/>
   
  <arg name="subscribe_scan"          default="false"/>         <!-- Assuming 2D scan if set, rtabmap will do 3DoF mapping instead of 6DoF -->
  <arg name="scan_topic"              default="/scan"/>
   
  <arg name="subscribe_scan_cloud"    default="false"/>         <!-- Assuming 3D scan if set -->
  <arg name="scan_cloud_topic"        default="/scan_cloud"/>
  
  <arg name="visual_odometry"         default="true"/>          <!-- Generate visual odometry -->
  <arg name="odom_topic"              default="/odom"/>         <!-- Odometry topic used if visual_odometry is false -->
  
  <arg name="namespace"               default="rtabmap"/>
  <arg name="wait_for_transform"      default="0.1"/>
  
  <!-- Nodes -->
  <group ns="$(arg namespace)">
  
    <node if="$(arg compressed)" name="republish_left"  type="republish" pkg="image_transport" args="compressed in:=$(arg left_image_topic) raw out:=$(arg left_image_topic)" />
    <node if="$(arg compressed)" name="republish_right" type="republish" pkg="image_transport" args="compressed in:=$(arg right_image_topic) raw out:=$(arg right_image_topic)" />
  
    <!-- Odometry -->
    <node if="$(arg visual_odometry)" pkg="rtabmap_ros" type="stereo_odometry" name="stereo_odometry" output="screen" args="$(arg rtabmap_args)" launch-prefix="$(arg launch_prefix)">
      <remap from="left/image_rect"        to="$(arg left_image_topic)"/>
      <remap from="right/image_rect"       to="$(arg right_image_topic)"/>
      <remap from="left/camera_info"       to="$(arg left_camera_info_topic)"/>
      <remap from="right/camera_info"      to="$(arg right_camera_info_topic)"/>
	  
      <param name="frame_id"                 type="string" value="$(arg frame_id)"/>
      <param name="wait_for_transform_duration"       type="double"   value="$(arg wait_for_transform)"/>
      <param name="approx_sync"              type="bool"   value="$(arg approximate_sync)"/>	  
      <param name="Odom/FillInfoData"        type="string" value="true"/> 
      <param name="Vis/InlierDistance"      type="string" value="0.01"/> <!-- when EstimationType=0 -->
      <param name="Vis/FeatureType"         type="string" value="2" />
      <param name="Stereo/MinDisparity:     type=:string" value="56" />
      <param name="Stereo/MaxDisparity"      type="string" value="256" />
      <param name="Stereo/WinHeight"         type="string" value="30" />
      <param name="Stereo/WinWidth"         type="string" value="30" />
      <param name="StereoBM/MinDisparity"  type="string" value="102" />
      <param name="StereoBM/NumDisparities" type="string" value="256" />
[... 67 lines cut ...]
      <remap unless="$(arg visual_odometry)" from="odom"  to="$(arg odom_topic)"/>
    </node>
  
  </group>
  
  <!-- Visualization RVIZ -->
  <node if="$(arg rviz)" pkg="rviz" type="rviz" name="rviz" args="$(arg rviz_cfg)"/>
  <node if="$(arg rviz)" pkg="nodelet" type="nodelet" name="points_xyzrgb" args="standalone rtabmap_ros/point_cloud_xyzrgb">
    <remap from="left/image"        to="$(arg left_image_topic)"/>
    <remap from="right/image"       to="$(arg right_image_topic)"/>
    <remap from="left/camera_info"  to="$(arg left_camera_info_topic)"/>
    <remap from="right/camera_info" to="$(arg right_camera_info_topic)"/>
    <remap from="cloud"             to="voxel_cloud" />

    <param name="decimation"  type="double" value="2"/>
    <param name="voxel_size"  type="double" value="0.02"/>
    <param name="approx_sync" type="bool"   value="$(arg approximate_sync)"/>
  </node>

</launch>
```

in the rviz i'm not getting the map frame after the odometry computed.

thanks in advance.

matlabbe, 2016-07-18:
In general, look in the terminal for warnings.

 * In RVIZ, you should set the fixed frame in Global options to `/map`, not `/left`. 
 * "`Vis/InlierDistance`" seems a little small to me, it is 10 cm by default instead of 1 cm. 
 * Odometry estimation time is pretty high ~500/700 ms. What is your computer spec? What is the resolution of the images? With as high estimation time, you cannot move fast, otherwise you may get odometry lost. You can reduce the local map size of the odometry: 
  ```xml
<param name="OdomF2M/MaxSize" type="string" value="1000" />

<!-- or try Frame-to-Frame odometry with optical flow -->
<param name="Odom/Strategy" type="string" value="1" />
<param name="Vis/CorType" type="string" value="1" />
```
 * Are you able to show "voxel_cloud" cloud in RVIZ? That nodelet uses the same approach as `rtabmap` node to create the point clouds. This would help to see if at least cv::stereoBM() can compute a disparity image.

Note that `StereoBM` parameters are not used by `stereo_odometry` node. They would be used for `rtabmap` node though, for cloud reconstruction. However, I noticed a bug that they are ignored in `rtabmap`. I opened an issue for this: https://github.com/introlab/rtabmap_ros/issues/97

Annaiyan, 2016-07-18:
i'm using the resolution of 752x640 and my graphic card - nvidia gf11.
ros-indigo-ubuntu 14.04.
with the same resolution and after adding the above odometry parameters,cloud_map is working fine for slow movements by hand.(this is for initial testing).i'm planning to reduce the resolution while mounting it on a moving vehicle.Let me test and update it here,if it is working perfect.
thanks for the support.

closed 2016-11-09 by matlabbe