introlab/rtabmap_ros #94: libcurart.so.6.5 error

Exception4U, 2016-06-30:
I have cuda 7.5 but rtabmap shows this error while trying to run this launch file

roslaunch rtabmap_ros demo_stereo_outdoor.launch

/*******************************************************************************************************/

core service [/rosout] found
process[republish_left-1]: started with pid [10024]
process[republish_right-2]: started with pid [10025]
process[stereo_camera/stereo_nodelet-3]: started with pid [10026]
process[stereo_camera/stereo_image_proc-4]: started with pid [10032]
process[rtabmap/stereo_odometry-5]: started with pid [10045]
/home/iiith/ros_ws/devel/lib/rtabmap_ros/stereo_odometry: error while loading shared libraries: libcudart.so.6.5: cannot open shared object file: No such file or directory
process[rtabmap/rtabmap-6]: started with pid [10050]
/home/iiith/ros_ws/devel/lib/rtabmap_ros/rtabmap: error while loading shared libraries: libcudart.so.6.5: cannot open shared object file: No such file or directory
process[rviz-7]: started with pid [10055]
[rtabmap/stereo_odometry-5] process has died [pid 10045, exit code 127, cmd /home/iiith/ros_ws/devel/lib/rtabmap_ros/stereo_odometry left/image_rect:=/stereo_camera/left/image_rect right/image_rect:=/stereo_camera/right/image_rect left/camera_info:=/stereo_camera/left/camera_info_throttle right/camera_info:=/stereo_camera/right/camera_info_throttle odom:=/stereo_odometry __name:=stereo_odometry __log:=/home/iiith/.ros/log/8cbfd5b6-3ece-11e6-b653-507b9d90c688/rtabmap-stereo_odometry-5.log].
log file: /home/iiith/.ros/log/8cbfd5b6-3ece-11e6-b653-507b9d90c688/rtabmap-stereo_odometry-5*.log
[rtabmap/rtabmap-6] process has died [pid 10050, exit code 127, cmd /home/iiith/ros_ws/devel/lib/rtabmap_ros/rtabmap --delete_db_on_start left/image_rect:=/stereo_camera/left/image_rect_color right/image_rect:=/stereo_camera/right/image_rect left/camera_info:=/stereo_camera/left/camera_info_throttle right/camera_info:=/stereo_camera/right/camera_info_throttle odom:=/stereo_odometry __name:=rtabmap __log:=/home/iiith/.ros/log/8cbfd5b6-3ece-11e6-b653-507b9d90c688/rtabmap-rtabmap-6.log].
log file: /home/iiith/.ros/log/8cbfd5b6-3ece-11e6-b653-507b9d90c688/rtabmap-rtabmap-6*.log
[ERROR] [1467315922.742422611]: PluginlibFactory: The plugin for class 'rtabmap_ros/MapCloud' failed to load.  Error: Failed to load library /home/iiith/ros_ws/devel/lib//librtabmap_ros.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libcudart.so.6.5: cannot open shared object file: No such file or directory)
[ERROR] [1467315922.801690800]: PluginlibFactory: The plugin for class 'rtabmap_ros/MapGraph' failed to load.  Error: Failed to load library /home/iiith/ros_ws/devel/lib//librtabmap_ros.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libcudart.so.6.5: cannot open shared object file: No such file or directory)
[ERROR] [1467315922.891479409]: PluginlibFactory: The plugin for class 'rtabmap_ros/Info' failed to load.  Error: Failed to load library /home/iiith/ros_ws/devel/lib//librtabmap_ros.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libcudart.so.6.5: cannot open shared object file: No such file or directory)
[ERROR] [1467315923.439263335]: PluginlibFactory: The plugin for class 'rtabmap_ros/OrbitOriented' failed to load.  Error: Failed to load library /home/iiith/ros_ws/devel/lib//librtabmap_ros.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libcudart.so.6.5: cannot open shared object file: No such file or directory)
[rviz-7] process has finished cleanly
log file: /home/iiith/.ros/log/8cbfd5b6-3ece-11e6-b653-507b9d90c688/rviz-7*.log
/********************************************************************************************************/

matlabbe, 2016-07-04:
It looks like the libcudart.so.6.5 is not found on the PATH or LD_LIBRARY_PATH. Is OpenCV built with CUDA 7.5? or an older version (6.5)? 

This may happen if you have built OpenCV with CUDA 6.5, then unsintall CUDA to reinstall 7.5 without re-compiling OpenCV. rtabmap will link on opencv, which still link on 6.5.

Another problem would be that rtabmap_ros is linking on OpenCV binaries (by cv_bridge) instead of OpenCV built from source. It would be then better to remove opencv binaries from the computer and re-build the [vision-opencv](https://github.com/ros-perception/vision_opencv) stack against your local OpenCV.

cheers

matlabbe, 2016-11-09:
Looks ok on Ubuntu 16.04/kinetic/cuda8.

closed 2016-11-09 by matlabbe