introlab/rtabmap_ros #95: error compiling rtabmap_ros

jacksonkr, 2016-07-04:
$ catkin_make
Base path: /home/ubuntu/catkin_ws
Source space: /home/ubuntu/catkin_ws/src
Build space: /home/ubuntu/catkin_ws/build
Devel space: /home/ubuntu/catkin_ws/devel
Install space: /home/ubuntu/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/ubuntu/catkin_ws/build"
####
####
#### Running command: "make -j1 -l1" in "/home/ubuntu/catkin_ws/build"
####
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_ResetPose
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_Info
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_ListLabels
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_SetLabel
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_KeyPoint
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_Point2f
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_NodeData
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_MapData
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_GetMap
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_PublishMap
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_MapGraph
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_Goal
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_Point3f
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_Link
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_OdomInfo
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_SetGoal
[ 18%] Built target rtabmap_ros_generate_messages_lisp
[ 19%] Built target rtabmap_ros_gencfg
[ 19%] Built target sensor_msgs_generate_messages_py
[ 19%] Built target geometry_msgs_generate_messages_py
[ 19%] Built target std_msgs_generate_messages_py
[ 40%] Built target rtabmap_ros_generate_messages_py
[ 40%] Built target sensor_msgs_generate_messages_cpp
[ 40%] Built target geometry_msgs_generate_messages_cpp
[ 40%] Built target std_msgs_generate_messages_cpp
[ 58%] Built target rtabmap_ros_generate_messages_cpp
[ 59%] Built target camera
[ 87%] Built target rtabmap_ros
[ 88%] Built target data_player
[ 89%] Built target map_assembler
[ 90%] Built target map_optimizer
[ 91%] Built target odom_msg_to_tf
[ 93%] Built target rgbd_odometry
[ 95%] Built target rtabmap
[ 95%] Built target rtabmap_ros_generate_messages
[ 96%] Building CXX object rtabmap_ros/CMakeFiles/rtabmapviz.dir/src/GuiWrapper.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
make[2]: *** [rtabmap_ros/CMakeFiles/rtabmapviz.dir/src/GuiWrapper.cpp.o] Error 4
make[1]: *** [rtabmap_ros/CMakeFiles/rtabmapviz.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j1 -l1" failed

matlabbe, 2016-07-05:
Try again, this error happens when the system runs out of memory. You can increase your swap space if needed. Normally, on a system with 2GB of RAM, it should build.

jacksonkr, 2016-07-05:
@matlabbe I'm bulding GTSAM as an option for rtabmap and I accidentally posted here instead of GTSAM's page. Regardless, thanks for your informative attitude!

jacksonkr, 2016-07-05:
I thought this was on unrelated issue with GTSAM but I was mistaken. I've restarted and am repeatedly running into the same issue.

matlabbe, 2016-07-05:
You can use "$ top" to see how much memory the compiler is using. Indeed, if GuiWrapper.cpp cannot be built, rtabmapviz won't be available.

jacksonkr, 2016-07-05:
I've been doing everything over SSH so far. I decided to hook up a keyboard to the bot itself and `make` ran with no hiccups. I have no idea what this means or how to even investigate it. Anyway, thanks @matlabbe for all your work with RTABMAP. You are a gentleman and a scholar!

closed 2016-07-05 by jacksonkr