introlab/rtabmap_ros #96: Create nodelets for rgbd_odometry, rtabmap and rtabmapviz nodes

matlabbe, 2016-07-06:
So by launching all these nodelets in camera_manager nodelet, we could avoid all serialization/deserialization of the image topics like in the standalone application.

Danmou, 2017-03-22:
What is the status on this? I see you have made some commits, but is it usable? I tried launching the nodes as nodelets (e.g. `<node pkg="nodelet" type="nodelet" name="rtabmap" output="screen" args="load rtabmap_ros/rtabmap camera_nodelet_manager">`) but it didn't seem to work - no error messages, but nothing published either. I am running rtabmap on an embedded platform so any optimization would be greatly appreciated.

matlabbe, 2017-03-22:
See [test_rtabmap_nodelets.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/tests/test_rtabmap_nodelets.launch) for an example of usage.

closed 2017-03-22 by matlabbe