introlab/rtabmap_ros #100: Reset odometry when odometry lost

willdzeng, 2016-07-19:
Just thinking maybe to call reset odometry service whenever it get lost, I tested a few times just now and it seems like working good.

For example, whenever it shows:

> [ WARN] [1468969468.604413838]: rtabmap: Could not get transform from odom to /base_link after 0.200000 seconds (for stamp=1468969468.277366)!
> [ WARN] [1468969468.604491661]: Could not get TF transform from odom to /base_link, sensors will not be synchronized with odometry pose.
> [ERROR] (2016-07-19 16:04:28.605) Rtabmap.cpp:881::process() RGB-D SLAM mode is enabled and no odometry is provided. Image 4016 is ignored!

I do 

`rosservice call /rtabmap/reset_odom`

It find the pose again by research the database.

Is there any downside of doing this?

matlabbe, 2016-07-20:
That depends what we want to do. By default, the parameters are set for "3D scanning" behavior. Odometry is not reset when lost to keep consistent the links between consecutive scans.

When calling `reset_odom`, `rtabmap` will start a new map, hiding the old one until a loop closure is found with the previous map. If you don't want rtabmap to start a new map when odometry is reset and wait until a first loop closure is found, you can set `Rtabmap/StartNewMapOnLoopClosure` to `true`.

If parameter `Odom/ResetCountdown` is set to `1` (default `0`=disabled), odometry will automatically reset one frame after being lost, i.e., it has the same effect than calling `reset_odom` service.

In **summary**, by setting `Rtabmap/StartNewMapOnLoopClosure=true` and  `Odom/ResetCountdown=1`, odometry will be automatically reset when lost and `rtabmap` will wait to localize with a previous location before continuing mapping.

cheers

willdzeng, 2016-07-21:
Cool, it's working as expected.

closed 2016-07-21 by willdzeng