introlab/rtabmap_ros #99: rtabmapviz crashes: /usr/include/boost/smart_ptr/shared_ptr.hpp:646: typename

willdzeng, 2016-07-19:
Hey mathiue,

I am try to run the new commits, and I had this crashes when the rtabmapviz running for a while( around 5s)

It works fine without rtabmapviz.

Btw, the new commits are pretty cool, is it octomap? 

rtabmapviz: /usr/include/boost/smart_ptr/shared_ptr.hpp:646: typename boost::detail::sp_dereference<T>::type boost::shared_ptr<T>::operator*() const [with T = boost::signals2::detail::signal1_impl<void, const pcl::visualization::MouseEvent&, boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void(const pcl::visualization::MouseEvent&)>, boost::function<void(const boost::signals2::connection&, const pcl::visualization::MouseEvent&)>, boost::signals2::mutex>; typename boost::detail::sp_dereference<T>::type = boost::signals2::detail::signal1_impl<void, const pcl::visualization::MouseEvent&, boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void(const pcl::visualization::MouseEvent&)>, boost::function<void(const boost::signals2::connection&, const pcl::visualization::MouseEvent&)>, boost::signals2::mutex>&]: Assertion `px != 0' failed.

[rtabmap/rtabmapviz-3] process has died [pid 15104, exit code -6, cmd /home/zeng/test/devel/lib/rtabmap_ros/rtabmapviz -d /home/zeng/test/src/rtabmap_ros/launch/config/rgbd_gui.ini rgb/image:=/camera/rgb/image_rect_color depth/image:=/camera/depth/image_rect_color rgb/camera_info:=/camera/rgb/camera_info scan:=/scan scan_cloud:=/scan_cloud __name:=rtabmapviz __log:=/home/zeng/.ros/log/f179e308-4d50-11e6-8ecb-7c5cf8c32cb0/rtabmap-rtabmapviz-3.log].
log file: /home/zeng/.ros/log/f179e308-4d50-11e6-8ecb-7c5cf8c32cb0/rtabmap-rtabmapviz-3*.log

matlabbe, 2016-07-19:
Hi Di,

I cannot reproduce the error. Yes, the new commits include the octomap and other features: see release [0.11.8](https://github.com/introlab/rtabmap/releases/tag/0.11.8). Note that the ocotmap is not shown by default, it should be activated here:
[image]

Decreasing the maximum tree depth (between 13 and 15) helps to increase visualization performance (less cubes drawn). Only level 16 has RGB colored cubes.

The octomap will indeed clear dynamic obstacles in the 3D map, but doing so there is a little time overhead for 3D ray tracing when adding new data to octomap.

Does your error easily reproducible on your system?

cheers,
Mathieu

willdzeng, 2016-07-19:
Yeah, whenever I scroll in rtabmapviz, it crashes. Let me try change the parameters to see if it still happens

matlabbe, 2016-07-19:
Is still crashing even if the octomap is not shown/checked?

willdzeng, 2016-07-19:
I am having problem to use "rtabmap" from command line.

rtabmap: error while loading shared libraries: librtabmap_core.so.0.11: cannot open shared object file: No such file or directory

How do I change the Octomap Parameter in launch file?

willdzeng, 2016-07-19:
Ohh, I fixed the ld libraires stuff,
since I installed by source  I did:

export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib/x86_64-linux-gnu

to fix the can't find library issue, maybe put this in README?

willdzeng, 2016-07-19:
I am using my old launch file.
by using the launch file, Octomap is showing by default.

matlabbe, 2016-07-20:
By default now (see this [kinetic related issue](https://github.com/introlab/rtabmap/issues/97)), rtabmap library doesn't use anymore multi-arch install folders. Libraries are installed in `/usr/local/lib` instead of `/usr/local/lib/x86_64-linux-gnu` by default. This would avoid modifying `LD_LIBRARY_PATH`. If you didn't clear your cmake cache, `x86_64-linux-gnu` will still be used.

On ROS, you should get the octomap on the referred published topics starting with `/rtabmap/octomap_****`. There is still an issue with the official octomap rviz plugins, which cannot show the colored version of the octomap (I think the octomap_rivz_plugins package should be rebuilt with ColorOctree support). For convenience, you can use a PointCloud2 display and subscribe to `/rtabmap/octomap_cloud`. For rtabmapviz, once you activated the octomap option in Preferences dialog and saved the config file (File->Save config) it should be activated when restarting rtabmapviz.

The ROS parameters for the octomap (the one published by `rtabmap` node) are [here](https://github.com/introlab/rtabmap_ros/blob/3c49f3139ddb2fda447e0d4ff5e8126f63462706/src/MapsManager.cpp#L152-L168):
```xml
<node pkg="rtabmap_ros" type="rtabmap" name="rtabmap">
   <param name="grid_cell_size" value="0.05"/>
   <param name="octomap_ground_is_obstacle" value="false"/>
   <param name="octomap_tree_depth" value="16"/>
   ...
</node>
```

The octomap is not supposed to be shown by default. I [fixed](https://github.com/introlab/rtabmap/commit/1dca6116be79d2b7b31a3aa5e9e57afdaa7f58d8) this bug.

I tried on another computer, not still able to reproduce the boost shared point error when zooming in the 3D map...

willdzeng, 2016-07-20:
hey mathieu,

Still having the same problem even if I canceled the octomap.

[image]

Btw, the ini file read by the launch file is in rtabmap_ros
`$(find rtabmap_ros)/launch/config/rgbd_gui.ini`
So I just copy 

```
General\octomap=false
General\octomap_depth=16
General\octomap_ground_is_obstacle=true
```
into the rgbd_gui.ini in the rtabmap_ros/launch/config

willdzeng, 2016-07-20:
Fixed this by uninstall PCL1.8 and install PCL1.7

closed 2016-07-20 by willdzeng

matlabbe, 2016-07-21:
I tried with PCL1.8 on Windows 10 and still not able to reproduce the error (well I don't have a Ubuntu 16.04 to test right now). There is maybe a combination of VTK version, Qt version, PCL version and OS version that may cause the problem.

willdzeng, 2016-07-21:
FYI, I am using 
Ubuntu 14.04
QMake version 2.01a
Qt version 4.8.6 
libboost: Version: 1.54.0.1ubuntu1
VTK: 5.8.0-14.lubuntu3

matlabbe, 2016-07-21:
Well, I am pretty much using the same configuration on my laptop: Ubuntu 14.04 with the default installed Qt(4.8.6)/VTK(5.8.0-14)/PCL(1.7.1-3)/boost(1.54.0.1).

I'll give a try with PCL 1.8 on my system too then, see if I can reproduce this.

matlabbe, 2016-07-21:
Did you rebuild `pcl_ros` and all dependencies too? I have some library conflicts (PCL 1.8 shadowed by PCL 1.7.1) when building `rtabmap_ros`, so `rtabmapviz` is crashing on start. I should rebuild `pcl_ros` with PCL 1.8 but I have many packages depending on `pcl_ros` binaries (which are on 1.7.1). As this is a complicated setup, I will let this issue as "won't be fixed". 

However, I can test `rtabmap` standalone with PCL 1.8 and it works.

willdzeng, 2016-07-21:
Yes, I build [pcl_conversions ](https://github.com/ros-perception/pcl_conversions) from source indigo branch, and it works