introlab/rtabmap_ros #102: 3d stereo SLAM bad performance

neweirik, 2016-07-26:
Hi, Im using the zed stereo camera on the tegra TX1. When trying to build a map it does not take mouch movement before I get this error message

[ERROR] (2016-07-26 07:37:37.499) Rtabmap.cpp:880::process() RGB-D SLAM mode is enabled and no odometry is provided. Image 6807 is ignored!
[ WARN] (2016-07-26 07:37:37.592) OdometryF2M.cpp:227::computeTransform() Registration failed: "Not enough inliers 0/20 between -1 and 0"

When the zed stands on my desk it works fine, slow movements is allso fine. But with "normal" movements like on a remote controlled car, it crashes. Is the an issue? Or Im I doing something wrong?

-Eirik

matlabbe, 2016-07-26:
Hi Eirik,

If you are on a car, I've found that using these parameters can be good (Optical Flow and 3D->2D PnP registration):
```ini
Odom/Strategy=1       # Frame-to-Frame odometry
Odom/GuessMotion=true # Use guess from previous motion for correspondences
Vis/EstimationType=1  # 2D->3D PnP
Vis/CorType=1         # Optical Flow
```

If your images have high resolution, you may want to increase `GFTT/MinDistance` (e.g. 10, default 5 pixels) to have more distributed features across the image.

Look also how much time is required for odometry, it should be under 100 ms to handle "normal" movements.

cheers,
Mathieu

willdzeng, 2016-07-28:
Hi Eirik,

You can try use VGA mode in zed and make the frame rate higher to see if the visual odometry getting better.

neweirik, 2016-08-02:
Thank you for the response. Im a little confused where to change these settings, the only parameters I can find is in the Parameters.h file from rtabmap, not rtabmap_ros. Im installing rtabmap from apt-get, so changing the parameterfile means building everything from source?  Im working on outdoor navigation, im generating a 2d map (proj_map) from rtabmap and then running a cost function on it, linking everything to movebase. Everything is really unstable and when odometry is lost I have to restart everything. I see that the default value for RTABMAP_PARAM(Odom, ResetCountdown is 0. In other works: it never resets. Does there not exist rosparam or yaml files for there settings?  Zed is in vga

matlabbe, 2016-08-02:
As you are using the binaries, I recommend to use [ros-shadow-fixed](http://wiki.ros.org/ShadowRepository) version (latest version).

To set [rtabmap's parameters](https://github.com/introlab/rtabmap/blob/master/corelib/include/rtabmap/core/Parameters.h), you can do it like that in the launch file:
```xml
<node pkg="rtabmap_ros" type="rgbd_odometry" name="rgbd_odometry">
   <param name="Odom/ResetCountdown" type="string" value="10"/>
</node>
```

If you use [rtabmap.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/rtabmap.launch), you can do this too:
```bash
$ roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--Odom/ResetCountdown 10"
```

Another way is using `*.ini` config file, `~/config.ini:`
```ini
[Core]
Odom\ResetCountdown=10
```
Use it in a launch file:
```xml
<node pkg="rtabmap_ros" type="rgbd_odometry" name="rgbd_odometry">
   <param name="config_path" type="string" value="~/config.ini"/>
</node>
```
Or with `rtabmap.launch`:
```bash
$ roslaunch rtabmap_ros rtabmap.launch cfg:=~/config.ini
```

cheers

neweirik, 2016-08-03:
Thank you for the quick response, now the performance of the navigation is a lot better! However I have one more question. When navigating on the projected map, the global cost map becomes the same size. When I look at your navigation video you are never setting goals outside the map (transparent square, guess thats your global cost map). I want to navigate on a map that is a lot bigger than the one projected from the point cloud, but goals outside the projected map is not accepted( im using rviz to set goals). Is it possible to make the projected map scale according to goals from rviz?  

-Eirik

matlabbe, 2016-08-03:
Hi Erik,

You can initialize the minimum size of the map with `grid_size` parameter:
<node pkg="rtabmap_ros" type="rtabmap" name="rtabmap">
   <param name="grid_size" type="double" value="50"/> 
</node>

cheers,
Mathieu

closed 2016-11-09 by matlabbe