introlab/rtabmap_ros #103: Installing in arm processor

Annaiyan, 2016-07-27:
Hi,

I'm installing the rtabmap_ros in arm processor  (NVIDIA Tegra K1). I have the following issues,
1.since the memory is very limited . Always getting the c++ internal compiler error (killed program cc1plus).This is due to the  low memory.
2.so i deleted some of the executables in cmake list - (rgbd_odometry, map_optimizer, map_assembler, camera) .

My objective is to use the stereo camera and get the odometry and cloud_map .

I want your recommendation to remove the rest of the things which are not mandatory for the above objective.

many thanks in advance.

matlabbe, 2016-07-27:
How much RAM do you have? You can increase the SWAP memory: https://devtalk.nvidia.com/default/topic/764828/-hint-configure-swap-on-jetson-tk1/

Use `$ make` and `$ catkin_make -j1` to make sure to launch only one compilation at a time.

Annaiyan, 2016-07-28:
I have 2gb of ram and i created a 2gb of swap.Then i did catkin_make -j1  and got this error.

/home/ubuntu/catkin_ws/src/rtabmap_ros/src/DbPlayerNode.cpp: In function 'int main(int, char**)':
/home/ubuntu/catkin_ws/src/rtabmap_ros/src/DbPlayerNode.cpp:106:25: error: invalid user-defined conversion from 'int' to 'const string& {aka const std::basic_string<char>
&}' [-fpermissive]
  if(!reader.init(startId))
                         ^
In file included from /usr/include/c++/4.8/string:53:0,                                                                                                            [0/515]
                 from /opt/ros/indigo/include/ros/platform.h:55,
                 from /opt/ros/indigo/include/ros/time.h:53,
                 from /opt/ros/indigo/include/ros/ros.h:38,
                 from /home/ubuntu/catkin_ws/src/rtabmap_ros/src/DbPlayerNode.cpp:1:
/usr/include/c++/4.8/bits/basic_string.tcc:212:5: note: candidate is: std::basic_string<_CharT, _Traits, _Alloc>::basic_string(const _CharT*, const _Alloc&) [with _CharT = char; _Traits = std::char_traits<char>; _Alloc = std::allocator<char>] <near match>
     basic_string<_CharT, _Traits, _Alloc>::
     ^
/usr/include/c++/4.8/bits/basic_string.tcc:212:5: note:   no known conversion for argument 1 from 'int' to 'const char*'
/home/ubuntu/catkin_ws/src/rtabmap_ros/src/DbPlayerNode.cpp:106:25: error: invalid conversion from 'int' to 'const char*' [-fpermissive]
  if(!reader.init(startId))
                         ^
In file included from /usr/include/c++/4.8/string:53:0,
                 from /opt/ros/indigo/include/ros/platform.h:55,
                 from /opt/ros/indigo/include/ros/time.h:53,
                 from /opt/ros/indigo/include/ros/ros.h:38,
                 from /home/ubuntu/catkin_ws/src/rtabmap_ros/src/DbPlayerNode.cpp:1:
/usr/include/c++/4.8/bits/basic_string.tcc:212:5: error:   initializing argument 1 of 'std::basic_string<_CharT, _Traits, _Alloc>::basic_string(const _CharT*, const _Alloc&) [with _CharT = char; _Traits = std::char_traits<char>; _Alloc = std::allocator<char>]' [-fpermissive]
     basic_string<_CharT, _Traits, _Alloc>::
     ^
/home/ubuntu/catkin_ws/src/rtabmap_ros/src/DbPlayerNode.cpp:130:2: error: 'OdometryEvent' is not a member of 'rtabmap'
  rtabmap::OdometryEvent odom = reader.getNextData();
  ^
/home/ubuntu/catkin_ws/src/rtabmap_ros/src/DbPlayerNode.cpp:130:25: error: expected ';' before 'odom'
  rtabmap::OdometryEvent odom = reader.getNextData();
                         ^
/home/ubuntu/catkin_ws/src/rtabmap_ros/src/DbPlayerNode.cpp:132:21: error: 'odom' was not declared in this scope
  while(ros::ok() && odom.data().id())
                     ^
/home/ubuntu/catkin_ws/src/rtabmap_ros/src/DbPlayerNode.cpp:450:14: error: 'uSleep' was not declared in this scope
    uSleep(100);
              ^
In file included from /home/ubuntu/catkin_ws/src/rtabmap_ros/src/DbPlayerNode.cpp:18:0:
/usr/local/lib/rtabmap-0.11/../../include/rtabmap-0.11/rtabmap/core/DBReader.h:76:13: error: 'rtabmap::SensorData rtabmap::DBReader::getNextData(rtabmap::CameraInfo*)' is
 private
  SensorData getNextData(CameraInfo * info = 0);
             ^
/home/ubuntu/catkin_ws/src/rtabmap_ros/src/DbPlayerNode.cpp:455:29: error: within this context
   odom = reader.getNextData();
                             ^
make[2]: *** [rtabmap_ros/CMakeFiles/data_player.dir/src/DbPlayerNode.cpp.o] Error 1
make[1]: *** [rtabmap_ros/CMakeFiles/data_player.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j1" failed

the above error is due to the memory issue or something else?

Annaiyan, 2016-07-29:
the issue solved by making more space of 4Gb and then i tried catkin_make -j1 --pkg rtabmap_ros.
got the successful compilation.
many thanks for the support.

closed 2016-07-29 by Annaiyan