introlab/rtabmap_ros #105: unable to start rtabmap

abhignans, 2016-07-29:
hello i have installed rtabmap_ros from source.
all went well but after i open the application it does not start and says segmentation error and application doesnt start forever
 here is the screen shot.
can anybody suggest me way to proceed or how to debug it even
[image]

willdzeng, 2016-08-01:
did you installed rtabmap from ROS before?
if it is, you can try remove it and re-install rtabmap

Can you print your
`ls /usr/local/lib`
and
`ls /usr/local/lib/x86_64-linux-gnu`
here?

matlabbe, 2016-11-09:
closed by inactivity

closed 2016-11-09 by matlabbe