introlab/rtabmap_ros #106: variables set to not found

ArpitSoni9, 2016-08-02:
steps followed : $ sudo apt-get install ros-indigo-rtabmap-ros

copying the downloaded package in my catkin workspace scr folder

then replacing config file with RTABMap 0.11.8 config file coz it was showing error

then errors are after catkin_make-----

CMake Warning at rtabmap_ros-indigo-devel/CMakeLists.txt:291 (MESSAGE):

Found RTAB-Map built without its GUI library. Node rtabmapviz will not be built!

CMake Error: The following variables are used in this project, but they are set to NOTFOUND. Please set them or make sure they are set and tested correctly in the CMake files: RTABMap_CORE_RELEASE linked by target "camera" in directory /home/hns9kor/arpit/src/rtabmap_ros-indigo-devel

linked by target "data_player" in directory /home/hns9kor/arpit/src/rtabmap_ros-indigo-devel
linked by target "map_assembler" in directory /home/hns9kor/arpit/src/rtabmap_ros-indigo-devel
linked by target "map_optimizer" in directory /home/hns9kor/arpit/src/rtabmap_ros-indigo-devel
linked by target "odom_msg_to_tf" in directory /home/hns9kor/arpit/src/rtabmap_ros-indigo-devel
linked by target "rgbd_odometry" in directory /home/hns9kor/arpit/src/rtabmap_ros-indigo-devel
linked by target "rtabmap" in directory /home/hns9kor/arpit/src/rtabmap_ros-indigo-devel
linked by target "rtabmap_ros" in directory /home/hns9kor/arpit/src/rtabmap_ros-indigo-devel
linked by target "stereo_camera" in directory /home/hns9kor/arpit/src/rtabmap_ros-indigo-devel
linked by target "stereo_odometry" in directory /home/hns9kor/arpit/src/rtabmap_ros-indigo-devel
RTABMap_UTILITE_RELEASE

linked by target "camera" in directory /home/hns9kor/arpit/src/rtabmap_ros-indigo-devel
linked by target "data_player" in directory /home/hns9kor/arpit/src/rtabmap_ros-indigo-devel
linked by target "map_assembler" in directory /home/hns9kor/arpit/src/rtabmap_ros-indigo-devel
linked by target "map_optimizer" in directory /home/hns9kor/arpit/src/rtabmap_ros-indigo-devel
linked by target "odom_msg_to_tf" in directory /home/hns9kor/arpit/src/rtabmap_ros-indigo-devel
linked by target "rgbd_odometry" in directory /home/hns9kor/arpit/src/rtabmap_ros-indigo-devel
linked by target "rtabmap" in directory /home/hns9kor/arpit/src/rtabmap_ros-indigo-devel
linked by target "rtabmap_ros" in directory /home/hns9kor/arpit/src/rtabmap_ros-indigo-devel
linked by target "stereo_camera" in directory /home/hns9kor/arpit/src/rtabmap_ros-indigo-devel
linked by target "stereo_odometry" in directory /home/hns9kor/arpit/src/rtabmap_ros-indigo-devel
-- Configuring incomplete, errors occurred!

See also "/home/hns9kor/arpit/build/CMakeFiles/CMakeOutput.log".

See also "/home/hns9kor/arpit/build/CMakeFiles/CMakeError.log".

make: * [cmake_check_build_system] Error 1

Invoking "make cmake_check_build_system" failed

matlabbe, 2016-08-02:
"copying the downloaded package in my catkin workspace scr folder" is wrong. Keep the binaries installed in `/opt/ros/indigo`. Make sure you source the `/opt/ros/indigo/setup.bash` to use them. See [installation instructions](http://wiki.ros.org/indigo/Installation/Ubuntu) from ROS.

If you want to build rtabmap from source, you should uninstall the binaries, then follow these [instructions](https://github.com/introlab/rtabmap_ros#build-from-source).

ArpitSoni9, 2016-08-04:
thanks.. rtabmap is working now.... now i want to navigate my robot using the map... how should i proceed????

ArpitSoni9, 2016-08-04:
thanks.... that worked... now i want to navigate my robot to given position in the map... how should i proceed???? like this tutorial http://wiki.ros.org/rtabmap_ros/Tutorials/MappingAndNavigationOnTurtlebot.... but my robot is not turtle bot.

matlabbe, 2016-08-04:
Look for [move_base](http://wiki.ros.org/move_base?distro=kinetic) examples. There is also the [Stereo outdoor navigation tutorial](http://wiki.ros.org/rtabmap_ros/Tutorials/StereoOutdoorNavigation) that you can look. If your robot is similar to turtlebot, you can check the launch file included in the link you referenced. 

When you have setup the navigation stack with your robot, the [navigation tuning guide](http://wiki.ros.org/navigation/Tutorials/Navigation%20Tuning%20Guide) is a good documentation about tuning parameters for your robot.

closed 2016-11-09 by matlabbe