introlab/rtabmap_ros #107: error while running demo launch file

Exception4U, 2016-08-04:
roslaunch rtabmap_ros demo_robot_mapping.launch 
... logging to /home/swahana/.ros/log/1046bfda-5a40-11e6-81ca-54a05086584a/roslaunch-AsusRed-3481.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://AsusRed:43570/

SUMMARY
========

PARAMETERS
 * /points_xyzrgb/depth/image_transport: compressedDepth
 * /points_xyzrgb/queue_size: 10
 * /points_xyzrgb/rgb/image_transport: compressed
 * /points_xyzrgb/voxel_size: 0.01
 * /rosdistro: indigo
 * /rosversion: 1.11.16
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/Optimizer/Slam2D: true
 * /rtabmap/rtabmap/RGBD/NeighborLinkRefining: true
 * /rtabmap/rtabmap/RGBD/OptimizeFromGraphEnd: false
 * /rtabmap/rtabmap/RGBD/ProximityBySpace: true
 * /rtabmap/rtabmap/RGBD/ProximityByTime: false
 * /rtabmap/rtabmap/Reg/Force3DoF: true
 * /rtabmap/rtabmap/Reg/Strategy: 1
 * /rtabmap/rtabmap/Vis/InlierDistance: 0.1
 * /rtabmap/rtabmap/depth/image_transport: compressedDepth
 * /rtabmap/rtabmap/frame_id: base_footprint
 * /rtabmap/rtabmap/rgb/image_transport: compressed
 * /rtabmap/rtabmap/subscribe_depth: True
 * /rtabmap/rtabmap/subscribe_scan: True
 * /rtabmap/rtabmap/wait_for_transform: True
 * /rtabmap/rtabmapviz/depth/image_transport: compressedDepth
 * /rtabmap/rtabmapviz/frame_id: base_footprint
 * /rtabmap/rtabmapviz/rgb/image_transport: compressed
 * /rtabmap/rtabmapviz/subscribe_depth: True
 * /rtabmap/rtabmapviz/subscribe_scan: True
 * /rtabmap/rtabmapviz/wait_for_transform: True
 * /use_sim_time: True

NODES
  /rtabmap/
    rtabmap (rtabmap_ros/rtabmap)
    rtabmapviz (rtabmap_ros/rtabmapviz)
  /
    points_xyzrgb (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
ERROR: cannot launch node of type [rtabmap_ros/rtabmap]: can't locate node [rtabmap] in package [rtabmap_ros]
ERROR: cannot launch node of type [rtabmap_ros/rtabmapviz]: can't locate node [rtabmapviz] in package [rtabmap_ros]
process[points_xyzrgb-3]: started with pid [3505]
[ERROR] [1470314234.768083520, 1368730028.579204662]: Failed to load nodelet [/points_xyzrgb] of type [rtabmap_ros/point_cloud_xyzrgb] even after refreshing the cache: Could not find library corresponding to plugin rtabmap_ros/point_cloud_xyzrgb. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1470314234.768154113, 1368730028.579204662]: The error before refreshing the cache was: Could not find library corresponding to plugin rtabmap_ros/point_cloud_xyzrgb. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[points_xyzrgb-3] process has died [pid 3505, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet standalone rtabmap_ros/point_cloud_xyzrgb rgb/image:=/data_throttled_image depth/image:=/data_throttled_image_depth rgb/camera_info:=/data_throttled_camera_info cloud:=voxel_cloud __name:=points_xyzrgb __log:=/home/swahana/.ros/log/1046bfda-5a40-11e6-81ca-54a05086584a/points_xyzrgb-3.log].
log file: /home/swahana/.ros/log/1046bfda-5a40-11e6-81ca-54a05086584a/points_xyzrgb-3*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

matlabbe, 2016-08-04:
Did you install the binaries or build rtabmap from source? Make sure to source the setup.bash, assuming you are on Indigo:
```bash
$ source /opt/ros/indigo/setup.bash
$ source ~/catkin_ws/devel/setup.bash
```

Exception4U, 2016-08-04:
I build rtabmap core pacakge from source ... later I build rtamap_ros ...

I sourced those above paths allready... still it donnot file those nodes

matlabbe, 2016-08-04:
The error is like rtabmap is not installed on the system. Can you run rtabmap standalone?
```bash
$ rtabmap
```
It should be installed in `~/catkin_ws/devel/bin` or `/usr/local/bin`.

closed 2016-11-09 by matlabbe