introlab/rtabmap_ros #108: rtabmap/obstacles

Annaiyan, 2016-08-08:
hi,
i'm getting good odometry as well as map from the stereo camera .i'm trying to extract the obstacles using the nodelet rtbamap/obstacles like below:

` <node pkg="nodelet" type="nodelet" name="stereo_nodelet"  args="manager"/>
   
        <node pkg="nodelet" type="nodelet" name="disparity2cloud" args="load rtabmap_ros/point_cloud_xyz stereo_nodelet">
	<remap from="disparity/image"       to="disparity"/>
	<remap from="disparity/camera_info" to="right/camera_info"/>
	<remap from="cloud"                 to="cloudXYZ"/>	
	<param name="voxel_size" type="double" value="0.05"/>
	<param name="decimation" type="int" value="0"/>
	<param name="max_depth" type="double" value="1"/>
	</node>
	<node pkg="nodelet" type="nodelet" name="obstacles_detection" args="load rtabmap_ros/obstacles_detection stereo_nodelet">
    	<remap from="cloud" to="cloudXYZ"/>
	

	<param name="frame_id" type="string" value="left"/>		
	<param name="map_frame_id" type="string" value="map"/>
	<param name="wait_for_transform" type="bool" value="true"/>
	<param name="min_cluster_size" type="int" value="10"/>
	<param name="max_obstacles_height" type="double" value="0.0"/>
        </node>`
The topic is publishing but there is no message/data in it. Am i using the nodelet in correct way?
how to solve this issue?

regards,
arun

matlabbe, 2016-08-08:
Normally the disparity image should match the left image, so the `left` camera_info would be used instead of the `right`. The `max_depth` is 1 meter, it is quite small if your cloud is over 1 meter. Is `cloudXYZ` empty?

For `obstacles_detection`, the `frame_id` should be in world coordinate: with x axis forward and z axis upward (default ros coordinate). For example, it could be your `base_link` frame. Your `left` frame may be in image coordinate: z forward, x right and y downward. That can be fixed by adding a static transform:
```xml
<arg name="pi/2" value="1.5707963267948966" />
<arg name="optical_rotate" value="0 0 0 -$(arg pi/2) 0 -$(arg pi/2)" />
<node pkg="tf" type="static_transform_publisher" name="camera_base_link"
        args="$(arg optical_rotate) base_link left 100" /> 
``` 

Summary:
```xml

<arg name="pi/2" value="1.5707963267948966" />
<arg name="optical_rotate" value="0 0 0 -$(arg pi/2) 0 -$(arg pi/2)" />
<node pkg="tf" type="static_transform_publisher" name="camera_base_link"
        args="$(arg optical_rotate) base_link left 100" />

<node pkg="nodelet" type="nodelet" name="disparity2cloud" args="load rtabmap_ros/point_cloud_xyz stereo_nodelet">
  <remap from="disparity/image"       to="disparity"/>
  <remap from="disparity/camera_info" to="left/camera_info"/>
  <remap from="cloud"                 to="cloudXYZ"/> 

  <param name="voxel_size" type="double" value="0.05"/>
  <param name="decimation" type="int"    value="1"/>
  <param name="max_depth" type="double"  value="0"/>
</node>

<node pkg="nodelet" type="nodelet" name="obstacles_detection" args="loadrtabmap_ros/obstacles_detection stereo_nodelet">
  <remap from="cloud" to="cloudXYZ"/>

  <param name="frame_id"             type="string" value="base_link"/>     
  <param name="wait_for_transform"   type="bool"   value="true"/>
  <param name="min_cluster_size"     type="int"    value="10"/>
  <param name="cluster_radius"       type="double" value="0.1"/>
  <param name="max_obstacles_height" type="double" value="0.0"/>
</node>`
```

As you voxelize at 5 cm, I increased the `cluster_radius` to 10 cm to make sure to find clusters with at least 10 points.

cheers

Annaiyan, 2016-08-08:
many thanks for the info.
i tried the above recommedation but the the rtabmap/cloudXYZ is empty.
is it advisable to remap from cloud_map if the cloudXYZ is empty?

matlabbe, 2016-08-08:
No, they don't mean the same thing. Can you show the stereo_view to see what the disparity image looks like? http://wiki.ros.org/image_view#image_view.2BAC8-diamondback.stereo_view

Do you use [stereo_img_proc](http://wiki.ros.org/stereo_image_proc) to rectify your images? If so, you can look at the `points` output if it is empty too.

Annaiyan, 2016-08-09:
Yes, i 'm using  stereo_img_proc to rectify the images and also i can see the points2 in rviz.
Here is the stereo_view,
(https://www.dropbox.com/s/31dw3j31mu6r45z/stereo_view.png?dl=0)

but still the cloudXYZ is empty!

matlabbe, 2016-08-09:
Indeed this is strange as the disparity has values. Maybe a bug with the disparity input of `point_cloud_xyz` nodelet. Can you record a small rosbag with (assuming your topics are in `stereo` namespace, change it if not):
```bash
$ rosbag record stereo/disparity stereo/left/camera_info
```

Annaiyan, 2016-08-09:
here is the bag file:
https://www.dropbox.com/s/lr7u0eu4j1cf3fe/disparity.bag?dl=0
let me know if it works in your system.

matlabbe, 2016-08-09:
The disparity image looks not standard. It contains almost only **negative disparity values**!? The info in the disparity image:
```
f: 320.769622803
T: -0.0
valid_window: 
  x_offset: 35
  y_offset: 7
  height: 4294967281
  width: 4294967225
  do_rectify: False
min_disparity: -35.0
max_disparity: 28.0
delta_d: 0.0625
```
`T`, the baseline, should be greater than `0`. In comparison, here is the disparity info from the [stereo outdoor tutorial](http://wiki.ros.org/rtabmap_ros/Tutorials/StereoOutdoorMapping):
```
f: 487.608734131
T: 0.11969166249
valid_window: 
  x_offset: 134
  y_offset: 7
  height: 4294967281
  width: 4294967154
  do_rectify: False
min_disparity: 0.0
max_disparity: 127.0
delta_d: 0.0625
```
Note how `min_disparity` is not negative here. Well, negative disparity means that points are behind the cameras, which is not possible, unless the cameras are not parallel. The point cloud generated by `stereo_img_proc` may be inverted.

The left camera_info should have `P(0,3)=0` (here it is `-44`). Only the right camera should have `P(0,3)<0`:
```
header: 
  seq: 9786
  stamp: 
    secs: 1470759511
    nsecs: 898991776
  frame_id: /left
height: 240
width: 376
distortion_model: plumb_bob
D: [0.039330000000000004, -0.06547, 0.0019100000000000002, 0.0021000000000000003, 0.0]
K: [283.8764, 0.0, 181.98479, 0.0, 282.28735, 126.73881, 0.0, 0.0, 1.0]
R: [0.9998400000000001, 0.0012200000000000002, -0.017720000000000003, -0.00142, 0.99994, -0.01123, 0.01771, 0.011250000000000001, 0.9997800000000001]
P: [320.76961, 0.0, 189.75299, -44.23252, 0.0, 320.76961, 118.12179, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
```
Make sure you have calibrated the left camera on the left and the right camera on the right. See [stereo_calibration](http://wiki.ros.org/camera_calibration/Tutorials/StereoCalibration) for example.

Annaiyan, 2016-08-10:
many thanks for point out.The problem is that the camera is not in parallel before (since it is two independent camera).Now i make the lens parallel (verifed using the calibration images).the disparity is positive now but still the cloudXYZ is empty.i have recorded the new rosbag, here is the link 
https://www.dropbox.com/s/q2nz2zq6phmao9r/disparity_calibrated.bag?dl=0

Let me know,if you are getting the cloudXYZ.

matlabbe, 2016-08-10:
There is nothing in the bag:
```bash
$ rosbag info disparity_calibrated.bag 
path:    disparity_calibrated.bag
version: 2.0
size:    4.0 KB
```

Is `T` now valid (`>0`) in your disparity image?

Annaiyan, 2016-08-10:
sorry for the inconvenience.here is the correct link; https://www.dropbox.com/s/q2nz2zq6phmao9r/disparity_calibrated.bag?dl=0

yes, T is now valid

matlabbe, 2016-08-10:
[image]

It is working on my side... I noticed however that the scale of the point cloud changes, are the cameras are rigidly fixed together? maybe also a bad synchronization between the left and right images. If you can have a **hardware synchronization** between the cameras it would be better.

Annaiyan, 2016-08-12:
many thanks for testing the rosbag. the camera's are not rigidly fixed and also yet to get hardware synchronized.
But now i'm getting the cloudXYZ without problem.Most of the time while calibrating, left and right camera gets interchanged. This leads to all the issues (most of them).

closed 2016-08-30 by Annaiyan