introlab/rtabmap_ros #109: Save projected maps in the database for fast reloading?

matlabbe, 2016-08-09:
The problem is when we kill `rtabmap` and restart it with a database created previously, it takes a lot of time (if the database is big) to regenerate the global occupancy grid map (in particular `proj_map`) on relocalization, as all point clouds should be loaded and projected again to create each projected map. Save projected maps in database for fast map re-generation when restarting fresh the robot. We don't need to re-create the projected maps from scratch every time (from raw point clouds). This will save a lot of time when we relocalize in a previous map in the database. We could do it for laser occupancy grid maps too.

Note that if we want to change parameters related to grid map generation, the occupancy map should have to be regenerated offline (like using database viewer for that). However, this kind of parameters are generally tuned just one time for the lifetime of the robot. It is safe to consider that the grid maps generated will be always fixed.

matlabbe, 2016-08-31:
Fixed by merging `occupancy` feature branch to `master` branch.

closed 2016-08-31 by matlabbe

willdzeng, 2016-08-31:
Great work!