introlab/rtabmap_ros #110: CoreWrapper.cpp: optimization outside rtabmap node

KamalDSOberoi, 2016-08-11:
Hello,

I read in the documentation that if we want to optimize the graph outside rtabmap node, we should set publishTf = false and RGBD/optimizeIterations = 0 for rtabmap node. 

In CoreWrapper.cpp file, these two variables are defined. In [this](https://github.com/introlab/rtabmap_ros/blob/master/src/CoreWrapper.cpp#L453-L463) part of the code, when publishTf = true and optimizeIterations != 0, then the first if statement is executed. But for the execution of else if statement, publishTf needs to be true, which seems to be contradicting with the documentation. I doubt that the statement should be **else if(!publishTf)**, so that when publishTf is false (according to the documentation), we use map_optimizer node. Maybe I am wrong but could you please clear this doubt. 
Thanks in advance!!
Regards

matlabbe, 2016-08-11:
The 
```cpp
	else if(publishTf)
	{
		UWARN("Graph optimization is disabled (%s=0), the tf between frame \"%s\" and odometry frame will not be published. You can safely ignore this warning if you are using map_optimizer node.",
				Parameters::kOptimizerIterations().c_str(), mapFrameId_.c_str());
	}
```
is a warning. If you set iterations to 0 (so disabling graph optimization), you will get this warning only if you wanted TF (`publish_tf:=true`) at the same time. In other words, having at the same time `Optimize/Iterations=0` and `publish_tf:=true` makes no sense, the warning just reminds you...

KamalDSOberoi, 2016-08-11:
Got it. Thanks for the clarification.

closed 2016-08-11 by KamalDSOberoi