introlab/rtabmap_ros #112: Navigation using the octomap

Annaiyan, 2016-08-30:
hello,
I'm getting the octomap from the stereo camera.Now i want to use this map with move_base (following the tutorial) like below:
      

       <group ns="planner">
       <remap from="cloudXYZ" to="/planner_cloud"/>
      <remap from="map" to="/rtabmap/proj_map"/>
      <remap from="move_base_simple/goal" to="/planner_goal"/>   
     
      <node pkg="move_base" type="move_base" respawn="false" name="move_base" output="screen">
       <rosparam file="$(find rtabmap_ros)/launch/azimut3/config/costmap_common_params.yaml" command="load" ns="global_costmap" />
       <rosparam file="$(find rtabmap_ros)/launch/azimut3/config/costmap_common_params.yaml" command="load" ns="local_costmap" />
       <rosparam file="$(find rtabmap_ros)/launch/azimut3/config/local_costmap_params.yaml" command="load" />
       <rosparam file="$(find rtabmap_ros)/launch/azimut3/config/global_costmap_params.yaml" command="load" />
       <rosparam file="$(find rtabmap_ros)/launch/azimut3/config/base_local_planner_params.yaml" command="load" />
     </node>
</group>

and this is my rqt_graph
[image]
 
i'm using the parameters of the yaml file as it is in the package without changing.the stereo_camera publishes /left as frame id.
I have recorded a bag file from the drone and using that trying to get the cmd_vel .But the cmd_vel is published and there is no data in it.

how to get the command velocity using the rosbag file?

matlabbe, 2016-08-31:
You may have to send a goal to move_base will playing the rosbag, so it can plan a path and send cmd_vel. Note that navigation stuff is kinda hard to test with a rosbag, because cmd_vel will have an effect on how the robot would move, there is a feedback between control and mapping.

Annaiyan, 2016-09-01:
thanks for your reply.
The problem is that in my drone onboard computer there is no graphics (to save memory & computation).So i can't launch rviz directly.I tried to push all the rostopic to another computer using  wifi (multiple machines - ros) but it is too slow to receive the map.
Is there a way to send the navgoal without rviz?

matlabbe, 2016-09-01:
See [rostopic pub](http://wiki.ros.org/rostopic#rostopic_pub).

Annaiyan, 2016-09-12:
i have problems in using the move_base with the map. I can give the 2D Nav Goal from rviz but still the _planner/move_base/goal_ is empty.It didn't receive the goal and this is my tf tree

[image]

i tried checking like this rostopic echo /planner/move_base/goal and there is no data in it.
Since i'm using multiple machines, one for launching the stereo_mapping.launch and other is for visualization.Do you think that is the problem for not receiving the goal?

thanks in advance!

matlabbe, 2016-09-12:
When using multiple machines, I usually set [ROS_IP](http://wiki.ros.org/ROS/EnvironmentVariables#ROS_IP.2BAC8-ROS_HOSTNAME) environment variable to all machines, to avoid hostname resolution issue (that may cause one machine not receiving data from another one).

Your TF tree seems odd to me, there should not be a "loop" in the tree. `/map_to_odom` node that publishes `/base_footprint -> /map` should not exist.

cheers

Annaiyan, 2016-09-12:
ok thanks for your suggestion.
i have changed the tf now and it is without loop like below,
[image]
 Does the above tf tree is in the correct format?

matlabbe, 2016-09-12:
No, what is /map_to_odom used for? A standard TF would be:
/map -> /odom -> /base_footprint -> /base_link -> /left

Annaiyan, 2016-09-20:
this is my tf tree now.
[image].

trying to use the move_base for navigation in the octomap.In the TF the camera frame is moving as expected but the base_footprint and base_link is not moving.I'm receving the messages in
 /planner/move_base/local_costmap/footprint
/planner/move_base/global_costmap/footprint

why it is not moving ?what could be the issue?

thanks in advance.

matlabbe, 2016-09-20:
`/rtabmap/stereo_odometry` should publish `/odom` -> `/base_footprint`. `/base_link` -> `/left` should be a fixed transform.

Annaiyan, 2016-09-27:
now the _planner/move_base/global_costmap/footprint_ is moving in rviz but keep jumping to the origin .Here is my rosbag (https://www.dropbox.com/s/km2ta3k8qd49h96/Footprint_tf.bag?dl=0).

I'm publishing the tf using the static publisher like below,

`<node pkg="tf" type="static_transform_publisher" name="camera_base_link" args="0 0 0 -1.5707963267948966 0 -1.5707963267948966 base_link left 700" />
<node pkg="tf" type="static_transform_publisher" name="robot_base" args="0 0 0 $(arg optical_rotate) base_footprint base_link 700" />
<node pkg="tf" type="static_transform_publisher" name="odom_to_robot" args="0 0 0 $(arg optical_rotate) odom base_footprint 700" />`

why the base_footprint is jumping always in rviz?

matlabbe, 2016-09-27:
Having:
```xml
<node pkg="tf" type="static_transform_publisher" name="odom_to_robot" args="0 0 0 $(arg optical_rotate) odom base_footprint 700" />
```
is wrong. This transform is already published by `stereo_odometry`, it is why you have jumps, because two nodes are publishing the same transform (one always null). You should remove `odom_to_robot`.

cheers

Annaiyan, 2016-09-27:
ok.I just removed that node but the stereo_odometry is not publishing the transform.
My tf tree look like below 
[image]

In my launch file i have this two nodes
`<node pkg="tf" type="static_transform_publisher" name="camera_base_link" args="0 0 0 -1.5707963267948966 0 -1.5707963267948966 base_link left 700" />
<node pkg="tf" type="static_transform_publisher" name="robot_base" args="0 0 0 $(arg optical_rotate) base_footprint base_link 700" />`

Now there is no data in the footprint message.
 how to solve this issue of not publishing the tf from stereo_odometry?

matlabbe, 2016-09-27:
It is maybe that `stereo_odometry` is not started or odometry cannot be computed. Make sure that inputs of `stereo_odometry` are published. If yes and if there is no error/log/status shown on terminal from `stereo_odometry`, it could be a synchronization problem (the node assumes **Exact** time synchronization by default for stereo). What is the configuration of `stereo_odometry`?

Annaiyan, 2016-09-28:
now i'm getting the footrpint moving in rviz.This is my stereo_odometry configuration

  ```
 <node if="$(arg visual_odometry)" pkg="rtabmap_ros" type="stereo_odometry" name="stereo_odometry" output="screen" args="$(arg rtabmap_args)" launch-prefix="$(arg launch_prefix)">
      <remap from="left/image_rect"        to="/stereo_camera/left/image_rect_color"/>
      <remap from="right/image_rect"       to="/stereo_camera/right/image_rect"/>
      <remap from="left/camera_info"       to="/stereo_camera/left/camera_info"/>
      <remap from="right/camera_info"      to="/stereo_camera/right/camera_info"/>	  
      <param name="frame_id"                 type="string" value="base_footprint"/>
      <param name="wait_for_transform_duration"       type="double"   value="$(arg wait_for_transform)"/>
      <param name="approx_sync"             type="bool"   value="$(arg approximate_sync)"/>	  
      <param name="Odom/FillInfoData"       type="string" value="true"/> 
      <param name="Vis/InlierDistance"      type="string" value="10"/> <!-- when EstimationType=0 -->
      <param name="OdomF2M/MaxSize" type="string" value="1000" />
      <param name="grid_size"               type="double" value="50"/>     
      <param name="Odom/Guessmotion"        type="string" value="true" /> 
      <param name="Odom/Strategy" type="string" value="1" />
      <param name="Odom/FeatureType" type="string" value="4"/>
      <param name="publish_tf" value="true" />
      <param name="Vis/CorType" type="string" value="1" />
      <param name="Vis/MinInliers" type="string" value="5"/>
      <param name="Vis/RoiRatios" type="string" value="0.0 0.0 0.0 0.0"/>
      <param name="Vis/MaxDepth" type="string" value="0"/>
      <param name="OdomBow/NNDR" type="string" value="0.9"/>
      
   </node>   
```
  The topic  `planner/move_base/global_costmap/footprint` is moving without jumping but it stops at one point.Also i'm getting this warning after every 4th odmetry like this
```

[ WARN] [1475061606.375618772, 1473858425.285184440]: The openni_points observation buffer has not been updated for 61.33 seconds, and it should be updated every 0.50 seconds.
[ INFO] [1475061606.381672637, 1473858425.285184440]: Odom: quality=174, std dev=3.286485m, update time=0.024262s
[ INFO] [1475061606.417739868, 1473858425.325434878]: Odom: quality=170, std dev=3.109237m, update time=0.025594s
[ INFO] [1475061606.442699592, 1473858425.345575969]: Odom: quality=180, std dev=2.772344m, update time=0.019213s
[ INFO] [1475061606.482480228, 1473858425.385831379]: Odom: quality=178, std dev=2.723185m, update time=0.025994s
[ INFO] [1475061606.518713706, 1473858425.426106187]: Odom: quality=182, std dev=2.934196m, update time=0.027834s
[ WARN] [1475061606.526125740, 1473858425.436500219]: The openni_points observation buffer has not been updated for 61.48 seconds, and it should be updated every 0.50 seconds.
[ INFO] [1475061606.542721046, 1473858425.446764675]: Odom: quality=180, std dev=3.296050m, update time=0.018933s
[ INFO] [1475061606.589578569, 1473858425.497621327]: Odom: quality=182, std dev=3.534057m, update time=0.034576s
[ INFO] [1475061606.608694650, 1473858425.517739127]: Odom: quality=181, std dev=3.520127m, update time=0.017854s
[ INFO] [1475061606.644169664, 1473858425.548340468]: Odom: quality=183, std dev=4.868070m, update time=0.020347s
[ WARN] [1475061606.669609179, 1473858425.578533945]: The openni_points observation buffer has not been updated for 61.63 seconds, and it should be updated every 0.50 seconds.
[ INFO] [1475061606.676461023, 1473858425.578533945]: Odom: quality=177, std dev=5.500457m, update time=0.020005s
[ INFO] [1475061606.709921507, 1473858425.620305667]: Odom: quality=176, std dev=5.506516m, update time=0.019842s
[ INFO] [1475061606.746257897, 1473858425.651257003]: Odom: quality=172, std dev=3.175560m, update time=0.020995s
[ INFO] [1475061606.793821588, 1473858425.701667305]: Odom: quality=171, std dev=7.613109m, update time=0.020164s
[ WARN] [1475061606.811986632, 1473858425.721809704]: The openni_points observation buffer has not been updated for 61.77 seconds, and it should be updated every 0.50 seconds.
```
should i change the topic name in `local_costmap_params_3d.yaml` ? 
How the generate the `planner/cmd_vel` using the recorded rosbag ?

thanks in advance

matlabbe, 2016-09-28:
The error "The openni_points observation buffer has not been updated for 61.48 seconds, and it should be updated every 0.50 seconds." is coming from `move_base`, not `stereo_odometry`. `openni_points` should match the point cloud you are sending to local costmap (I assume it may be renamed `/planner_cloud`).

To generate /cmd_vel, move_needs to receive a goal, either a goal recorded in the bag or one sent while the bag is running. Note the it doesn't really make sense to test move_base cmd_vel with a rosbag, since move_base would mostly fail every time if the robot doesn't really follow the output cmd_vel.

Note that the odomerty standard deviation is very high (2 to 7 meters). What is your stereo baseline? With baseline around 10 cm, the standard deviation should be a couple of cm at most. Verify your camera calibration. You can also visualize the /planner_cloud in RVIZ with a 1 meter grid to see if the scale looks real.

If odometry stops at some point, it is maybe that odomety is lost. See "Lost Odometry" section on this [page](https://github.com/introlab/rtabmap/wiki/Kinect-mapping#lostodometry).

cheers

Annaiyan, 2016-09-28:
thanks for your suggestions.
I have baseline of 19 cm and here is my rosbag of `/planner_cloud` https://www.dropbox.com/s/18d83nr8clns2rh/planner_cloud.bag?dl=0

I tried to visualize but i cannot visualize the planner_cloud in rviz.
can you check my rosbag file ?

thanks in advance.

matlabbe, 2016-09-28:
The point clouds are mostly empty. You should have at least a more dense point cloud, otherwise obstacle detection stereo odometry won't work. This may be a **calibration issue**. If you move with the camera, images from both cameras should be taken **exactly** at the same time, generally a hardware clock is required between the two cameras for synchronization if you are not using a real stereo camera.

Follow carefully this tutorial: http://wiki.ros.org/camera_calibration/Tutorials/StereoCalibration, the resulting planner_cloud should be a lot more dense! unless you are looking a textureless wall.

Annaiyan, 2016-10-06:
i have a doubt regarding the local_costmap_params,
```
local_costmap:
  global_frame: odom
  robot_base_frame: base_footprint
  update_frequency: 2.0
  publish_frequency: 2.0
  static_map: true
  rolling_window: true
  width: 4.0
  height: 4.0
  resolution: 0.025
  origin_x: -2.0
  origin_y: -2.0

  observation_sources: point_cloud_sensor

  # assuming receiving a cloud from rtabmap_ros/obstacles_detection node
  point_cloud_sensor: {
    sensor_frame: base_footprint,
    data_type: PointCloud2, 
    topic: stereo_camera/points2, 
    expected_update_rate: 0.5, 
    marking: true, 
    clearing: true,
    min_obstacle_height: -99999.0,
    max_obstacle_height: 99999.0}

```
topic should be `stereo_camera/points2 `or `rtabmap_cloudmap`?

and in `move_base `node it shoule like (if it is `stereo_camera/points2`) this ?
```
<group ns="planner">       
      <remap from="stereo_camera/points2" to="/planner_cloud"/>

</group>
```

matlabbe, 2016-10-07:
If /planner_cloud is the obstacles_cloud output of obstacles_detection nodelet, then yes. To avoid multiple renaming, you could do also directly:
```
point_cloud_sensor: {
    sensor_frame: base_footprint,
    data_type: PointCloud2, 
    topic: /obstacles_cloud, 
    expected_update_rate: 0.5, 
    marking: true, 
    clearing: true,
    min_obstacle_height: -99999.0,
    max_obstacle_height: 99999.0}
```

`/rtabmap/cloud_map` should not be used for the obstacle layer of the costmaps.

Annaiyan, 2016-10-07:
ok. In the obstacle detection nodelet the remapping should be from /stereo_camera/points2 to planner_cloud like this 

```
node pkg="nodelet" type="nodelet" name="obstacles_detection" args="load rtabmap_ros/obstacles_detection stereo_nodelet">
<remap from="cloud" to="cloudXYZ"/>
<remap from="stereo_camera/points2" to="/planner_cloud"/>
<param name="frame_id" type="string" value="base_footprint"/>     
<param name="map_frame_id" type="string" value="map"/>
<param name="wait_for_transform" type="bool" value="true"/>
<param name="min_cluster_size" type="int" value="10"/>
<param name="max_obstacles_height" type="double" value="20"/>
</node>
```
or it should be `<remap from="obstacles_cloud" to="/planner_cloud"/>?

Thanks in advance
`

matlabbe, 2016-10-07:
If you look at the [Stereo Outdoor Navigation tutorial](http://wiki.ros.org/rtabmap_ros/Tutorials/StereoOutdoorNavigation), and the file [az3_mapping_robot_stereo_nav.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/azimut3/az3_mapping_robot_stereo_nav.launch), with my example above setting directly `/obstacles_cloud` to `point_cloud_sensor`:
```xml
<node pkg="nodelet" type="nodelet" name="obstacles_detection" args="load rtabmap_ros/obstacles_detection stereo_nodelet">
<remap from="cloud" to="cloudXYZ"/>
<remap from="obstacles_cloud" to="/obstacles_cloud"/> <!-- /obstacles_cloud is the topic of planner_cloud_sensor above -->
<param name="frame_id" type="string" value="base_footprint"/>     
<param name="map_frame_id" type="string" value="map"/>
<param name="wait_for_transform" type="bool" value="true"/>
<param name="min_cluster_size" type="int" value="10"/>
<param name="max_obstacles_height" type="double" value="20"/>
</node>
```
The leading `/` is to get the topic in the public namespace, in case you enclosed `move_base` or `obstacles_detection` under a group un the launch file.

Annaiyan, 2016-10-07:
many thanks for the quick response.
I tried the _stereo_outdoorA.bag_  with above configuration and i get the rosgraph like below

[image]

the obstacle_cloud is not generated in the graph, if i use the above config .

if you don mind , can you check it out with your rosbag - `stereo_outdoorA.bag` ?

matlabbe, 2016-10-11:
Not sure exactly which launch files you are using, but if I add `move_base` stuff at the end of [demo_stereo_outdoor.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/demo/demo_stereo_outdoor.launch):
```xml
   <remap from="openni_points" to="/stereo_camera/obstacles"/>
   <remap from="map" to="/rtabmap/proj_map"/>

   <node pkg="move_base" type="move_base" respawn="false" name="move_base" output="screen">
      <rosparam file="$(find rtabmap_ros)/launch/azimut3/config/costmap_common_params.yaml" command="load" ns="global_costmap" />
      <rosparam file="$(find rtabmap_ros)/launch/azimut3/config/costmap_common_params.yaml" command="load" ns="local_costmap" />
      <rosparam file="$(find rtabmap_ros)/launch/azimut3/config/local_costmap_params_3d.yaml" command="load" />
      <rosparam file="$(find rtabmap_ros)/launch/azimut3/config/global_costmap_params.yaml" command="load" ns="global_costmap"/>
      <rosparam file="$(find rtabmap_ros)/launch/azimut3/config/base_local_planner_params.yaml" command="load" />
   </node>

   <group ns="/stereo_camera" >   
      <!-- Create point cloud for the local costmap -->
      <node pkg="nodelet" type="nodelet" name="disparity2cloud" args="load rtabmap_ros/point_cloud_xyz stereo_nodelet">
         <remap from="disparity/image"       to="disparity"/>
         <remap from="disparity/camera_info" to="right/camera_info_throttle"/>
         <remap from="cloud"                 to="cloudXYZ"/>
         
         <param name="voxel_size" type="double" value="0.05"/>
         <param name="decimation" type="int" value="4"/>
         <param name="max_depth" type="double" value="4"/>
      </node>
      <node pkg="nodelet" type="nodelet" name="obstacles_detection" args="load rtabmap_ros/obstacles_detection stereo_nodelet">
         <remap from="cloud" to="cloudXYZ"/>
         <remap from="obstacles" to="obstacles"/>

         <param name="frame_id" type="string" value="base_footprint"/>      
         <param name="map_frame_id" type="string" value="map"/>
         <param name="wait_for_transform" type="bool" value="true"/>
         <param name="min_cluster_size" type="int" value="20"/>
         <param name="max_obstacles_height" type="double" value="0.0"/>
      </node>
   </group>
```

Then do:
```bash
$ roslaunch rtabmap_ros demo_stereo_outdoor.launch
$ rosbag play --clock stereo_outdoorA.bag
```

In RVIZ, showing `/stereo_camera/obstacles` (red points on the trees) and the global costmap:
[image]

The graph:
[image]

I think with this example you should be able to connect correctly the `obstacles` cloud to `move_base` local costmap.

cheers

Annaiyan, 2016-10-12:
I'm also using exactly the same launch file demo_stereo_outdoor.launch without changes.But i'm not getting the map frame.
 I do like this:
```
$ roslaunch rtabmap_ros demo_stereo_outdoor.launch
$ rosbag play --clock stereo_outdoorA.bag

```
Below is my rqt_tf_tree,
[image]

and rosgraph:

[image]

what could be the possible problems for not generating the map frame?

thanks in advance.

matlabbe, 2016-10-12:
The map frame is published after the rtabmap's callback is called for the first time. If the callback is not called, it is because one of the input topic is not published or they are badly synchronized. Make sure you did "--clock" when running the bag. Look also on the terminal for possible errors. Also, which version are you using? 0.11.8 or latest from source?

cheers

Annaiyan, 2016-10-12:
I'm using the latest version from source.Besides, i 'm using the stereo_outdoorA.bag which i assume, it is from well synchronized stereo camera.
In the terminal, there is no reasonable error except the name changes of the parameter.

matlabbe, 2016-10-12:
Right now if I do the demo:
```bash
$ roslaunch rtabmap_ros demo_stereo_outdoor.launch
$ rosbag play --clock stereo_outdoorA.bag
```
[image]

Yes the bag contains synchronized (exact stamps) stereo images:
```bash
$ rosbag info stereo_outdoorA.bag 
path:        stereo_outdoorA.bag
version:     2.0
duration:    3:55s (235s)
start:       Nov 11 2014 15:28:20.90 (1415737700.90)
end:         Nov 11 2014 15:32:16.65 (1415737936.65)
size:        668.3 MB
messages:    54083
compression: none [848/848 chunks]
types:       sensor_msgs/CameraInfo      [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/CompressedImage [8f7a12909da2c9d3332d540a0977563f]
             tf2_msgs/TFMessage          [94810edda583a504dfda3829e70d7eec]
topics:      /stereo_camera/left/camera_info_throttle              3741 msgs    : sensor_msgs/CameraInfo     
             /stereo_camera/left/image_raw_throttle/compressed     3753 msgs    : sensor_msgs/CompressedImage
             /stereo_camera/right/camera_info_throttle             3741 msgs    : sensor_msgs/CameraInfo     
             /stereo_camera/right/image_raw_throttle/compressed    3745 msgs    : sensor_msgs/CompressedImage
             /tf                                                  39103 msgs    : tf2_msgs/TFMessage          (2 connections)

```

Annaiyan, 2016-10-13:
Tried some times again to launch the demo_stereo_outdoor.launch file.
From my observation, the map frame is generating sometimes and most of the time it is not generating with the above rosbag file.
i don't know why this is happening!

Do you recommend to uninstall the rtabmap_ros as well as rtabmap and  install it again?

matlabbe, 2016-10-13:
Yeah you can try to do a clean unsintall and install again. Make sure you don't have a roscore still opened between the trials, so that TF time is correctly reset.

Annaiyan, 2016-10-14:
I reinstalled everything again but it is not much helpful.The same problem continues still.

matlabbe, 2016-10-16:
For info, what are your computer/os specifications?

Annaiyan, 2016-10-16:
Ubuntu 14.04, trusty and ros- indigo - `os spec`
Hardware spec ( command - lshw)
```
  *-core
       description: Motherboard
       product: 08RCYC
       vendor: Dell Inc.
       physical id: 0
       version: A00
           *-firmware
          description: BIOS
          vendor: Dell Inc.
          physical id: 0
          version: A06
          date: 05/01/2014
          size: 64KiB
          capacity: 13MiB
          capabilities: pci pnp upgrade shadowing cdboot bootselect edd int13floppy1200 int13floppy720 int13floppy2880 int5printscreen int9keyboard int14serial int17printer acpi usb smartbattery biosbootspecification netboot uefi
     *-cpu
          description: CPU
          product: Intel(R) Core(TM) i7-4600U CPU @ 2.10GHz
          vendor: Intel Corp.
          physical id: 42
          bus info: cpu@0
          version: Intel(R) Core(TM) i7-4600U CPU @ 2.10GHz
          slot: SOCKET 0
          size: 2700MHz
          capacity: 2700MHz
          width: 64 bits
          clock: 100MHz
          capabilities: x86-64 fpu fpu_exception wp vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm ida arat epb xsaveopt pln pts dtherm tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm cpufreq
          configuration: cores=2 enabledcores=2 threads=4
        *-cache:0
             description: L1 cache
             physical id: 43
             slot: CPU Internal L1
             size: 128KiB
             capacity: 128KiB
             capabilities: internal write-back
        *-cache:1
             description: L2 cache
             physical id: 44
             slot: CPU Internal L2
             size: 512KiB
             capacity: 512KiB
             capabilities: internal write-back unified
        *-cache:2
             description: L3 cache
             physical id: 45
             slot: CPU Internal L3
             size: 4MiB
             capacity: 4MiB
             capabilities: internal write-back unified
     *-memory
          description: System Memory
          physical id: 46
          slot: System board or motherboard
          size: 8GiB
        *-bank:0
             description: SODIMM DDR3 Synchronous 1600 MHz (0,6 ns)
             product: HMT451S6AFR8A-PB
             vendor: Hynix/Hyundai
             physical id: 0
             serial: F2AB8282
             slot: DIMM A
             size: 4GiB
             width: 64 bits
             clock: 1600MHz (0.6ns)
        *-bank:1
             description: SODIMM DDR3 Synchronous 1600 MHz (0,6 ns)
[... 305 lines cut ...]
             bus info: scsi@2:0.0.0
             logical name: /dev/cdrom
             logical name: /dev/sr0
             version: A101
             capabilities: removable audio cd-r cd-rw dvd dvd-r dvd-ram
             configuration: ansiversion=5 status=nodisc
  *-battery
       product: DELL Y6KM755
       vendor: LG
       physical id: 1
       version: 05/22/2015
       serial: 857D
       slot: Sys. Battery Bay
       capacity: 99900mWh
       configuration: voltage=11,1V
  *-power UNCLAIMED
       description: To Be Filled By O.E.M.
       product: To Be Filled By O.E.M.
       vendor: To Be Filled By O.E.M.
       physical id: 2
       version: To Be Filled By O.E.M.
       serial: To Be Filled By O.E.M.
       capacity: 32768mWh

```

matlabbe, 2016-10-16:
What is the output on the terminal of demo_stereo_outdoor.launch? (beginning with after you started the bag)

Annaiyan, 2016-10-16:
i did like below:
```
rosbag play --clock clutter_1.bag 
roslaunch rtabmap_ros demo_stereo_outdoor.launch rtabmap_args:="--delete_db_on_start" approximate_sync:=true
```
Output from the terminal:
```
SUMMARY
========

PARAMETERS
 * /octomap_server/frame_id: map
 * /octomap_server/latch: False
 * /octomap_server/resolution: 0.05
 * /octomap_server/sensor_model/max_range: 5.0
 * /rosdistro: indigo
 * /rosversion: 1.11.20
 * /rtabmap/rtabmap/Grid/DepthDecimation: 4
 * /rtabmap/rtabmap/Grid/FlatObstacleDetected: true
 * /rtabmap/rtabmap/Kp/DetectorStrategy: 0
 * /rtabmap/rtabmap/Kp/MaxDepth: 10
 * /rtabmap/rtabmap/Kp/MaxFeatures: 200
 * /rtabmap/rtabmap/Odom/GuessMotion: true
 * /rtabmap/rtabmap/Odom/Strategy: True
 * /rtabmap/rtabmap/Optimizer/Strategy: 0
 * /rtabmap/rtabmap/RGBD/LoopClosureReextractFeatures: true
 * /rtabmap/rtabmap/Rtabmap/TimeThr: 700
 * /rtabmap/rtabmap/SURF/HessianThreshold: 1000
 * /rtabmap/rtabmap/Stereo/OpticalFlow: false
 * /rtabmap/rtabmap/StereoBM/MinDisparity: 0
 * /rtabmap/rtabmap/StereoBM/NumDisparities: 256
 * /rtabmap/rtabmap/StereoBM/SpeckleRange: 30
 * /rtabmap/rtabmap/StereoBM/SpeckleWindowSize: 320
 * /rtabmap/rtabmap/Vis/CorType: True
 * /rtabmap/rtabmap/Vis/EstimationType: True
 * /rtabmap/rtabmap/Vis/MaxDepth: 0
 * /rtabmap/rtabmap/approx_sync: True
 * /rtabmap/rtabmap/frame_id: base_footprint
 * /rtabmap/rtabmap/queue_size: 30
 * /rtabmap/rtabmap/subscribe_depth: False
 * /rtabmap/rtabmap/subscribe_stereo: True
 * /rtabmap/stereo_odometry/GFTT/MinDistance: 10
 * /rtabmap/stereo_odometry/GFTT/QualityLevel: 0.00001
 * /rtabmap/stereo_odometry/Odom/FillInfoData: true
 * /rtabmap/stereo_odometry/Odom/MaxDepth: 0
 * /rtabmap/stereo_odometry/Odom/Strategy: 1
 * /rtabmap/stereo_odometry/OdomF2M/MaxSize: 1000
 * /rtabmap/stereo_odometry/approx_sync: True
 * /rtabmap/stereo_odometry/frame_id: base_footprint
 * /rtabmap/stereo_odometry/odom_frame_id: odom
 * /rtabmap/stereo_odometry/odom_sensor_sync: True
 * /rtabmap/stereo_odometry/wait_for_transform_duration: 2.0
 * /stereo_camera/disparity2cloud/decimation: 4
 * /stereo_camera/disparity2cloud/max_depth: 4.0
 * /stereo_camera/disparity2cloud/voxel_size: 0.05
 * /stereo_camera/obstacles_detection/frame_id: base_footprint
 * /stereo_camera/obstacles_detection/map_frame_id: map
 * /stereo_camera/obstacles_detection/max_obstacles_height: 0.0
 * /stereo_camera/obstacles_detection/min_cluster_size: 20

NODES
  /rtabmap/
    rtabmap (rtabmap_ros/rtabmap)
    stereo_odometry (rtabmap_ros/stereo_odometry)
  /stereo_camera/
    disparity2cloud (nodelet/nodelet)
    obstacles_detection (nodelet/nodelet)
    stereo_nodelet (nodelet/nodelet)
  /
    camera_base_link (tf/static_transform_publisher)
[... 675 lines cut ...]
[ INFO] [1476652020.617991347, 1475586101.284071172]: Odom: quality=142, std dev=1.079134m, update time=0.018054s
[ INFO] [1476652020.660595323, 1475586101.324330510]: Odom: quality=141, std dev=0.985473m, update time=0.019085s
[ INFO] [1476652020.686887916, 1475586101.354521175]: Odom: quality=133, std dev=1.172335m, update time=0.018374s
[ INFO] [1476652020.718469027, 1475586101.384744242]: Odom: quality=142, std dev=1.337198m, update time=0.018050s
[ INFO] [1476652020.755919148, 1475586101.425858813]: Odom: quality=145, std dev=1.138644m, update time=0.019558s
[ INFO] [1476652020.800100849, 1475586101.467362889]: Odom: quality=129, std dev=1.396659m, update time=0.021236s
[ INFO] [1476652020.824270098, 1475586101.487501262]: Odom: quality=128, std dev=0.960710m, update time=0.020910s
[ INFO] [1476652020.856763958, 1475586101.527989659]: Odom: quality=134, std dev=0.825264m, update time=0.021377s
[ INFO] [1476652020.894822633, 1475586101.558178408]: Odom: quality=130, std dev=0.797045m, update time=0.019280s
[ INFO] [1476652020.923762447, 1475586101.588364468]: Odom: quality=127, std dev=1.034502m, update time=0.023060s
[ INFO] [1476652020.965246677, 1475586101.628563462]: Odom: quality=134, std dev=1.060976m, update time=0.021141s
[ INFO] [1476652020.988042583, 1475586101.658939924]: Odom: quality=129, std dev=0.948301m, update time=0.020620s
[ INFO] [1476652021.022778544, 1475586101.689132565]: Odom: quality=130, std dev=1.411086m, update time=0.020158s
[ INFO] [1476652021.078451552, 1475586101.748457989]: Assembled 0 obstacle and 0 ground clouds (596 points, 0.000028s)
[ INFO] [1476652021.078631607, 1475586101.748457989]: rtabmap (23): Rate=1.00s, Limit=0.700s, RTAB-Map=0.0637s, Maps update=0.0002s pub=0.0002s (local map=22, WM=22)
[ INFO] [1476652021.099557141, 1475586101.768578631]: Odom: quality=131, std dev=1.253508m, update time=0.040107s
[ INFO] [1476652021.128730001, 1475586101.798753684]: Odom: quality=135, std dev=0.964704m, update time=0.028399s
[ INFO] [1476652021.156564393, 1475586101.818869267]: Odom: quality=129, std dev=1.063098m, update time=0.027076s
[ INFO] [1476652021.185727283, 1475586101.849793421]: Odom: quality=125, std dev=1.125140m, update time=0.020331s
[ INFO] [1476652021.217506333, 1475586101.880029090]: Odom: quality=122, std dev=1.164462m, update time=0.017529s
[ INFO] [1476652021.257938543, 1475586101.920259975]: Odom: quality=122, std dev=0.991640m, update time=0.020143s
[ INFO] [1476652021.293861737, 1475586101.960505956]: Odom: quality=106, std dev=2.041096m, update time=0.018359s
[ INFO] [1476652021.329062374, 1475586101.991453315]: Odom: quality=113, std dev=1.237105m, update time=0.018831s
[ INFO] [1476652021.359445880, 1475586102.021831964]: Odom: quality=119, std dev=0.930878m, update time=0.019144s
```

matlabbe, 2016-10-17:
Looks ok for me, we can see odometry and rtabmap working. The frame `/map -> /odom` should be published. Well RVIZ died for unknown reason though.

matlabbe, 2016-10-17:
Well you have an `octomap_server` running, make sure if it publishes `/map` frame it doesn't create a conflict with the /map transform published by rtabmap.

Annaiyan, 2016-10-18:
Finally, it started working :).
In the launch file, i have attached the stereo_nodelet and move_base node at the end initially.Now i removed both the nodes and tried.I'm getting the map frame generated!
I think the nodes should be placed in the right place.
Right now,
Nodes are in this order,
1.Stereo_image_proc
2.Stereo_nodelet (obstacle detection)
3.Stereo_odometry
4.rtabmap
5.rviz.

cheers,

matlabbe, 2016-10-18:
Normally, the order that nodes are launched in a launch file is not determined. I'll close this issue, though you can re-open it if you have the problem again.

cheers

closed 2016-10-18 by matlabbe