introlab/rtabmap_ros #114: Add nh.getParam() for GuiWrapper.cpp to parse parameter in launch file

willdzeng, 2016-09-02:
Hi Mathieu,

I noticed that the GuiWrapper.cpp doesn't have the parser for rtabmap GUI parameters, something like in the [CoreWrapper.cpp](https://github.com/introlab/rtabmap_ros/blob/master/src/CoreWrapper.cpp#L311).
For example,parameters like the following in rgbd_gui.ini
```
General\octomap=false
General\octomap_depth=16
General\octomap_ground_is_obstacle=true
```

Do you think it's good to have it? I can help you do it if you want.

matlabbe, 2016-09-03:
Hi Di,

The GUI parameters are hard-coded, in contrast to CORE parameters, so it may take some efforts to convert them all (see for example [Preferences::resetSettings()](https://github.com/introlab/rtabmap/blob/4bd9e4f84751b212d7867efb0939479c59998f2d/guilib/src/PreferencesDialog.cpp#L1171-L1438) where they are all listed). The mechanism to manage them is by saving/loading configuration files (*.ini), similar to RVIZ. When you hit "ctrl-s", the parameters will be loaded the next session. Like RVIZ, the config file can be set like this:
```bash
$ rosrun rtabmap rtabmapviz -d gui_params.ini
```
or if rtabmap node was started in `rtabmap` namespace (e.g., with [rtabmap.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/rtabmap.launch)):
```bash
$ export ROS_NAMESPACE=rtabmap && rosrun rtabmap rtabmapviz _frame_id:=camera_link -d gui_params.ini
```

Note for the octomap pameters, those ones in the GUI are for the GUI only, not for the `octomap_***` topics sent by rtabmap node. The octomap parameters for `rtabmap` node are [here](https://github.com/introlab/rtabmap_ros/blob/8f5c50b3f03850b7b2880b51172290ad90c7c510/src/MapsManager.cpp#L146-L147) (the ROS wiki needs an update to list them).

`General\octomap_ground_is_obstacle=true` (or [octomap_ground_is_obstacle](https://github.com/introlab/rtabmap_ros/blob/8f5c50b3f03850b7b2880b51172290ad90c7c510/src/MapsManager.cpp#L272) for `rtabmap`) doesn't exist anymore, see "[Grid/3DGroundIsObstacle](https://github.com/introlab/rtabmap/blob/4bd9e4f84751b212d7867efb0939479c59998f2d/corelib/include/rtabmap/core/Parameters.h#L490)". I recommend to use RVIZ instead of rtabmapviz to visualize the OctoMap.

cheers,
Mathieu

willdzeng, 2016-09-06:
Hi Mathieu,

OK, I got you.

I noticed [here](https://github.com/introlab/rtabmap_ros/blob/8f5c50b3f03850b7b2880b51172290ad90c7c510/src/MapsManager.cpp#L946) if there is no subscriber of octomap, it won't calculate the octomap. So we don't really need a parameter to enable and disable octomap. Is it right?

closed 2016-09-06 by willdzeng

matlabbe, 2016-09-08:
Yes, for convenience. For `rtabmapviz` it is different, as the octomap is built in `rtabmapviz` so a parameter is required (`rtabmapviz` doesn't subscribe to octomap from `rtabmap` node).