introlab/rtabmap_ros #115: The implementation of obstacle_detection

Severn-Vergil, 2016-09-12:
Hi, Mathieu
I try to figure out how obstacle_detection of rtabmap_ros works, and finally I just find that the ground and obstacle indices of the pointcloud are assigned by the grid_.segmentcloud in obstacle_detection.cpp, and I want to know the principle of dividing ground and obstacle, how can I check that part from rtabmap?
Thank you in advance!

matlabbe, 2016-09-12:
See [OccupancyGrid.hpp](https://github.com/introlab/rtabmap/blob/master/corelib/include/rtabmap/core/impl/OccupancyGrid.hpp).

cheers

Severn-Vergil, 2016-09-13:
Thank you, Mathieu
The normal filter check the normal vector of all points in the pointcloud, and calculate the angle with vector (0, 0, 1), which means that if there are some "smooth" obstacles on the ground, they may be divided into the part of ground as well, are they? I think it is enough for current usage.

matlabbe, 2016-09-13:
Yes, smooth obstacles will be segmented as ground. A carpet for example. The voxel size, sensor's resolution and sensor's noise can also determine what is the smallest object that can be detected. 

cheers

closed 2016-09-27 by Severn-Vergil