introlab/rtabmap_ros #116: Installation fails on Kinetic

chrisspen, 2016-09-19:
Attempting to install via package on Ubuntu 16.04/ROS Kinetic via:

    sudo apt-get install ros-kinetic-rtabmap-ros

fails with the errors:

    $ sudo apt-get install ros-kinetic-rtabmap-ros
    Reading package lists... Done
    Building dependency tree       
    Reading state information... Done
    ros-kinetic-rtabmap-ros is already the newest version (0.11.8-0xenial-20160724-131632-0700).
    0 upgraded, 0 newly installed, 0 to remove and 20 not upgraded.
    13 not fully installed or removed.
    After this operation, 0 B of additional disk space will be used.
    Do you want to continue? [Y/n] y
    Setting up libopenni0 (1.5.4.0-14) ...
    mv: cannot stat '/var/lib/ni/licenses.xml.dpkg-old': No such file or directory
    dpkg: error processing package libopenni0 (--configure):
     subprocess installed post-installation script returned error exit status 1
    dpkg: dependency problems prevent configuration of libopenni-dev:
     libopenni-dev depends on libopenni0 (= 1.5.4.0-14); however:
      Package libopenni0 is not configured yet.

    dpkg: error processing package libopenni-dev (--configure):
     dependency problems - leaving unconfigured
    dpkg: dependency problems prevent configuration of openni-utils:
     openni-utils depends on libopenni0; however:
      Package libopenni0 is not configured yet.No apport report written because the error message indicates its a followup error from a previous failure.

    dpkg: error processing package openni-utils (--configure):
     dependency problems - leaving unconfigured
    dpkg: dependency problems prevent configuration of libopenni-sensor-pointclouds0:
     libopenni-sensor-pointclouds0 depends on libopenni0; however:
      Package libopenni0 is not configured yet.
     libopenni-sensor-pointclouds0 depends on openni-utils; however:
      Package openni-utils is not configured yet.

    dpkg: error processing package libopenni-sensor-pointclouds0 (--configure):
    No apport report written because the error message indicates its a followup error from a previous failure.
                                                                                                              No apport report written because MaxReports is reached already
                                                                                                                                                                             dependency problems - leaving unconfigured
    dpkg: dependency problems prevent configuration of libpcl-io1.7:amd64:
     libpcl-io1.7:amd64 depends on libopenni0; however:
      Package libopenni0 is not configured yet.

    dpkg: error processing package libpcl-io1.7:amd64 (--configure):
     dependency problems - leaving unconfigured
    No apport report written because MaxReports is reached already
                                                                  dpkg: dependency problems prevent configuration of libpcl-visualization1.7:amd64:
     libpcl-visualization1.7:amd64 depends on libpcl-io1.7; however:
      Package libpcl-io1.7:amd64 is not configured yet.

    dpkg: error processing package libpcl-visualization1.7:amd64 (--configure):
     dependency problems - leaving unconfigured
    No apport report written because MaxReports is reached already
                                                                  dpkg: dependency problems prevent configuration of libpcl1.7:
     libpcl1.7 depends on libpcl-io1.7; however:
      Package libpcl-io1.7:amd64 is not configured yet.
     libpcl1.7 depends on libpcl-visualization1.7; however:
      Package libpcl-visualization1.7:amd64 is not configured yet.

    dpkg: error processing package libpcl1.7 (--configure):
     dependency problems - leaving unconfigured
    No apport report written because MaxReports is reached already
                                                                  dpkg: dependency problems prevent configuration of libpcl-dev:
     libpcl-dev depends on libopenni-dev; however:
      Package libopenni-dev is not configured yet.
     libpcl-dev depends on libpcl1.7 (= 1.7.2-14build1); however:
      Package libpcl1.7 is not configured yet.

    dpkg: error processing package libpcl-dev (--configure):
     dependency problems - leaving unconfigured
    No apport report written because MaxReports is reached already
                                                                  dpkg: dependency problems prevent configuration of ros-kinetic-pcl-conversions:
     ros-kinetic-pcl-conversions depends on libpcl-dev; however:
      Package libpcl-dev is not configured yet.
     ros-kinetic-pcl-conversions depends on libpcl1.7; however:
      Package libpcl1.7 is not configured yet.

    dpkg: error processing package ros-kinetic-pcl-conversions (--configure):
     dependency problems - leaving unconfigured
    No apport report written because MaxReports is reached already
                                                                  dpkg: dependency problems prevent configuration of ros-kinetic-pcl-ros:
     ros-kinetic-pcl-ros depends on libpcl-io1.7; however:
      Package libpcl-io1.7:amd64 is not configured yet.
     ros-kinetic-pcl-ros depends on libpcl-dev; however:
      Package libpcl-dev is not configured yet.
     ros-kinetic-pcl-ros depends on ros-kinetic-pcl-conversions; however:
      Package ros-kinetic-pcl-conversions is not configured yet.

    dpkg: error processing package ros-kinetic-pcl-ros (--configure):
     dependency problems - leaving unconfigured
    No apport report written because MaxReports is reached already
                                                                  dpkg: dependency problems prevent configuration of ros-kinetic-costmap-2d:
     ros-kinetic-costmap-2d depends on ros-kinetic-pcl-conversions; however:
      Package ros-kinetic-pcl-conversions is not configured yet.
     ros-kinetic-costmap-2d depends on ros-kinetic-pcl-ros; however:
      Package ros-kinetic-pcl-ros is not configured yet.

    dpkg: error processing package ros-kinetic-costmap-2d (--configure):
     dependency problems - leaving unconfigured
    No apport report written because MaxReports is reached already
                                                                  dpkg: dependency problems prevent configuration of ros-kinetic-rtabmap:
     ros-kinetic-rtabmap depends on libpcl-io1.7; however:
      Package libpcl-io1.7:amd64 is not configured yet.
     ros-kinetic-rtabmap depends on libpcl-visualization1.7; however:
      Package libpcl-visualization1.7:amd64 is not configured yet.
     ros-kinetic-rtabmap depends on libopenni-dev; however:
      Package libopenni-dev is not configured yet.
     ros-kinetic-rtabmap depends on libpcl-dev; however:
      Package libpcl-dev is not configured yet.

    dpkg: error processing package ros-kinetic-rtabmap (--configure):
     dependency problems - leaving unconfigured
    No apport report written because MaxReports is reached already
                                                                  dpkg: dependency problems prevent configuration of ros-kinetic-rtabmap-ros:
     ros-kinetic-rtabmap-ros depends on ros-kinetic-rtabmap; however:
      Package ros-kinetic-rtabmap is not configured yet.
     ros-kinetic-rtabmap-ros depends on ros-kinetic-costmap-2d; however:
      Package ros-kinetic-costmap-2d is not configured yet.
     ros-kinetic-rtabmap-ros depends on ros-kinetic-pcl-conversions; however:
      Package ros-kinetic-pcl-conversions is not configured yet.
     ros-kinetic-rtabmap-ros depends on ros-kinetic-pcl-ros; however:
      Package ros-kinetic-pcl-ros is not configured yet.

    dpkg: error processing package ros-kinetic-rtabmap-ros (--configure):
     dependency problems - leaving unconfigured
    No apport report written because MaxReports is reached already
                                                                  Processing triggers for libc-bin (2.23-0ubuntu3) ...
    Errors were encountered while processing:
     libopenni0
     libopenni-dev
     openni-utils
     libopenni-sensor-pointclouds0
     libpcl-io1.7:amd64
     libpcl-visualization1.7:amd64
     libpcl1.7
     libpcl-dev
     ros-kinetic-pcl-conversions
     ros-kinetic-pcl-ros
     ros-kinetic-costmap-2d
     ros-kinetic-rtabmap
     ros-kinetic-rtabmap-ros
    E: Sub-process /usr/bin/dpkg returned an error code (1)

chrisspen, 2016-09-19:
Nevermind. Searching around for similar errors shows that this is likely caused by old libopenni files from 14.04.

I fixed this by running:

    sudo apt-get purge libopenni*

and then reinstalling the rtabmap.

closed 2016-09-19 by chrisspen

huangzonghao, 2018-06-09:
thanks buddy!