introlab/rtabmap_ros #117: There are a lot of issues or errors when run the cmake of the source for the indigo version in the last two steps

78226415, 2016-09-27:
Hi everyone,
     I have caught a problem when  run the cmake of the source for the indigo version, the command like:

$ cd ~
$ git clone https://github.com/introlab/rtabmap.git rtabmap
$ cd rtabmap/build
$ cmake ..  [<---double dots included]

it is one of the last two steps, all steps above this step are built successful. My environment is ubuntu14.04 LTS, ros indigo.
    The error message are:
/usr/include/ni/XnCppWrapper.h: At global scope:
/usr/include/ni/XnCppWrapper.h:9717:9: error: ‘XnStatus’ does not name a type
  inline XnStatus NodeInfoList::FilterList(Context& context, Query& query)
         ^
/usr/include/ni/XnCppWrapper.h: In member function ‘void xn::ProductionNode::GetContext(xn::Context&) const’:
/usr/include/ni/XnCppWrapper.h:9724:66: error: ‘xnGetRefContextFromNodeHandle’ was not declared in this scope
   context.TakeOwnership(xnGetRefContextFromNodeHandle(GetHandle()));
                                                                  ^
/usr/include/ni/XnCppWrapper.h: In member function ‘xn::Context xn::ProductionNode::GetContext() const’:
/usr/include/ni/XnCppWrapper.h:9729:66: error: ‘xnGetRefContextFromNodeHandle’ was not declared in this scope
   XnContext* pContext = xnGetRefContextFromNodeHandle(GetHandle());
                                                                  ^
/usr/include/ni/XnCppWrapper.h:9731:28: error: ‘xnContextRelease’ was not declared in this scope
   xnContextRelease(pContext);
                            ^
/usr/include/ni/XnCppWrapper.h: In member function ‘xn::NodeInfoList& xn::NodeInfo::GetNeededNodes() const’:
/usr/include/ni/XnCppWrapper.h:9739:60: error: ‘xnNodeInfoGetNeededNodes’ was not declared in this scope
    XnNodeInfoList* pList = xnNodeInfoGetNeededNodes(m_pInfo);
                                                            ^
/usr/include/ni/XnCppWrapper.h: In member function ‘void xn::NodeInfo::SetUnderlyingObject(XnNodeInfo*)’:
/usr/include/ni/XnCppWrapper.h:9753:3: error: ‘m_bOwnerOfNode’ was not declared in this scope
   m_bOwnerOfNode = FALSE;
   ^
/usr/include/ni/XnCppWrapper.h: At global scope:
/usr/include/ni/XnCppWrapper.h:9758:9: error: ‘XnBool’ does not name a type
  inline XnBool FrameSyncCapability::CanFrameSyncWith(Generator& other) const
         ^
/usr/include/ni/XnCppWrapper.h:9763:9: error: ‘XnStatus’ does not name a type
  inline XnStatus FrameSyncCapability::FrameSyncWith(Generator& other)
         ^
/usr/include/ni/XnCppWrapper.h:9768:9: error: ‘XnStatus’ does not name a type
  inline XnStatus FrameSyncCapability::StopFrameSyncWith(Generator& other)
         ^
/usr/include/ni/XnCppWrapper.h:9773:9: error: ‘XnBool’ does not name a type
  inline XnBool FrameSyncCapability::IsFrameSyncedWith(Generator& other) const
         ^
/usr/include/ni/XnCppWrapper.h:9778:9: error: ‘XnStatus’ does not name a type
  inline XnStatus NodeInfo::GetInstance(ProductionNode& node) const
         ^
/usr/include/ni/XnCppWrapper.h:9800:9: error: ‘XnStatus’ does not name a type
  inline XnStatus Device::Create(Context& context, Query* pQuery/*=NULL*/, EnumerationErrors* pErrors/*=NULL*/)
         ^
/usr/include/ni/XnCppWrapper.h:9809:9: error: ‘XnStatus’ does not name a type
  inline XnStatus Recorder::Create(Context& context, const XnChar* strFormatName /*= NULL*/)
         ^
/usr/include/ni/XnCppWrapper.h:9818:9: error: ‘XnStatus’ does not name a type
  inline XnStatus Player::Create(Context& context, const XnChar* strFormatName)
         ^
/usr/include/ni/XnCppWrapper.h:9827:9: error: ‘XnStatus’ does not name a type
  inline XnStatus DepthGenerator::Create(Context& context, Query* pQuery/*=NULL*/, EnumerationErrors* pErrors/*=NULL*/)
         ^
/usr/include/ni/XnCppWrapper.h:9836:9: error: ‘XnStatus’ does not name a type
  inline XnStatus MockDepthGenerator::Create(Context& context, const XnChar* strName /* = NULL */)
         ^
/usr/include/ni/XnCppWrapper.h:9845:9: error: ‘XnStatus’ does not name a type
  inline XnStatus MockDepthGenerator::CreateBasedOn(DepthGenerator& other, const XnChar* strName /* = NULL */)
         ^
/usr/include/ni/XnCppWrapper.h:9856:9: error: ‘XnStatus’ does not name a type
  inline XnStatus ImageGenerator::Create(Context& context, Query* pQuery/*=NULL*/, EnumerationErrors* pErrors/*=NULL*/)
         ^
[... 280 lines cut ...]
                          ^
/usr/include/pcl-1.7/pcl/io/openni_grabber.h:459:39: error: ‘const XnMapOutputMode’ has no member named ‘nFPS’
           else if (mode1.nFPS < mode2.nFPS)
                                       ^
In file included from /usr/include/pcl-1.7/pcl/io/oni_grabber.h:48:0,
                 from /home/liguangxi/rtabmap/corelib/src/CameraRGBD.cpp:45:
/usr/include/pcl-1.7/pcl/io/openni_camera/openni_device_oni.h: In member function ‘bool openni_wrapper::DeviceONI::hasDataLeft()’:
/usr/include/pcl-1.7/pcl/io/openni_camera/openni_device_oni.h:89:24: error: ‘class xn::Player’ has no member named ‘IsEOF’
       return (!player_.IsEOF ());
                        ^
In file included from /home/liguangxi/rtabmap/corelib/src/../include/rtabmap/core/CameraRGBD.h:42:0,
                 from /home/liguangxi/rtabmap/corelib/src/CameraRGBD.cpp:28:
/usr/include/pcl-1.7/pcl/io/openni_camera/openni_image.h: In member function ‘unsigned int openni_wrapper::Image::getHeight() const’:
/usr/include/pcl-1.7/pcl/io/openni_camera/openni_image.h:188:3: warning: control reaches end of non-void function [-Wreturn-type]
   }
   ^
/usr/include/pcl-1.7/pcl/io/openni_camera/openni_image.h: In member function ‘unsigned int openni_wrapper::Image::getWidth() const’:
/usr/include/pcl-1.7/pcl/io/openni_camera/openni_image.h:182:3: warning: control reaches end of non-void function [-Wreturn-type]
   }
   ^
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/CameraRGBD.cpp.o] Error 1
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
make: *** [all] Error 2

Could some one help me? thanks.

willdzeng, 2016-09-27:
Did you installed openNI?

matlabbe, 2016-09-27:
Similar posts:
 * https://github.com/CognitiveRobotics/organized_segmentation_tools/issues/1
 * http://official-rtab-map-forum.67519.x6.nabble.com/Compilitaion-td955.html
It is a OpenNI vs PCL problem.

78226415, 2016-09-28:
use the GTSAM3.2.0, not GTSAM3.2.1, will get it, thank you very much!

closed 2016-09-28 by 78226415