introlab/rtabmap_ros #118: Could not find a package configuration file provided by "RTABMap"   (requested version 0.11.10)

78226415, 2016-09-28:
Hi everyone,
     I have a building problem when build the last one step for the source, in the Install RTAB-Map ros-pkg in your src folder of your Catkin workspace. The command lines are:
$ cd ~/catkin_ws
$ git clone https://github.com/introlab/rtabmap_ros.git src/rtabmap_ros
$ catkin_make

then an error is happen, the message is :
`-- Using these message generators: gencpp;genlisp;genpy
CMake Error at rtabmap_ros/CMakeLists.txt:21 (find_package):
  By not providing "FindRTABMap.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "RTABMap", but
  CMake did not find one.

  Could not find a package configuration file provided by "RTABMap"
  (requested version 0.11.10) with any of the following names:

    RTABMapConfig.cmake
    rtabmap-config.cmake

  Add the installation prefix of "RTABMap" to CMAKE_PREFIX_PATH or set
  "RTABMap_DIR" to a directory containing one of the above files.  If
  "RTABMap" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/liguangxi/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/liguangxi/catkin_ws/build/CMakeFiles/CMakeError.log".
make: *** [cmake_check_build_system] Error 1
`
it seems can not find the RTABMap package, but i have made it installed at last two or three steps in the whole installation.
Would some one help me? thanks

matlabbe, 2016-09-28:
Hi,

Where rtabmap libraries are installed (step 2 from [build-from-source](https://github.com/introlab/rtabmap_ros#build-from-source))? It should be installed in your catkin workspace (`~/catkin_ws/devel`) or in `/usr/local`. CMake should be able to find the installed `RTABMapConfig.cmake` file in one of those paths (note that if installed in `~/catkin_ws/devel`, make sure you did `$ source ~/catkin_ws/devel/setup.bash`).

cheers

78226415, 2016-09-29:
I used this command line instand of it :
$ sudo apt-get install ros-indigo-rtabmap-ros

thanks all the same!

closed 2016-09-29 by 78226415

alex4men, 2018-05-21:
> Where rtabmap libraries are installed (step 2 from build-from-source)? It should be installed in your catkin workspace (~/catkin_ws/devel) or in /usr/local

But step 2 tells to install it to the `~/rtabmap directory`.
I faced the same error as @78226415 when tried to follow the instruction of building from source.
I fixed the problem by the command: `export RTABMap_DIR=~/rtabmap/build/`.
But maybe the `cmake` option `-DCMAKE_INSTALL_PREFIX=~/catkin_ws/devel` could help either.

@matlabbe it would be cool if you could revise this step in the instruction.

Thanks!

matlabbe, 2018-05-21:
Hi,

With:
```bash
$ cd ~
$ git clone https://github.com/introlab/rtabmap.git rtabmap
$ cd rtabmap/build
$ cmake .. 
$ make
$ sudo make install
```
the last line "`sudo make install`" will install rtabmap in `/usr/local` directory. If your **PATH** has `/usr/local/bin` in it, CMake should be able to find RTABMapConfig.cmake installed in `/usr/local/lib/rtabmap-0.XX` without the need to export RTABMap_DIR. Same thing if rtabmap is installed in your catkin devel space (e.g., `~/catkin_ws/devel`), cmake should be able to find it if you did source `~/catkin_ws/devel/setup.bash` (which puts `~/catkin_ws/devel` in the **PATH**).

If you don't install rtabmap (not doing the last line above), RTABMap_DIR should be indeed set to rtabmap/build directory so that CMake can find it.

cheers,
Mathieu

alex4men, 2018-05-21:
Ooops, sorry, I really missed `sudo make install` command 🙃
I appreciate your help!