introlab/rtabmap_ros #121: Build error

QinZiwen, 2016-10-01:
I use Ubuntu14.04 , ros indigo and kinect2, but when I build, some error show me:
```
$ catkin_make
-- +++ processing catkin package: 'rtabmap_ros'
-- ==> add_subdirectory(rtabmap_ros)
-- Using these message generators: gencpp;genlisp;genpy
CMake Error at rtabmap_ros/CMakeLists.txt:21 (find_package):
  Could not find a configuration file for package "RTABMap" that is
  compatible with requested version "0.11.10".

  The following configuration files were considered but not accepted:

    /opt/ros/indigo/lib/x86_64-linux-gnu/rtabmap-0.11/RTABMapConfig.cmake, version: 0.11.7

-- Configuring incomplete, errors occurred!
See also "/home/turtlebot/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/turtlebot/catkin_ws/build/CMakeFiles/CMakeError.log".
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
```
help me, Thank you very much!

matlabbe, 2016-10-01:
You should update (`$ git pull`) your `rtabmap` source repository and `make install` it again.

QinZiwen, 2016-10-02:
@matlabbe Thank you. I " git pull my rtabmap source", but when I "make -j4", some error show me:
```
[  9%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/vertigo/g2o/edge_se3Switchable.cpp.o
[ 10%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/vertigo/g2o/edge_switchPrior.cpp.o
[ 10%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/vertigo/g2o/types_g2o_robust.cpp.o
[ 11%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/vertigo/g2o/vertex_switchLinear.cpp.o
In file included from /usr/local/include/gtsam/geometry/Pose2.h:24:0,
                 from /home/turtlebot/Project_slam/rtabmap/corelib/src/OptimizerGTSAM.cpp:39:
/usr/local/include/gtsam/base/Matrix.h: In function ‘void gtsam::inplace_QR(MATRIX&)’:
/usr/local/include/gtsam/base/Matrix.h:316:71: error: expected ‘;’ before ‘::’ token
   Eigen::internal::householder_qr_inplace_blocked<MATRIX, HCoeffsType>::run(A, hCoeffs, 48, temp.data());
                                                                       ^
In file included from /usr/local/include/gtsam/geometry/Pose2.h:26:0,
                 from /home/turtlebot/Project_slam/rtabmap/corelib/src/OptimizerGTSAM.cpp:39:
/usr/local/include/gtsam/geometry/Point2.h: In static member function ‘static gtsam::Vector gtsam::Point2::Logmap(const gtsam::Point2&)’:
/usr/local/include/gtsam/geometry/Point2.h:175:86: error: could not convert ‘Eigen::DenseBase<Derived>::operator<<(const Scalar&) [with Derived = Eigen::Matrix<double, -1, 1>; Eigen::DenseBase<Derived>::Scalar = double]((* &(& dp)->gtsam::Point2::x())).Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, 1> >((* &(& dp)->gtsam::Point2::y()))’ from ‘Eigen::CommaInitializer<Eigen::Matrix<double, -1, 1> >’ to ‘gtsam::Vector {aka Eigen::Matrix<double, -1, 1>}’
   static inline Vector Logmap(const Point2& dp) { return (Vector(2) << dp.x(), dp.y()); }
                                                                                      ^
In file included from /usr/local/include/gtsam/geometry/Pose2.h:27:0,
                 from /home/turtlebot/Project_slam/rtabmap/corelib/src/OptimizerGTSAM.cpp:39:
/usr/local/include/gtsam/geometry/Rot2.h: In static member function ‘static gtsam::Vector gtsam::Rot2::Logmap(const gtsam::Rot2&)’:
/usr/local/include/gtsam/geometry/Rot2.h:173:25: error: could not convert ‘Eigen::DenseBase<Derived>::operator<<(const Scalar&) [with Derived = Eigen::Matrix<double, -1, 1>; Eigen::DenseBase<Derived>::Scalar = double]((* &(& r)->gtsam::Rot2::theta()))’ from ‘Eigen::CommaInitializer<Eigen::Matrix<double, -1, 1> >’ to ‘gtsam::Vector {aka Eigen::Matrix<double, -1, 1>}’
       return (Vector(1) << r.theta());
                         ^
In file included from /home/turtlebot/Project_slam/rtabmap/corelib/src/OptimizerGTSAM.cpp:39:0:
/usr/local/include/gtsam/geometry/Pose2.h: In static member function ‘static gtsam::Matrix gtsam::Pose2::wedge(double, double, double)’:
/usr/local/include/gtsam/geometry/Pose2.h:178:20: error: could not convert ‘(&(&(&(&(&(&(& Eigen::DenseBase<Derived>::operator<<(const Scalar&) [with Derived = Eigen::Matrix<double, -1, -1>; Eigen::DenseBase<Derived>::Scalar = double]((* &0.0)).Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >((* &(- w))))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >((*(const Scalar*)(& vx))))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >((*(const Scalar*)(& w))))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >((* &0.0)))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >((*(const Scalar*)(& vy))))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >((* &0.0)))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >((* &0.0)))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >((* &0.0))’ from ‘Eigen::CommaInitializer<Eigen::Matrix<double, -1, -1> >’ to ‘gtsam::Matrix {aka Eigen::Matrix<double, -1, -1>}’
         0., 0.,  0.);
                    ^
In file included from /usr/local/include/gtsam/geometry/Pose3.h:30:0,
                 from /home/turtlebot/Project_slam/rtabmap/corelib/src/OptimizerGTSAM.cpp:40:
/usr/local/include/gtsam/geometry/Rot3.h: In static member function ‘static gtsam::Rot3 gtsam::Rot3::rodriguez(double, double, double)’:
/usr/local/include/gtsam/geometry/Rot3.h:197:51: error: no matching function for call to ‘gtsam::Rot3::rodriguez(Eigen::CommaInitializer<Eigen::Matrix<double, -1, 1> >&)’
       { return rodriguez((Vector(3) << wx, wy, wz));}
                                                   ^
/usr/local/include/gtsam/geometry/Rot3.h:197:51: note: candidates are:
/usr/local/include/gtsam/geometry/Rot3.h:164:17: note: static gtsam::Rot3 gtsam::Rot3::rodriguez(const Vector&, double)
     static Rot3 rodriguez(const Vector& w, double theta);
                 ^
/usr/local/include/gtsam/geometry/Rot3.h:164:17: note:   candidate expects 2 arguments, 1 provided
/usr/local/include/gtsam/geometry/Rot3.h:172:17: note: static gtsam::Rot3 gtsam::Rot3::rodriguez(const gtsam::Point3&, double)
     static Rot3 rodriguez(const Point3& w, double theta);
                 ^
/usr/local/include/gtsam/geometry/Rot3.h:172:17: note:   candidate expects 2 arguments, 1 provided
[... cut ...]
     inline gtsam::Vector localCoordinates(const SwitchVariableSigmoid& t2) const { return gtsam::Vector1(t2.value() - value()); }
                                                                                                                                 ^
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/OptimizerGTSAM.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
make: *** [all] Error 2
```

My gtsam version is 3.2.1, The error is appear because my gtsam version mistake?

QinZiwen, 2016-10-03:
new error appear:
```
[  8%] Built target rtabmap_ros_gencfg
Linking CXX executable /home/turtlebot/catkin_ws/devel/lib/rtabmap_ros/icp_odometry
[  8%] Built target _rtabmap_ros_generate_messages_check_deps_OdomInfo
[  8%] Built target _rtabmap_ros_generate_messages_check_deps_Point2f
Linking CXX executable /home/turtlebot/catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry
Linking CXX executable /home/turtlebot/catkin_ws/devel/lib/rtabmap_ros/rgbdicp_odometry
[  9%] Built target rtabmap
Linking CXX executable /home/turtlebot/catkin_ws/devel/lib/rtabmap_ros/stereo_odometry
/usr/bin/ld: warning: libfreenect.so.0.2, needed by /opt/ros/indigo/lib/x86_64-linux-gnu/librtabmap_core.so, may conflict with libfreenect.so.0.5
CMakeFiles/icp_odometry.dir/src/ICPOdometryNode.cpp.o: In function `main':
ICPOdometryNode.cpp:(.text+0x110): undefined reference to `rtabmap::Parameters::getDefaultOdometryParameters(bool, bool, bool)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/turtlebot/catkin_ws/devel/lib/rtabmap_ros/icp_odometry] Error 1
make[1]: *** [rtabmap_ros/CMakeFiles/icp_odometry.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: warning: libfreenect.so.0.2, needed by /opt/ros/indigo/lib/x86_64-linux-gnu/librtabmap_core.so, may conflict with libfreenect.so.0.5
CMakeFiles/stereo_odometry.dir/src/StereoOdometryNode.cpp.o: In function `main':
StereoOdometryNode.cpp:(.text+0x110): undefined reference to `rtabmap::Parameters::getDefaultOdometryParameters(bool, bool, bool)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/turtlebot/catkin_ws/devel/lib/rtabmap_ros/stereo_odometry] Error 1
make[1]: *** [rtabmap_ros/CMakeFiles/stereo_odometry.dir/all] Error 2
/usr/bin/ld: warning: libfreenect.so.0.2, needed by /opt/ros/indigo/lib/x86_64-linux-gnu/librtabmap_core.so, may conflict with libfreenect.so.0.5
CMakeFiles/rgbd_odometry.dir/src/RGBDOdometryNode.cpp.o: In function `main':
RGBDOdometryNode.cpp:(.text+0x110): undefined reference to `rtabmap::Parameters::getDefaultOdometryParameters(bool, bool, bool)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/turtlebot/catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry] Error 1
make[1]: *** [rtabmap_ros/CMakeFiles/rgbd_odometry.dir/all] Error 2
/usr/bin/ld: warning: libfreenect.so.0.2, needed by /opt/ros/indigo/lib/x86_64-linux-gnu/librtabmap_core.so, may conflict with libfreenect.so.0.5
CMakeFiles/rgbdicp_odometry.dir/src/RGBDICPOdometryNode.cpp.o: In function `main':
RGBDICPOdometryNode.cpp:(.text+0x110): undefined reference to `rtabmap::Parameters::getDefaultOdometryParameters(bool, bool, bool)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/turtlebot/catkin_ws/devel/lib/rtabmap_ros/rgbdicp_odometry] Error 1
make[1]: *** [rtabmap_ros/CMakeFiles/rgbdicp_odometry.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```

matlabbe, 2016-10-03:
Can you do:
```
$ catkin_make -j1 VERBOSE=1
```

Looks like you have still rtabmap binaries installed (`/opt/ros/indigo/...`). You should remove them:
```
$ sudo apt-get remove ros-indigo-rtabmap
```

cheers

QinZiwen, 2016-10-04:
Thanks for your help, but there are new errors:
```
[ 10%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap.dir/src/CoreNode.cpp.o
Linking CXX executable /home/turtlebot/catkin_ws/devel/lib/rtabmap_ros/icp_odometry
Linking CXX executable /home/turtlebot/catkin_ws/devel/lib/rtabmap_ros/rgbdicp_odometry
Linking CXX executable /home/turtlebot/catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry
In file included from /home/turtlebot/catkin_ws/src/rtabmap_ros/src/CoreNode.cpp:28:0:
/home/turtlebot/catkin_ws/src/rtabmap_ros/include/rtabmap_ros/CoreWrapper.h:47:32: fatal error: rtabmap_ros/GetMap.h: No such file or directory
 #include "rtabmap_ros/GetMap.h"
                                ^
compilation terminated.
[ 10%] Built target icp_odometry
Scanning dependencies of target stereo_odometry
[ 10%] Building CXX object rtabmap_ros/CMakeFiles/stereo_odometry.dir/src/StereoOdometryNode.cpp.o
[ 10%] Built target rgbdicp_odometry
[ 11%] Built target find_object_2d_generate_messages_lisp
[ 11%] Built target rgbd_odometry
[ 11%] Built target find_object_2d_generate_messages_cpp
[ 12%] Built target find_object_2d_generate_messages_py
[ 16%] Built target frontier_exploration_generate_messages_cpp
[ 16%] Built target turtlebot_exploration_3d
[ 21%] Built target frontier_exploration_generate_messages_lisp
[ 26%] Scanning dependencies of target rtabmap_ros_generate_messages_cpp
Built target frontier_exploration_generate_messages_py
[ 27%] Generating C++ code from rtabmap_ros/MapData.msg
Scanning dependencies of target rtabmap_ros_generate_messages_py
[ 28%] Generating Python from MSG rtabmap_ros/MapData
[ 28%] Generating Python from MSG rtabmap_ros/NodeData
[ 29%] Generating Python from MSG rtabmap_ros/OdomInfo
[ 29%] Generating C++ code from rtabmap_ros/NodeData.msg
[ 29%] Generating Python from MSG rtabmap_ros/MapGraph
[ 30%] Generating Python from MSG rtabmap_ros/Goal
[ 30%] Generating Python from MSG rtabmap_ros/UserData
[ 30%] Generating Python from MSG rtabmap_ros/Link
[ 31%] Generating C++ code from rtabmap_ros/OdomInfo.msg
[ 32%] Generating Python from MSG rtabmap_ros/Info
[ 32%] Generating Python from MSG rtabmap_ros/RGBDImage
[ 34%] Generating Python from MSG rtabmap_ros/KeyPoint
[ 34%] Generating C++ code from rtabmap_ros/MapGraph.msg
[ 34%] Generating Python from MSG rtabmap_ros/Point2f
[ 35%] Generating Python from MSG rtabmap_ros/Point3f
[ 35%] Generating Python code from SRV rtabmap_ros/GetMap
[ 35%] [ 35%] Generating Python code from SRV rtabmap_ros/ListLabels
Generating C++ code from rtabmap_ros/Goal.msg
[ 36%] Generating Python code from SRV rtabmap_ros/SetGoal
[ 36%] Generating Python code from SRV rtabmap_ros/ResetPose
[ 37%] Generating Python code from SRV rtabmap_ros/SetLabel
[ 38%] Generating C++ code from rtabmap_ros/UserData.msg
[ 38%] Generating Python code from SRV rtabmap_ros/PublishMap
[ 38%] Generating Python msg __init__.py for rtabmap_ros
[ 39%] Generating Python srv __init__.py for rtabmap_ros
[ 39%] Built target rtabmap_ros_generate_messages_py
Scanning dependencies of target rtabmap_ros_generate_messages_lisp
[ 39%] [ 39%] Generating C++ code from rtabmap_ros/Link.msg
Generating Lisp code from rtabmap_ros/MapData.msg
[ 40%] Generating Lisp code from rtabmap_ros/NodeData.msg
[ 40%] Generating Lisp code from rtabmap_ros/OdomInfo.msg
[ 41%] Generating C++ code from rtabmap_ros/Info.msg
[ 42%] Generating Lisp code from rtabmap_ros/MapGraph.msg
[ 42%] Generating Lisp code from rtabmap_ros/Goal.msg
[ 42%] Generating Lisp code from rtabmap_ros/UserData.msg
[ 42%] Generating C++ code from rtabmap_ros/RGBDImage.msg
[ 43%] Generating Lisp code from rtabmap_ros/Link.msg
[ 43%] Generating Lisp code from rtabmap_ros/Info.msg
[ 44%] Generating Lisp code from rtabmap_ros/RGBDImage.msg
[ 45%] Generating C++ code from rtabmap_ros/KeyPoint.msg
[ 45%] Generating Lisp code from rtabmap_ros/KeyPoint.msg
[ 46%] Generating Lisp code from rtabmap_ros/Point2f.msg
[ 46%] Generating Lisp code from rtabmap_ros/Point3f.msg
[ 46%] Generating Lisp code from rtabmap_ros/GetMap.srv
[... 59 lines cut ...]
[ 91%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/CommonDataSubscriber.cpp.o
[ 92%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/CoreWrapper.cpp.o
[ 92%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/impl/CommonDataSubscriberDepth.cpp.o
/home/turtlebot/catkin_ws/src/rtabmap_ros/src/MapsManager.cpp: In member function ‘std::map<int, rtabmap::Transform> MapsManager::updateMapCaches(const std::map<int, rtabmap::Transform>&, const rtabmap::Memory*, bool, bool, const std::map<int, rtabmap::Signature>&)’:
/home/turtlebot/catkin_ws/src/rtabmap_ros/src/MapsManager.cpp:451:73: error: ‘class rtabmap::OccupancyGrid’ has no member named ‘isGridFromDepth’
       data = memory->getSignatureDataConst(iter->first, occupancyGrid_->isGridFromDepth() && !occupancySavedInDB, !occupancyGrid_->isGridFromDepth() && !occupancySavedInDB, false, true);
                                                                         ^
/home/turtlebot/catkin_ws/src/rtabmap_ros/src/MapsManager.cpp:451:132: error: ‘class rtabmap::OccupancyGrid’ has no member named ‘isGridFromDepth’
       data = memory->getSignatureDataConst(iter->first, occupancyGrid_->isGridFromDepth() && !occupancySavedInDB, !occupancyGrid_->isGridFromDepth() && !occupancySavedInDB, false, true);
                                                                                                                                    ^
[... cut ...]
                          ^
/home/turtlebot/catkin_ws/src/rtabmap_ros/src/MapsManager.cpp:480:27: error: ‘class rtabmap::OccupancyGrid’ has no member named ‘isGridFromDepth’
          !occupancyGrid_->isGridFromDepth() && generateGrid?&scan:0,
                           ^
make[2]: *** [rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/MapsManager.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [rtabmap_ros/CMakeFiles/rtabmap_ros.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```

matlabbe, 2016-10-04:
The `isGridFromDepth` method was [added](https://github.com/introlab/rtabmap/commit/6f00cba872db876dde29d4dc3709235d99e58d58) yesterday to rtabmap. Update your rtabmap source and make install again.

For the `GetMap.h` error, it is an header created when ros services are generated (here [GetMap.srv](https://github.com/introlab/rtabmap_ros/blob/master/srv/GetMap.srv)). Make sure to re-run cmake in your `catkin_ws/build` directory so the header can be generated.

cheers

QinZiwen, 2016-10-06:
Sorry for the delay in my reply, my lab have been power cut in pass two days. when I run kinect2's launch , some errors show as following:
```
$ roslaunch kinect2_bridge kinect2_bridge.launch publish_tf:=true 
$ roslaunch rtabmap_ros rgbd_mapping_kinect2.launch resolution:=qhd
... logging to /home/turtlebot/.ros/log/09ca26f4-8b6b-11e6-bf86-6480999b1afe/roslaunch-turtlebot-HP-ZBook-14-G2-5931.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.103:33172/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.20
 * /rtabmap/rtabmap/Vis/InlierDistance: 0.02
 * /rtabmap/rtabmap/Vis/MinInliers: 20
 * /rtabmap/rtabmap/approx_sync: True
 * /rtabmap/rtabmap/frame_id: kinect2_base_link
 * /rtabmap/rtabmap/subscribe_depth: True
 * /rtabmap/rtabmapviz/approx_sync: True
 * /rtabmap/rtabmapviz/frame_id: kinect2_base_link
 * /rtabmap/rtabmapviz/subscribe_depth: True
 * /rtabmap/rtabmapviz/subscribe_odom_info: True
 * /rtabmap/visual_odometry/GFTT/MinDistance: 7
 * /rtabmap/visual_odometry/Odom/FillInfoData: true
 * /rtabmap/visual_odometry/Odom/Strategy: 0
 * /rtabmap/visual_odometry/OdomF2M/MaxSize: 1000
 * /rtabmap/visual_odometry/Vis/CorNNType: 3
 * /rtabmap/visual_odometry/Vis/FeatureType: 6
 * /rtabmap/visual_odometry/Vis/InlierDistance: 0.02
 * /rtabmap/visual_odometry/Vis/MaxDepth: 10.0
 * /rtabmap/visual_odometry/Vis/MaxFeatures: 1000
 * /rtabmap/visual_odometry/Vis/MinInliers: 20
 * /rtabmap/visual_odometry/approx_sync: True
 * /rtabmap/visual_odometry/frame_id: kinect2_base_link

NODES
  /rtabmap/
    rtabmap (rtabmap_ros/rtabmap)
    rtabmapviz (rtabmap_ros/rtabmapviz)
    visual_odometry (rtabmap_ros/rgbd_odometry)
  /
    kinect2_base_link (tf/static_transform_publisher)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[kinect2_base_link-1]: started with pid [5949]
process[rtabmap/visual_odometry-2]: started with pid [5950]
process[rtabmap/rtabmap-3]: started with pid [5960]
process[rtabmap/rtabmapviz-4]: started with pid [5962]
[rtabmap/visual_odometry-2] process has died [pid 5950, exit code -11, cmd /home/turtlebot/catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry rgb/image:=/kinect2/qhd/image_color_rect depth/image:=/kinect2/qhd/image_depth_rect rgb/camera_info:=/kinect2/qhd/camera_info __name:=visual_odometry __log:=/home/turtlebot/.ros/log/09ca26f4-8b6b-11e6-bf86-6480999b1afe/rtabmap-visual_odometry-2.log].
log file: /home/turtlebot/.ros/log/09ca26f4-8b6b-11e6-bf86-6480999b1afe/rtabmap-visual_odometry-2*.log
[rtabmap/rtabmap-3] process has died [pid 5960, exit code -11, cmd /home/turtlebot/catkin_ws/devel/lib/rtabmap_ros/rtabmap --delete_db_on_start rgb/image:=/kinect2/qhd/image_color_rect depth/image:=/kinect2/qhd/image_depth_rect rgb/camera_info:=/kinect2/qhd/camera_info __name:=rtabmap __log:=/home/turtlebot/.ros/log/09ca26f4-8b6b-11e6-bf86-6480999b1afe/rtabmap-rtabmap-3.log].
log file: /home/turtlebot/.ros/log/09ca26f4-8b6b-11e6-bf86-6480999b1afe/rtabmap-rtabmap-3*.log
[rtabmap/rtabmapviz-4] process has died [pid 5962, exit code -11, cmd /home/turtlebot/catkin_ws/devel/lib/rtabmap_ros/rtabmapviz -d /home/turtlebot/catkin_ws/src/rtabmap_ros/launch/config/rgbd_gui.ini rgb/image:=/kinect2/qhd/image_color_rect depth/image:=/kinect2/qhd/image_depth_rect rgb/camera_info:=/kinect2/qhd/camera_info __name:=rtabmapviz __log:=/home/turtlebot/.ros/log/09ca26f4-8b6b-11e6-bf86-6480999b1afe/rtabmap-rtabmapviz-4.log].
log file: /home/turtlebot/.ros/log/09ca26f4-8b6b-11e6-bf86-6480999b1afe/rtabmap-rtabmapviz-4*.log
```

willdzeng, 2016-10-06:
looks like visual_odometry crashed, did you subscribed the right topic?
try to pull the new rtabmap and rtabmap_ros and build again.

QinZiwen, 2016-10-07:
The topic right, and I am try to pull the new rtabmap and rtabmap_ros, build again, no error. But when I run, the errors above show again. In .ros/log/, there no have the log file that my terminal cue.

willdzeng, 2016-10-07:
feel like it's because you didn't clean the old rtabmap in indigo
try
`sudo apt-get purge ros-indigo-rtabmap ros-indigo-rtabmap-ros
`
and do
`sudo ldconfig`

after you rebuild and installed both.

matlabbe, 2016-10-07:
The nodes may not load the right version of `librtabmap_core.so`, so crashing when loading shared libraries. Can you show the output of:
```
$ ldd /home/turtlebot/catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry
```

QinZiwen, 2016-10-08:
```
$ ldd /home/turtlebot/catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry
	linux-vdso.so.1 =>  (0x00007ffeca1e1000)
	libroscpp.so => /opt/ros/indigo/lib/libroscpp.so (0x00007ff67ce29000)
	librosconsole.so => /opt/ros/indigo/lib/librosconsole.so (0x00007ff67cbfd000)
	libboost_system.so.1.54.0 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.54.0 (0x00007ff67c9f9000)
	libnodeletlib.so => /opt/ros/indigo/lib/libnodeletlib.so (0x00007ff67c79f000)
	librtabmap_core.so.0.11 => /usr/local/lib/librtabmap_core.so.0.11 (0x00007ff67bf8e000)
	librtabmap_utilite.so.0.11 => /usr/local/lib/librtabmap_utilite.so.0.11 (0x00007ff67bd71000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007ff67ba6d000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007ff67b857000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007ff67b492000)
	libroscpp_serialization.so => /opt/ros/indigo/lib/libroscpp_serialization.so (0x00007ff67b28f000)
	librostime.so => /opt/ros/indigo/lib/librostime.so (0x00007ff67b065000)
	libxmlrpcpp.so => /opt/ros/indigo/lib/libxmlrpcpp.so (0x00007ff67ae47000)
	libcpp_common.so => /opt/ros/indigo/lib/libcpp_common.so (0x00007ff67ac3f000)
	libboost_thread.so.1.54.0 => /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.54.0 (0x00007ff67aa29000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007ff67a80b000)
	libboost_filesystem.so.1.54.0 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.54.0 (0x00007ff67a5f5000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007ff67a2ef000)
	librosconsole_log4cxx.so => /opt/ros/indigo/lib/librosconsole_log4cxx.so (0x00007ff67a0db000)
	librosconsole_backend_interface.so => /opt/ros/indigo/lib/librosconsole_backend_interface.so (0x00007ff679ed9000)
	liblog4cxx.so.10 => /usr/lib/liblog4cxx.so.10 (0x00007ff679af1000)
	libboost_regex.so.1.54.0 => /usr/lib/x86_64-linux-gnu/libboost_regex.so.1.54.0 (0x00007ff6797ea000)
	libbondcpp.so => /opt/ros/indigo/lib/libbondcpp.so (0x00007ff6795ba000)
	libtinyxml.so.2.6.2 => /usr/lib/x86_64-linux-gnu/libtinyxml.so.2.6.2 (0x00007ff6793a5000)
	libclass_loader.so => /opt/ros/indigo/lib/libclass_loader.so (0x00007ff679186000)
	libroslib.so => /opt/ros/indigo/lib/libroslib.so (0x00007ff678f74000)
	libconsole_bridge.so.0.2 => /usr/lib/x86_64-linux-gnu/libconsole_bridge.so.0.2 (0x00007ff678d6a000)
	libopencv_video.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_video.so.2.4 (0x00007ff678b16000)
	libopencv_imgproc.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.2.4 (0x00007ff678686000)
	libopencv_highgui.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4 (0x00007ff67843b000)
	libopencv_gpu.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_gpu.so.2.4 (0x00007ff6781f1000)
	libopencv_features2d.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_features2d.so.2.4 (0x00007ff677f4e000)
	libopencv_core.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4 (0x00007ff677b17000)
	libopencv_calib3d.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.2.4 (0x00007ff677881000)
	libpcl_common.so.1.7 => /usr/lib/libpcl_common.so.1.7 (0x00007ff6775c7000)
	libvtkCommon.so.5.8 => /usr/lib/libvtkCommon.so.5.8 (0x00007ff677051000)
	libpcl_io.so.1.7 => /usr/lib/libpcl_io.so.1.7 (0x00007ff676c22000)
	libpcl_kdtree.so.1.7 => /usr/lib/libpcl_kdtree.so.1.7 (0x00007ff676905000)
	libpcl_search.so.1.7 => /usr/lib/libpcl_search.so.1.7 (0x00007ff676566000)
	libpcl_sample_consensus.so.1.7 => /usr/lib/libpcl_sample_consensus.so.1.7 (0x00007ff67591d000)
	libpcl_filters.so.1.7 => /usr/lib/libpcl_filters.so.1.7 (0x00007ff6750e0000)
	libpcl_features.so.1.7 => /usr/lib/libpcl_features.so.1.7 (0x00007ff6737e1000)
	libpcl_segmentation.so.1.7 => /usr/lib/libpcl_segmentation.so.1.7 (0x00007ff672cbc000)
	libpcl_surface.so.1.7 => /usr/lib/libpcl_surface.so.1.7 (0x00007ff67269f000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007ff672486000)
	libfreenect.so.0.5 => /opt/ros/indigo/lib/libfreenect.so.0.5 (0x00007ff672277000)
	libOpenNI2.so.0 => /usr/lib/libOpenNI2.so.0 (0x00007ff67201c000)
	libdc1394.so.22 => /usr/lib/x86_64-linux-gnu/libdc1394.so.22 (0x00007ff671da8000)
	libg2o_core.so => /opt/ros/indigo/lib/libg2o_core.so (0x00007ff671b52000)
	libg2o_types_slam2d.so => /opt/ros/indigo/lib/libg2o_types_slam2d.so (0x00007ff671913000)
	libg2o_types_slam3d.so => /opt/ros/indigo/lib/libg2o_types_slam3d.so (0x00007ff6716c6000)
	libg2o_types_sba.so => /opt/ros/indigo/lib/libg2o_types_sba.so (0x00007ff67147f000)
	libg2o_stuff.so => /opt/ros/indigo/lib/libg2o_stuff.so (0x00007ff671264000)
	libg2o_csparse_extension.so => /opt/ros/indigo/lib/libg2o_csparse_extension.so (0x00007ff67105f000)
	libcxsparse.so.3.1.2 => /usr/lib/x86_64-linux-gnu/libcxsparse.so.3.1.2 (0x00007ff670e35000)
	libcholmod.so.2.1.2 => /usr/lib/x86_64-linux-gnu/libcholmod.so.2.1.2 (0x00007ff670b62000)
	libgtsam.so.4 => /usr/local/lib/libgtsam.so.4 (0x00007ff6705b6000)
	libgomp.so.1 => /usr/lib/x86_64-linux-gnu/libgomp.so.1 (0x00007ff6703a7000)
	libcvsba.so => /usr/local/lib/libcvsba.so (0x00007ff670174000)
	liboctomap.so.1.6 => /opt/ros/indigo/lib/liboctomap.so.1.6 (0x00007ff66ff2c000)
	liboctomath.so.1.6 => /opt/ros/indigo/lib/liboctomath.so.1.6 (0x00007ff66fd26000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007ff66fb22000)
	libtbb.so.2 => /usr/lib/libtbb.so.2 (0x00007ff66f8ee000)
	/lib64/ld-linux-x86-64.so.2 (0x00007ff67d18e000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007ff66f6e6000)
[... cut ...]
	libkrb5.so.26 => /usr/lib/x86_64-linux-gnu/libkrb5.so.26 (0x00007ff652e3c000)
	libasn1.so.8 => /usr/lib/x86_64-linux-gnu/libasn1.so.8 (0x00007ff652b9b000)
	libhcrypto.so.4 => /usr/lib/x86_64-linux-gnu/libhcrypto.so.4 (0x00007ff652968000)
	libroken.so.18 => /usr/lib/x86_64-linux-gnu/libroken.so.18 (0x00007ff652753000)
	libwind.so.0 => /usr/lib/x86_64-linux-gnu/libwind.so.0 (0x00007ff65252a000)
	libheimbase.so.1 => /usr/lib/x86_64-linux-gnu/libheimbase.so.1 (0x00007ff65231c000)
	libhx509.so.5 => /usr/lib/x86_64-linux-gnu/libhx509.so.5 (0x00007ff6520d3000)
```

QinZiwen, 2016-10-08:
@willdzeng , I follow you running this command, but the error show again.

willdzeng, 2016-10-08:
An easy way to fix is just to reinstall ubuntu and ROS and build rtabmap from source again.
Sometimes linkage problem just wired and hard to fix for some reason.

One thing it worked for me is go to rtbamap's build folder and do a 
`sudo make uninstall`
and make install again.
'sudo make install'

matlabbe, 2016-10-11:
Sorry `rgbd_odometry` is a nodelet, so we cannot see which `librtabmap_ros.so` is linked. Can you show `ldd` for `rtabmapviz`? similarly to this:
```bash
~/catkin_ws/devel/lib/rtabmap_ros$ ldd rtabmapviz | grep rtabmap
	librtabmap_ros.so => /home/mathieu/catkin_ws/devel/lib/librtabmap_ros.so (0x00007feb8dd7e000)
	librtabmap_core.so.0.11 => /home/mathieu/catkin_ws/devel/lib/librtabmap_core.so.0.11 (0x00007feb8acea000)
	librtabmap_utilite.so.0.11 => /home/mathieu/catkin_ws/devel/lib/librtabmap_utilite.so.0.11 (0x00007feb8aacc000)
	librtabmap_gui.so.0.11 => /home/mathieu/catkin_ws/devel/lib/librtabmap_gui.so.0.11 (0x00007feb8a57a000)
```

QinZiwen, 2016-10-12:
```
$ ldd rtabmapviz | grep rtabmap
librtabmap_ros.so => /home/turtlebot/catkin_ws/devel/lib/librtabmap_ros.so (0x00007fb5d651d000)
librtabmap_core.so.0.11 => /usr/local/lib/librtabmap_core.so.0.11 (0x00007fb5d3529000)
librtabmap_utilite.so.0.11 => /usr/local/lib/librtabmap_utilite.so.0.11 (0x00007fb5d330c000)
librtabmap_gui.so.0.11 => /usr/local/lib/librtabmap_gui.so.0.11 (0x00007fb5d2db8000)
```

matlabbe, 2016-10-12:
I don't have a cue of what is the problem. Is standalone `$ rtabmap` crash too? You can look at the issue [Debugging "process has died" errors on startup](https://github.com/introlab/rtabmap_ros/issues/28) for more debugging tips. I've also seen on some machines that sometimes Release version of the compiled libraries crashes while Debug version doesn't.

QinZiwen, 2016-10-13:
@matlabbe Thank you very much!

QinZiwen, 2016-10-13:
When I rebuild rtabmap, terminal show some thing as follows which I think that it maybe is a bug:
```
In file included from /usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:52:0,
                 from /usr/include/pcl-1.7/pcl/visualization/cloud_viewer.h:39,
                 from /home/turtlebot/Project_slam/rtabmap/examples/NoEventsExample/main.cpp:32:
/usr/include/pcl-1.7/pcl/visualization/area_picking_event.h: In member function ‘bool pcl::visualization::AreaPickingEvent::getPointsIndices(std::vector<int>&) const’:
/usr/include/pcl-1.7/pcl/visualization/area_picking_event.h:64:30: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
           if (nb_points_ == -1)
```

willdzeng, 2016-10-13:
that's just a warning in pcl 1.7 library.

QinZiwen, 2016-10-13:
I type " roslaunch rtabmap_ros rgbd_mapping.launch launch_prefix:="xterm -e gdb -ex run --args" to debug rtabmap for tacking some useful thing.The log is as follows:
```
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/turtlebot/catkin_ws/devel/lib/rtabmap_ros/rtabmapviz...done.
Starting program: /home/turtlebot/catkin_ws/devel/lib/rtabmap_ros/rtabmapviz -d /home/turtlebot/catkin_ws/src/rtabmap_ros/launch/config/rgbd_gui.ini rgb/image:=/camera/rgb/image_rect_color depth/image:=/camera/depth_registered/image_raw rgb/camera_info:=/camera/rgb/camera_info left/image_rect:=/stereo_camera/left/image_rect_color right/image_rect:=/stereo_camera/right/image_rect left/camera_info:=/stereo_camera/left/camera_info right/camera_info:=/stereo_camera/right/camera_info scan:=/scan scan_cloud:=/scan_cloud __name:=rtabmapviz __log:=/home/turtlebot/.ros/log/7f5138ec-9109-11e6-956c-6480999b1afe/rtabmap-rtabmapviz-4.log
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
l
[New Thread 0x7fffbc409700 (LWP 16537)]

---Type <return> to continue, or q <return> to quit---l
Program received signal SIGSEGV, Segmentation fault.
0x00007fffe70f4adc in boost::math::lanczos::lanczos_initializer<boost::math::lanczos::lanczos17m64, long double>::init::init() ()
   from /usr/lib/libpcl_io.so.1.7
(gdb) l
38
39      void my_handler(int s){
40              ROS_INFO("rtabmapviz: ctrl-c catched! Exiting Qt app...");
41              spinner->stop();
42              exit(-1);
43      }
44
45      int main(int argc, char** argv)
46      {
47              ROS_INFO("Starting node...");
(gdb) break 47
Breakpoint 1 at 0x50cdf8: file /home/turtlebot/catkin_ws/src/rtabmap_ros/src/GuiNode.cpp, line 47.
(gdb) info break
Num     Type           Disp Enb Address            What
1       breakpoint     keep y   0x000000000050cdf8 in main(int, char**) 
                                                   at /home/turtlebot/catkin_ws/src/rtabmap_ros/src/GuiNode.cpp:47
(gdb) r
The program being debugged has been started already.
Start it from the beginning? (y or n) y

Starting program: /home/turtlebot/catkin_ws/devel/lib/rtabmap_ros/rtabmapviz -d /home/turtlebot/catkin_ws/src/rtabmap_ros/launch/config/rgbd_gui.ini rgb/image:=/camera/rgb/image_rect_color depth/image:=/camera/depth_registered/image_raw rgb/camera_info:=/camera/rgb/camera_info left/image_rect:=/stereo_camera/left/image_rect_color right/image_rect:=/stereo_camera/right/image_rect left/camera_info:=/stereo_camera/left/camera_info right/camera_info:=/stereo_camera/right/camera_info scan:=/scan scan_cloud:=/scan_cloud __name:=rtabmapviz __log:=/home/turtlebot/.ros/log/7f5138ec-9109-11e6-956c-6480999b1afe/rtabmap-rtabmapviz-4.log
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffbc409700 (LWP 16540)]

Program received signal SIGSEGV, Segmentation fault.
0x00007fffe70f4adc in boost::math::lanczos::lanczos_initializer<boost::math::lanczos::lanczos17m64, long double>::init::init() ()
   from /usr/lib/libpcl_io.so.1.7
(gdb) n
Single stepping until exit from function _ZN5boost4math7lanczos19lanczos_initializerINS1_12lanczos17m64EeE4initC2Ev,
which has no line number information.
[Thread 0x7fffbc409700 (LWP 16540) exited]

Program terminated with signal SIGSEGV, Segmentation fault.
The program no longer exists.
(gdb) n
The program is not being run.
(gdb) 
```

jacksonkr, 2016-10-13:
As far as I can tell, this fits in with this issue so I'm adding it:

```
inking CXX executable /home/jackson/catkin_ws/devel/lib/rtabmap_ros/icp_odometry
Linking CXX executable /home/jackson/catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry
/usr/bin/ld: warning: libfreenect.so.0.2, needed by /home/jackson/catkin_ws/devel/lib/x86_64-linux-gnu/librtabmap_core.so, may conflict with libfreenect.so.0.5
CMakeFiles/icp_odometry.dir/src/ICPOdometryNode.cpp.o: In function `main':
ICPOdometryNode.cpp:(.text+0x110): undefined reference to `rtabmap::Parameters::getDefaultOdometryParameters(bool, bool, bool)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/jackson/catkin_ws/devel/lib/rtabmap_ros/icp_odometry] Error 1
make[1]: *** [rtabmap_ros/CMakeFiles/icp_odometry.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: warning: libfreenect.so.0.2, needed by /home/jackson/catkin_ws/devel/lib/x86_64-linux-gnu/librtabmap_core.so, may conflict with libfreenect.so.0.5
CMakeFiles/rgbd_odometry.dir/src/RGBDOdometryNode.cpp.o: In function `main':
RGBDOdometryNode.cpp:(.text+0x110): undefined reference to `rtabmap::Parameters::getDefaultOdometryParameters(bool, bool, bool)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/jackson/catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry] Error 1
make[1]: *** [rtabmap_ros/CMakeFiles/rgbd_odometry.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j2 -l2" failed
```

matlabbe, 2016-10-13:
@QinZiwen Looks like a PCL error. I found this post: http://answers.ros.org/question/194699/segmentation-fault-when-using-correspondencerejectorsampleconsensus/

maybe related to C++11 or a bug in that PCL version:

> Removing "-std=c++0x" from my CMakeLists.txt worked as charm.

Can you show the output of CMake for the rtabmap library?

@jacksonkr Make sure you have the rtabmap library updated and installed. This definition of `getDefaultOdometryParameters` was [committed on August 7](https://github.com/introlab/rtabmap/commit/6ef9034a671ff23985fa35c917284198e156325e#diff-e1e477bdc507eec6bc8cb02ad6f6fdbf). In general, when updating the source code, follow [this](https://github.com/introlab/rtabmap_ros#update-to-new-version) to make sure that `rtabmap` and `rtabmap_ros` are compatible.

jacksonkr, 2016-10-13:
@matlabbe I had installed this last night using that page and it worked flawlessly so I'm doing the same on a different machine (similar setup) and it's a no go. Would you rather I move this to a new ticket or move it to the official forums?

matlabbe, 2016-10-13:
@jacksonkr yeah I moved the question to https://github.com/introlab/rtabmap_ros/issues/127

willdzeng, 2016-10-13:
area_picking_event.h:64:30: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
if (nb_points_ == -1)

is this really causing an issue?

matlabbe, 2016-10-13:
I see this warning all the time, should not be the problem here.

QinZiwen, 2016-10-14:
@matlabbe 
```
turtlebot@turtlebot-HP-ZBook-14-G2:~/Project_slam/rtabmap/build$ cmake -DCMAKE_INSTALL_PREFIX=~/catkin_ws/devel -DCMAKE_BUILD_TYPE=Debug .. 
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- checking for module 'eigen3'
--   found eigen3, version 3.2.0
-- Found eigen: /usr/include/eigen3  
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- Found openni: /usr/lib/libOpenNI.so  
-- Found libusb-1.0: /usr/include  
-- checking for module 'flann'
--   found flann, version 1.8.4
-- Found Flann: /usr/lib/x86_64-linux-gnu/libflann_cpp_s.a  
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- Found qhull: /usr/lib/x86_64-linux-gnu/libqhull.so  
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- Found PCL_COMMON: /usr/lib/libpcl_common.so  
-- Found PCL_OCTREE: /usr/lib/libpcl_octree.so  
-- Found PCL_IO: /usr/lib/libpcl_io.so  
-- Found PCL_KDTREE: /usr/lib/libpcl_kdtree.so  
-- Found PCL_SEARCH: /usr/lib/libpcl_search.so  
-- Found PCL_SAMPLE_CONSENSUS: /usr/lib/libpcl_sample_consensus.so  
-- Found PCL_FILTERS: /usr/lib/libpcl_filters.so  
-- Found PCL_FEATURES: /usr/lib/libpcl_features.so  
-- Found PCL_KEYPOINTS: /usr/lib/libpcl_keypoints.so  
-- Found PCL_GEOMETRY: /usr/include/pcl-1.7  
-- Found PCL_SEGMENTATION: /usr/lib/libpcl_segmentation.so  
-- Found PCL_VISUALIZATION: /usr/lib/libpcl_visualization.so  
-- Found PCL_OUTOFCORE: /usr/lib/libpcl_outofcore.so  
-- Found PCL_REGISTRATION: /usr/lib/libpcl_registration.so  
-- Found PCL_RECOGNITION: /usr/lib/libpcl_recognition.so  
-- Found PCL_SURFACE: /usr/lib/libpcl_surface.so  
-- Found PCL_PEOPLE: /usr/lib/libpcl_people.so  
-- Found PCL_TRACKING: /usr/lib/libpcl_tracking.so  
-- Found PCL_APPS: /usr/lib/libpcl_apps.so  
-- Try OpenMP C flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Try OpenMP CXX flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Found OpenMP: -fopenmp  
-- Found OpenMP
-- Found OpenCV: /usr/include/opencv;/usr/include
-- Found PCL: /usr/include/pcl-1.7;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/vtk-5.8
-- Found ZLIB: /usr/include
-- Looking for Q_WS_X11
-- Looking for Q_WS_X11 - found
-- Looking for Q_WS_WIN
-- Looking for Q_WS_WIN - not found
-- Looking for Q_WS_QWS
-- Looking for Q_WS_QWS - not found
-- Looking for Q_WS_MAC
-- Looking for Q_WS_MAC - not found
-- Found Qt4: /usr/bin/qmake (found version "4.8.6") 
-- Found QVTK: /usr/lib/libQVTK.so  
-- Found Freenect: /usr/include
-- Found OpenNI2: /usr/include/openni2
-- Found DC1394: /usr/include/dc1394
-- Found CSPARSE: /usr/include/suitesparse  
-- Found g2o: /opt/ros/indigo/include;/usr/include/suitesparse;/usr/include/suitesparse
-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include;/usr/include;/usr/local/include/gtsam/3rdparty/Eigen/
-- Found cvsba: /usr/local/include
-- Found octomap: /opt/ros/indigo/include
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Found Pthreads
-- --------------------------------------------
-- Info :
--   Version : 0.11.11
--   CMAKE_INSTALL_PREFIX = /home/turtlebot/catkin_ws/devel
--   CMAKE_BUILD_TYPE =     Debug
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -Wno-deprecated -fopenmp -std=c++11
--   With OpenCV 2 nonfree module (SIFT/SURF) = NO (not found, License: BSD)
--   With Freenect             = YES (License: Apache v2 and/or GPLv2)
--   With OpenNI2              = YES (License: Apache v2)
--   With Freenect2            = NO (libfreenect2 not found)
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--   With g2o                  = YES (License: BSD)
--   With GTSAM                = YES (License: BSD)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = YES (License: GPLv2)
--   With ZED                  = NO (ZED sdk not found)
--   With RealSense            = NO (librealsense not found)
--   With OCTOMAP              = YES (License: BSD)
--   With Qt4                  = YES (License: Open Source or Commercial)
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/turtlebot/Project_slam/rtabmap/build
```

matlabbe, 2016-10-14:
Ok, we will try [without](https://github.com/introlab/rtabmap/blob/403bfc2b44771ac0d60d48ac9eeecac7113463c9/CMakeLists.txt#L323-L337) `-std=c++11`. Do:
```bash
$ cd /home/turtlebot/Project_slam/rtabmap/build
$ cmake -DWITH_G2O=OFF -DWITH_GTSAM=OFF ..
# make sure on the cmake status there is no "-std=c++11" in CMAKE_CXX_FLAGS
$ make
$ ../bin/rtabmap
```

QinZiwen, 2016-10-17:
@matlabbe So sorry for the delay in my reply. The problem has been solve, thank you for your help."cmake -DWITH_G2O=OFF -DWITH_GTSAM=OFF .." means trying to without g2o and gtsam?

willdzeng, 2016-10-17:
yes, you will build the rtabmap without g2o and gtsam.

matlabbe, 2016-10-17:
So it is indeed a problem of libraries supporting c++11 and others not. The best is that **all** libraries are built with **or** without c++11. Upgrading to Ubuntu 16.04 may help for c++11 compatibility.

QinZiwen, 2016-10-17:
@matlabbe @willdzeng Thanks for your help during this time!

QinZiwen, 2016-10-17:
@matlabbe I had read your paper "Appearance-Based Loop Closure Detection for Online Large-Scale and Long-Term Operation". Does this paper main idea is memory management? In addition, what difference Rtabmap has compared with other SLAM?

matlabbe, 2016-10-17:
Yes the main novelty is the memory management for long-term **online** loop closure detection. In that particular paper, you can compare loop closure detection approach with other BOW (bag-of-words) approaches. I'm working on a paper comparing the whole SLAM system.

cheers

closed 2016-10-17 by matlabbe

svg091, 2019-06-28:
Hi 

I am getting below error, please help to solve the above error

-- The C compiler identification is GNU 7.2.1
-- The CXX compiler identification is GNU 7.2.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building on a 64 bit system
-- Using GNU gold as linker.
-- Disabling gc-sections on gold binutils < 2.26, see: https://sourceware.org/bugzilla/show_bug.cgi?id=17639
-- No build type specified, defaulting to Release
-- Configuring OSRM in release mode
-- Configuring release mode optimizations
-- Performing Test LTO_AVAILABLE
-- Performing Test LTO_AVAILABLE - Success
-- Performing Test HAS_COLOR_FLAG
-- Performing Test HAS_COLOR_FLAG - Success
-- Setting linker optimizations
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   date_time
--   chrono
--   filesystem
--   iostreams
--   program_options
--   regex
--   system
--   thread
--   unit_test_framework
--   atomic
-- Found Intel TBB
-- TBB interface version: 4001
-- Found EXPAT: /usr/lib64/libexpat.so (found version "2.1.0") 
-- Found BZip2: /usr/lib64/libbz2.so (found version "1.0.6") 
-- Looking for BZ2_bzCompressInit
-- Looking for BZ2_bzCompressInit - found
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.27.1") 
-- Found Lua: /usr/local/lib/liblua.a;/usr/lib64/libm.so;dl (found suitable version "5.3.5", minimum required is "5.2") 
-- Using Lua 5.3.5
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.2") 
-- Found ZLIB: /usr/lib64/libz.so (found version "1.2.8") 
-- Found Protozero: /home/vishu/route_eta_18_0_1/third_party/protozero/include (found suitable version "1.6.2", minimum required is "1.5.1") 
-- Found Osmium: /home/vishu/route_eta_18_0_1/third_party/libosmium/include  
-- Configuring done
CMake Warning at CMakeLists.txt:164 (add_executable):
  Cannot generate a safe runtime search path for target osrm-extract because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libtbb.so.2] in /usr/lib64 may be hidden by files in:
[... 271 lines cut ...]
  Cannot generate a safe runtime search path for target alias-bench because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libtbb.so.2] in /usr/lib64 may be hidden by files in:
      /usr/local/lib
    runtime library [libtbbmalloc.so.2] in /usr/lib64 may be hidden by files in:
      /usr/local/lib

  Some of these libraries may not be found correctly.

CMake Warning at src/benchmarks/CMakeLists.txt:44 (add_executable):
  Cannot generate a safe runtime search path for target packedvector-bench
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libtbb.so.2] in /usr/lib64 may be hidden by files in:
      /usr/local/lib
    runtime library [libtbbmalloc.so.2] in /usr/lib64 may be hidden by files in:
      /usr/local/lib

  Some of these libraries may not be found correctly.

-- Generating done
-- Build files have been written to:

matlabbe, 2019-07-02:
Those errors don't seem related to `rtabmap_ros`...