introlab/rtabmap_ros #123: cannot catkin_make, build error, ros indigo

aaravrav142, 2016-10-05:
Hi, 

I followed all the tutorials for installation from the wiki page and was able to install rtabmap with all optional dependencies (except freenect2 since I do not have usb 3.0). I put the rtabmap_ros package in the catkin_ws/src and when running catkin_make, I am getting the following errors. I tried to remove the package and re-installed - got same error. Can you help.

`~/catkin_ws$ catkin_make
Base path: /home/alexr/catkin_ws
Source space: /home/alexr/catkin_ws/src
Build space: /home/alexr/catkin_ws/build
Devel space: /home/alexr/catkin_ws/devel
Install space: /home/alexr/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/alexr/catkin_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/alexr/catkin_ws/build"
####
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target _homer_mapnav_msgs_generate_messages_check_deps_RegionsOfInterest
[  0%] [  0%] Built target _homer_mapnav_msgs_generate_messages_check_deps_PointOfInterest
Built target _homer_mapnav_msgs_generate_messages_check_deps_SaveMap
[  0%] Built target nav_msgs_generate_messages_py
[  0%] Built target _homer_mapnav_msgs_generate_messages_check_deps_DeletePointOfInterest
[  0%] Built target _homer_mapnav_msgs_generate_messages_check_deps_ModifyMap
[  0%] Built target _homer_mapnav_msgs_generate_messages_check_deps_StopNavigation
[  0%] Built target _homer_mapnav_msgs_generate_messages_check_deps_NavigateToPOI
[  0%] Built target _homer_mapnav_msgs_generate_messages_check_deps_GetPointsOfInterest
[  0%] Built target _homer_mapnav_msgs_generate_messages_check_deps_RegionOfInterest
[  0%] Built target _homer_mapnav_msgs_generate_messages_check_deps_ModifyPOI
[  0%] Built target _homer_mapnav_msgs_generate_messages_check_deps_DoMapping
[  0%] Built target _homer_mapnav_msgs_generate_messages_check_deps_LoadMap
[  0%] Built target _homer_mapnav_msgs_generate_messages_check_deps_StartNavigation
[  0%] Built target _homer_mapnav_msgs_generate_messages_check_deps_MapLayers
[  0%] Built target _homer_mapnav_msgs_generate_messages_check_deps_TargetUnreachable
[  0%] [  0%] Built target geometry_msgs_generate_messages_cpp
Built target _homer_mapnav_msgs_generate_messages_check_deps_PointsOfInterest
[  0%] Built target nav_msgs_generate_messages_cpp
[  0%] [  0%] Built target geometry_msgs_generate_messages_lisp
Built target nav_msgs_generate_messages_lisp
[  1%] [  4%] Built target _homer_mapnav_msgs_generate_messages_check_deps_GetRegionsOfInterest
Built target Math
[  5%] Built target String
[  7%] Built target TinyXML
[  7%] Built target Thread
[  7%] Built target _or_msgs_generate_messages_check_deps_KeyPointMatch
[  7%] Built target _or_msgs_generate_messages_check_deps_KeyPointMatchArray
[  7%] Built target _or_msgs_generate_messages_check_deps_OrMatchResult
[  7%] Built target _or_msgs_generate_messages_check_deps_RecognizeImage
[  7%] Built target _or_msgs_generate_messages_check_deps_KeyPoint
[  7%] Built target _or_msgs_generate_messages_check_deps_ExtractKeyPoints
[  7%] Built target _or_msgs_generate_messages_check_deps_Point2D
[  7%] [  7%] Built target sensor_msgs_generate_messages_py
Built target _or_msgs_generate_messages_check_deps_MatchResult
[  7%] [  7%] Built target _or_msgs_generate_messages_check_deps_OrLearnCommand
Built target _or_msgs_generate_messages_check_deps_BoundingBox2D
[  7%] [  7%] Built target _or_msgs_generate_messages_check_deps_OrLearningStatus
Built target _or_msgs_generate_messages_check_deps_OrCommand
[  7%] Built target sensor_msgs_generate_messages_cpp
[  7%] Built target sensor_msgs_generate_messages_lisp
[  7%] Built target rtimulib_ros
[  7%] Built target razor_imu_9dof_gencfg
[  7%] Built target imu_filter_madgwick_gencfg
[  7%] Built target Explorer
[  7%] Built target _or_msgs_generate_messages_check_deps_OrObjectNames
[  7%] Built target SpeedControl
[  8%] [  8%] Built target MappingMath
Built target std_srvs_generate_messages_cpp
[  8%] [  8%] Built target actionlib_msgs_generate_messages_py
Built target std_msgs_generate_messages_lisp
[  8%] Built target std_msgs_generate_messages_cpp
[  8%] Built target actionlib_msgs_generate_messages_cpp
[  8%] Built target std_msgs_generate_messages_py
[  8%] Built target actionlib_msgs_generate_messages_lisp
[  8%] Built target roscpp_generate_messages_cpp
[  8%] Built target roscpp_generate_messages_lisp
[  8%] Built target roscpp_generate_messages_py
[  8%] Built target rosgraph_msgs_generate_messages_lisp
[  8%] Built target rosgraph_msgs_generate_messages_py
[  8%] [  8%] Built target rosgraph_msgs_generate_messages_cpp
Built target actionlib_generate_messages_lisp
[  8%] Built target tf2_msgs_generate_messages_cpp
[  8%] Built target tf2_msgs_generate_messages_lisp
[  8%] [  8%] Built target tf2_msgs_generate_messages_py
Built target tf_generate_messages_lisp
[  8%] [  8%] Built target tf_generate_messages_py
Built target actionlib_generate_messages_cpp
[  8%] [  8%] Built target actionlib_generate_messages_py
Built target tf_generate_messages_cpp
[  8%] [  8%] Built target std_srvs_generate_messages_lisp
Built target std_srvs_generate_messages_py
[  8%] Built target ConnectedComponentAnalyzer
[  9%] [ 10%] Built target rbx1_nav_gencfg
Built target complementary_filter
[ 11%] Built target ParallelSurf
[ 11%] Built target rosaria_gencfg
[ 12%] Built target PointOfInterestGUI
[ 12%] Automoc for target rviz_imu_plugin
[ 12%] Built target demo_pioneer_joy2twist_pioneer_node
[ 13%] Built target MapTab
[ 13%] Built target rviz_imu_plugin_automoc
[ 13%] Built target _rosaria_generate_messages_check_deps_BumperState
[ 14%] Built target demo_pioneer_joy2twist_pioneer_pan_node
[ 14%] Built target demo_pioneer_visual_servo_pioneer_node
[ 14%] Built target demo_pioneer_visual_servo_pioneer_pan_node
[ 14%] Built target _rtabmap_ros_generate_messages_check_deps_KeyPoint
[ 14%] [ 14%] Built target _rtabmap_ros_generate_messages_check_deps_GetMap
Built target _rtabmap_ros_generate_messages_check_deps_SetGoal
[ 14%] Built target _rtabmap_ros_generate_messages_check_deps_SetLabel
[ 14%] Built target _rtabmap_ros_generate_messages_check_deps_PublishMap
[ 14%] Built target _rtabmap_ros_generate_messages_check_deps_Point2f
[ 14%] Built target _rtabmap_ros_generate_messages_check_deps_MapGraph
[ 14%] Built target _rtabmap_ros_generate_messages_check_deps_Link
[ 14%] [ 14%] Built target _rtabmap_ros_generate_messages_check_deps_Info
Built target _rtabmap_ros_generate_messages_check_deps_Goal
[ 14%] Built target _rtabmap_ros_generate_messages_check_deps_ListLabels
[ 14%] Built target _rtabmap_ros_generate_messages_check_deps_RGBDImage
[ 14%] Built target _rtabmap_ros_generate_messages_check_deps_MapData
[ 14%] Built target _rtabmap_ros_generate_messages_check_deps_Point3f
[ 14%] Built target _rtabmap_ros_generate_messages_check_deps_UserData
[ 14%] Built target _rtabmap_ros_generate_messages_check_deps_OdomInfo
[ 14%] Built target rtabmap_ros_gencfg
Linking CXX executable /home/alexr/catkin_ws/devel/lib/rtabmap_ros/icp_odometry
[ 14%] Built target _rtabmap_ros_generate_messages_check_deps_NodeData
Linking CXX executable /home/alexr/catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry
[ 14%] Built target _rtabmap_ros_generate_messages_check_deps_ResetPose
Linking CXX executable /home/alexr/catkin_ws/devel/lib/rtabmap_ros/rgbdicp_odometry
[ 14%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap.dir/src/CoreNode.cpp.o
/usr/local/lib/libopencv_nonfree.so.2.4: undefined reference to `cv::ocl::integral(cv::ocl::oclMat const&, cv::ocl::oclMat&)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/alexr/catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry] Error 1
make[1]: *** [rtabmap_ros/CMakeFiles/rgbd_odometry.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/local/lib/libopencv_nonfree.so.2.4: undefined reference to `cv::ocl::integral(cv::ocl::oclMat const&, cv::ocl::oclMat&)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/alexr/catkin_ws/devel/lib/rtabmap_ros/rgbdicp_odometry] Error 1
make[1]: *** [rtabmap_ros/CMakeFiles/rgbdicp_odometry.dir/all] Error 2
/usr/local/lib/libopencv_nonfree.so.2.4: undefined reference to `cv::ocl::integral(cv::ocl::oclMat const&, cv::ocl::oclMat&)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/alexr/catkin_ws/devel/lib/rtabmap_ros/icp_odometry] Error 1
make[1]: *** [rtabmap_ros/CMakeFiles/icp_odometry.dir/all] Error 2
In file included from /home/alexr/catkin_ws/src/rtabmap_ros/src/CoreNode.cpp:28:0:
/home/alexr/catkin_ws/src/rtabmap_ros/include/rtabmap_ros/CoreWrapper.h:47:32: fatal error: rtabmap_ros/GetMap.h: No such file or directory
 #include "rtabmap_ros/GetMap.h"
                                ^
compilation terminated.
make[2]: *** [rtabmap_ros/CMakeFiles/rtabmap.dir/src/CoreNode.cpp.o] Error 1
make[1]: *** [rtabmap_ros/CMakeFiles/rtabmap.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
`

Thanks

matlabbe, 2016-10-05:
Looks at these posts: 
 * https://github.com/introlab/rtabmap_ros/issues/11
 * http://official-rtab-map-forum.67519.x6.nabble.com/catkin-make-ros-pkg-failed-td22.html#a24

On your machine, you may need to disable OpenCV when building OpenCV:
```
$ cmake -D WITH_OPENCL=OFF
```

Make sure that `rtabmap` library was built against your OpenCV built from source "`/usr/local`". If you built OpenCV from source **after** building `rtabmap` and `rtabmap_ros` for the first time, clean the build directories (`rtabmap/build` and `catkin_ws/build`) to make sure that the correct OpenCV libraries are linked.

aaravrav142, 2016-10-06:
Hi Mathieu,

Using the suggestion I can do catkin_make now. However after the successful build, when I try to open the standalone rtabmap via $ rtabmap I am getting the following error. I am on ubuntu 14.04 ros indigo

$ rtabmap
rtabmap: error while loading shared libraries: librtabmap_core.so.0.11: cannot open shared object file: No such file or directory

willdzeng, 2016-10-06:
Did you do 
`sudo make install`
when you build rtabmap?

if you did, try use
`sudo ldconfig`
to fix the linkage problem.

if failed again, you need to uninstall rtabmap, try remove everything in /usr/local/lib that has rtabmap, and build and install rtabmap again.

aaravrav142, 2016-10-11:
Ok, I fixed the issue by @willdzeng  suggestion. Thanks. 

Regards
Alex

closed 2016-10-11 by aaravrav142