introlab/rtabmap_ros #124: rtabmap_ros/MapGraph display not working

Annaiyan, 2016-10-10:
Hi,

Recently i have uninstalled rviz and installed it from source like this
`sudo apt-get install ros-indigo-rviz`
 this the rviz path /`opt/ros/indigo/share/rviz`

rviz is working but the plugin for the rtabmap_ros/MapGraph is throwing the following error:

`[ERROR] [1476099795.724155220, 1415737735.869798101]: PluginlibFactory: The plugin for class 'rtabmap_ros/MapGraph' failed to load.  Error: MultiLibraryClassLoader: Could not create class of type rtabmap_ros::MapGraphDisplay
`
How to make the plugin working with rviz?

thanks in advance!

matlabbe, 2016-10-11:
If `rtabmap_ros` was built against your old RVIZ, then make sure to rebuild `rtabmap_ros` so it can be linked to right rviz plugin library.

Annaiyan, 2016-10-12:
yes, you are right. It was building against old rviz .
Initially i deleted the build and devel but it didn't work as expected. Once again i tried the same and restarted the system.

Surprisingly it works now!

many thanks for the support.

closed 2016-10-12 by Annaiyan