introlab/rtabmap_ros #131: CMake Warning: Cannot generate a safe runtime search path for target _

jacksonkr, 2016-10-26:
os: ubuntu 14.04.5 x86_64
kernel: 4.4.0-46-generic
rtabmap-ros: 0.11.8-56-gea85929
rtabmap: 0.11.8
opencv: 2.4.8 [without nonfree]
pcl: 1.7.1
+freenect
+stereo1394
+octomap

Its building and running fine but I'm getting some crazy warnings. Can I these be ignored?

```
$ catkin_make
Base path: /home/jackson/catkin_ws
Source space: /home/jackson/catkin_ws/src
Build space: /home/jackson/catkin_ws/build
Devel space: /home/jackson/catkin_ws/devel
Install space: /home/jackson/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/jackson/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/jackson/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/jackson/catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/jackson/catkin_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/jackson/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.18
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - rtabmap_ros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'rtabmap_ros'
-- ==> add_subdirectory(rtabmap_ros)
-- Using these message generators: gencpp;genlisp;genpy
-- rtabmap_ros: 12 messages, 6 services
-- WITH costmap_2d
-- WITH octomap
-- WITH rviz
-- Configuring done
CMake Warning at rtabmap_ros/CMakeLists.txt:296 (add_executable):
  Cannot generate a safe runtime search path for target camera because files
  in some directories may conflict with libraries in implicit directories:

    runtime library [libopencv_videostab.so.2.4] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libopencv_video.so.2.4] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libopencv_superres.so.2.4] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libopencv_stitching.so.2.4] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libopencv_photo.so.2.4] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libopencv_ocl.so.2.4] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libopencv_objdetect.so.2.4] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libopencv_ml.so.2.4] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libopencv_legacy.so.2.4] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libopencv_imgproc.so.2.4] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
[... 713 lines cut ...]
Scanning dependencies of target odom_msg_to_tf
Scanning dependencies of target rtabmapviz
Scanning dependencies of target map_optimizer
Scanning dependencies of target stereo_camera
[ 92%] Building CXX object rtabmap_ros/CMakeFiles/data_player.dir/src/DbPlayerNode.cpp.o
[ 93%] Building CXX object rtabmap_ros/CMakeFiles/odom_msg_to_tf.dir/src/OdomMsgToTFNode.cpp.o
[ 94%] [ 95%] [ 96%] Building CXX object rtabmap_ros/CMakeFiles/map_optimizer.dir/src/MapOptimizerNode.cpp.o
Building CXX object rtabmap_ros/CMakeFiles/map_assembler.dir/src/MapAssemblerNode.cpp.o
Building CXX object rtabmap_ros/CMakeFiles/stereo_camera.dir/src/StereoCameraNode.cpp.o
[ 97%] [ 98%] [100%] Building CXX object rtabmap_ros/CMakeFiles/rtabmapviz.dir/src/GuiWrapper.cpp.o
Building CXX object rtabmap_ros/CMakeFiles/rtabmapviz.dir/src/GuiNode.cpp.o
Building CXX object rtabmap_ros/CMakeFiles/rtabmapviz.dir/src/PreferencesDialogROS.cpp.o
Linking CXX executable /home/jackson/catkin_ws/devel/lib/rtabmap_ros/odom_msg_to_tf
Linking CXX executable /home/jackson/catkin_ws/devel/lib/rtabmap_ros/stereo_camera
Linking CXX executable /home/jackson/catkin_ws/devel/lib/rtabmap_ros/data_player
Linking CXX executable /home/jackson/catkin_ws/devel/lib/rtabmap_ros/map_optimizer
Linking CXX executable /home/jackson/catkin_ws/devel/lib/rtabmap_ros/map_assembler
Linking CXX executable /home/jackson/catkin_ws/devel/lib/rtabmap_ros/rtabmapviz
[100%] Built target stereo_camera
[100%] Built target odom_msg_to_tf
[100%] Built target data_player
[100%] Built target map_optimizer
[100%] Built target map_assembler
[100%] Built target rtabmapviz
```

matlabbe, 2016-10-30:
Make sure that in your `PATH` (or `LD_LIBRARY_PATH`), the opencv libraries found first are those corresponding to those used on compilation (cmake path).

closed 2016-10-31 by jacksonkr

gs14iitbbs, 2017-03-27:
Hi @jacksonkr I have installed ubuntu 14.04 + ROS indigo + openCv 2.4.8.

But I am getting similar error as you have discussed above..

-- Build files have been written to: /home/gaurav/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build
gaurav@gaurav-HP-Z420-Workstation:~/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build$ cmake .. -DROS_BUILD_TYPE=Release
[rosbuild] Building package ORB_SLAM2
[rosbuild] using multiarch 'x86_64-linux-gnu' for finding Boost
-- Using CATKIN_DEVEL_PREFIX: /home/gaurav/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/devel
-- Using CMAKE_PREFIX_PATH: /home/gaurav/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/devel;/home/gaurav/catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/gaurav/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/devel;/home/gaurav/catkin_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Skip enable_testing() for dry packages
-- Using CATKIN_TEST_RESULTS_DIR: /home/gaurav/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.19
-- Using these message generators: gencpp;genlisp;genpy
[rosbuild] Including /opt/ros/indigo/share/roslisp/rosbuild/roslisp.cmake
[rosbuild] Including /opt/ros/indigo/share/roscpp/rosbuild/roscpp.cmake
[rosbuild] Including /opt/ros/indigo/share/rospy/rosbuild/rospy.cmake
Build type: Release
-- Using flag -std=c++11.
-- Configuring done
CMake Warning at /opt/ros/indigo/share/ros/core/rosbuild/public.cmake:509 (add_executable):
  Cannot generate a safe linker search path for target RGBD because files in
  some directories may conflict with libraries in implicit directories:

    link library [libopencv_videostab.so.2.4.8] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    link library [libopencv_video.so.2.4.8] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    link library [libopencv_superres.so.2.4.8] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    link library [libopencv_stitching.so.2.4.8] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    link library [libopencv_photo.so.2.4.8] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    link library [libopencv_ocl.so.2.4.8] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    link library [libopencv_objdetect.so.2.4.8] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    link library [libopencv_ml.so.2.4.8] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    link library [libopencv_legacy.so.2.4.8] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    link library [libopencv_imgproc.so.2.4.8] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    link library [libopencv_highgui.so.2.4.8] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    link library [libopencv_gpu.so.2.4.8] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    link library [libopencv_flann.so.2.4.8] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    link library [libopencv_features2d.so.2.4.8] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    link library [libopencv_core.so.2.4.8] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    link library [libopencv_contrib.so.2.4.8] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    link library [libopencv_calib3d.so.2.4.8] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  CMakeLists.txt:90 (rosbuild_add_executable)

[... 125 lines cut ...]
  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  CMakeLists.txt:70 (rosbuild_add_executable)

-- Generating done
-- Build files have been written to: /home/gaurav/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build
gaurav@gaurav-HP-Z420-Workstation:~/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build$ 

--------------------------------------------------------------------------------------------------------------

And when I make  it  

-- Generating done
-- Build files have been written to: /home/gaurav/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build
gaurav@gaurav-HP-Z420-Workstation:~/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build$ make
[  0%] Built target rospack_genmsg_libexe
[  0%] Built target rosbuild_precompile
make[2]: *** No rule to make target `/usr/lib/x86_64-linux-gnu/libopencv_video.so.2.4.8', needed by `../RGBD'.  Stop.
make[1]: *** [CMakeFiles/RGBD.dir/all] Error 2
make: *** [all] Error 2
gaurav@gaurav-HP-Z420-Workstation:~/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build$ 

--------------------------------------------------------------------------------------------------------------

Please guide me how can I solve this issue

jacksonkr, 2017-03-28:
@gs14iitbbs In my original post I wasn't getting errors, I was getting warnings. I think you may have better luck at the [Official RTAB-Map Forums](http://official-rtab-map-forum.67519.x6.nabble.com/)

an99990, 2022-10-21:
@jacksonkr what was the solution ?
How do you fix your LD_LIBRARY_PATH ?

cgm-free, 2023-03-17:
# If the package has been found, several variables will
# be set, you can find the full list with descriptions
# in the OpenCVConfig.cmake file.
# Print some message showing some of them
message(STATUS "OpenCV library status:")
message(STATUS "    config: ${OpenCV_DIR}")
message(STATUS "    version: ${OpenCV_VERSION}")
message(STATUS "    libraries: ${OpenCV_LIBS}")
message(STATUS "    include path: ${OpenCV_INCLUDE_DIRS}")

If this is shown：
config: /usr/lib/x86_64-linux-gnu/cmake/opencv4

You can add this code to change OpenCV_DIR:
SET(OpenCV_DIR /usr/local/lib/cmake/opencv4/)   # 设置OpenCV_DIR

 Refer to：
https://blog.csdn.net/u013454780/article/details/128357962?spm=1001.2014.3001.5501#:~:text=%E6%9C%89%E8%AD%A6%E5%91%8A%E4%BF%A1%E6%81%AF-,%E8%A7%A3%E5%86%B3%EF%BC%9A,SET(OpenCV_DIR%20/usr/local/lib/cmake/opencv4/)%20%20%20%23%20%E8%AE%BE%E7%BD%AEOpenCV_DIR,-SET(OpenCV_DIR%20/usr

rohansingh080699, 2024-04-17:
@matlabbe oh okay thanks

TheSeanParker, 2024-12-31:
Unfortunately by the ubuntu apt-get install the PCL 
```
sudo apt update
sudo apt install libpcl-dev
```
the PCL isntallled location is /usr/lib/x86_64-linux-gnu ;if you installed by yourself ,
```
cd PCL 
cmake ..
make 
sudo make install
```
the PCL lib will be installed in /usr/local/lib/
as we all know that: Linux(cover the ubuntu) find the lib the order is /usr/lib、/lib 和/usr/local/lib;
so your algorithm first find the PCL lib order is /usr/local/lib,the next is /usr/lib/x86_64-linux-gnu,so when CompileAndLink algorithm find the PCL in /usr/local/lib,but when run the algorithm exe,the ROS find PCL in /usr/lib/x86_64-linux-gnu,so the solution is delete lib in /usr/local/lib,
`sudo rm -r /usr/local/lib/libpcl*`
and you will not see the error of 
```
  Cannot generate a safe runtime search path for target fastlio_mapping
  because files in some directories may conflict with libraries in implicit
  directories:
    runtime library [libpcl_common.so.1.12] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
  Some of these libraries may not be found correctly.
```
and the interlinkage is my article ,hope you like and marked ,thanks