introlab/rtabmap_ros #133: compile is regularly failing due to high memory useage

tfoote, 2016-11-14:
This job is frequently failing to compile due to high memory usage: http://build.ros.org/view/Jbin_arm_uThf/job/Jbin_arm_uThf__rtabmap_ros__ubuntu_trusty_armhf__binary/

This is causing a regression on Jade armhf.

matlabbe, 2016-11-15:
The build for armhf on Indigo and Kinetic seems working: 
http://build.ros.org/job/Ibin_arm_uThf__rtabmap_ros__ubuntu_trusty_armhf__binary/
http://build.ros.org/job/Kbin_uxhf_uXhf__rtabmap_ros__ubuntu_xenial_armhf__binary/

I think the faster solution is to blacklist rtabmap_ros on Jade for armhf for the moment.

cheers

tfoote, 2016-11-15:
I pushed the new release with this regression. If you'd like to blacklist it we can do that also.

matlabbe, 2016-11-15:
Yes, you can blacklist it for Jade armhf as I don't think I will fix it in the short term unless the build on Kinetic starts failing.