introlab/rtabmap_ros #134: demo_two_kinects.launch dumped

Severn-Vergil, 2016-11-24:
Hi everyone, 
I'm trying to use demo_two_kinects.launch and slightly change the second camera orientation to 90 degrees in counterclockwise, and translation in z direction, about 0.14m. I find that the visualization module can't keep stable in 10 seconds, either by rtabmapviz or rviz. However the proj_map is published stably and can be visualized in rviz. 
And here's the fatal error: 
[FATAL] (2016-11-23 20:57:45.144) util3d.cpp:370::cloudFromDepthRGB() Condition ((model.imageHeight() == 0 && model.imageWidth() == 0) || (model.imageHeight() == imageRgb.rows && model.imageWidth() == imageRgb.cols)) not met! 
If you have any idea about the error, pls post it! Thank you in advance! 

[image]
[image]
[image]

Severn-Vergil, 2016-11-27:
Here's my launch file:

<launch>
  
  
  <arg name="database_path"               default="rtabmap.db"/>
  <arg name="localization"                default="false"/>
  <arg name="rgbd_odometry"               default="false"/>
  <arg name="args"                        default="--delete_db_on_start"/>
  <arg name="rtabmapviz"                  default="true"/>
  <arg name="rviz"                        default="false" />
  <arg name="gui_cfg"                     default="~/.ros/rtabmap_gui.ini" />
  <arg name="rviz_cfg"                    default="-d $(find rtabmap_ros)/launch/config/rgbd.rviz" />
  <arg name="approx_sync"                 default="true"/>               
  <arg name="wait_for_transform_duration" default="0.2"/>

  
  <arg name="depth_cameras"       default="2" />

  <arg name="rgb0/image"          default="/camera2/rgb/image_rect_color" />
  <arg name="depth0/image"        default="/camera2/depth_registered/image_raw" />
  <arg name="rgb0/camera_info"    default="/camera2/rgb/camera_info" />

  <arg name="rgb1/image"          default="/camera1/rgb/image_rect_color" />
  <arg name="depth1/image"        default="/camera1/depth_registered/image_raw" />
  <arg name="rgb1/camera_info"    default="/camera1/rgb/camera_info" />   
  
  <arg name="rgbd_namespace"          default="/camera"/>
  <arg name="rgb_topic"               default="$(arg rgbd_namespace)/rgb/image_rect_color" />
  <arg name="depth_topic"             default="$(arg rgbd_namespace)/depth_registered/image_raw" />
  <arg name="camera_info_topic"       default="$(arg rgbd_namespace)/rgb/camera_info" />
     
  
  <group ns="rtabmap">

        <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="$(arg args)">
                <param name="database_path"                 type="string"   value="$(arg database_path)"/>
                <param name="frame_id"                      type="string"   value="base_link"/>
                <param name="odom_frame_id"                 type="string"   value="odom"/>
                <param name="wait_for_transform_duration"   type="double"   value="$(arg wait_for_transform_duration)"/>
                <param name="subscribe_depth"               type="bool"     value="true"/>
                <param name="subscribe_scan"                type="bool"     value="false"/>
                <param name="gen_scan"                      type="bool"     value="true"/>
                <param name="depth_cameras"                 type="int"      value="$(arg depth_cameras)"/>
                
                <remap from="rgb0/image"                    to="$(arg rgb0/image)"/>
                <remap from="depth0/image"                  to="$(arg depth0/image)"/>
                <remap from="rgb0/camera_info"              to="$(arg rgb0/camera_info)"/>

                <remap from="rgb1/image"                    to="$(arg rgb1/image)"/>
                <remap from="depth1/image"                  to="$(arg depth1/image)"/>
                <remap from="rgb1/camera_info"              to="$(arg rgb1/camera_info)"/>

                
                <remap from="proj_map" to="/map"/>

                
                <param name="Kp/MaxDepth"                  type="string"    value="4.0"/>      
                <param name="RGBD/AngularUpdate"           type="string"    value="0.1"/>    
                <param name="RGBD/LinearUpdate"            type="string"    value="0.1"/>                              
                <param name="RGBD/ProximityBySpace"        type="string"    value="true"/>   
                <param name="RGBD/OptimizeFromGraphEnd"    type="string"    value="false"/>  
                <param name="Reg/Strategy"                 type="string"    value="2"/>      
                <param name="Reg/Force3DoF"                type="string"    value="true"/>                   
                <param name="Icp/CoprrespondenceRatio"     type="string"    value="0.3"/>
                <param name="Vis/MinInliers"               type="string"    value="5"/>      
                <param name="Vis/InlierDistance"           type="string"    value="0.1"/>    
                <param name="Rtabmap/TimeThr"              type="string"    value="700"/>
                <param name="Optimizer/Slam2D"             type="string"    value="true"/>

[... 53 lines cut ...]
  
        <node if="$(arg rviz)" pkg="rviz" type="rviz" name="rviz" args="$(arg rviz_cfg)"/>
        <node if="$(arg rviz)" pkg="nodelet" type="nodelet" name="points_xyzrgb" args="standalone rtabmap_ros/point_cloud_xyzrgb">
                <remap from="rgb/image"         to="$(arg rgb_topic)"/>
                <remap from="depth/image"       to="$(arg depth_topic)"/>
                <remap from="rgb/camera_info"   to="$(arg camera_info_topic)"/>
                <remap from="cloud"             to="voxel_cloud" />

                <param name="decimation"  type="double" value="2"/>
                <param name="voxel_size"  type="double" value="0.02"/>
                <param name="approx_sync" type="bool"   value="$(arg approx_sync)"/>
        </node> 

        <node pkg="nodelet" type="nodelet" name="obstacles_detection" args="standalone rtabmap_ros/obstacles_detection">
                
                <remap from="cloud"             to="voxel_cloud" />                
                
                <remap from="obstacles"         to="voxel_obstacles" />
                <remap from="ground"            to="voxel_ground" />
                
                <param name="frame_id"                type="string" value="base_link"/>     
                <param name="wait_for_transform"      type="bool"   value="true"/>

        </node>  
</launch>

Severn-Vergil, 2016-11-27:
<launch>
  <arg name="database_path"               default="rtabmap.db"/>
  <arg name="localization"                default="false"/>
  <arg name="rgbd_odometry"               default="false"/>
  <arg name="args"                        default="--delete_db_on_start"/>
  <arg name="rtabmapviz"                  default="true"/>
  <arg name="rviz"                        default="false" />
  <arg name="gui_cfg"                     default="~/.ros/rtabmap_gui.ini" />
  <arg name="rviz_cfg"                    default="-d $(find rtabmap_ros)/launch/config/rgbd.rviz" />
  <arg name="approx_sync"                 default="true"/>               
  <arg name="wait_for_transform_duration" default="0.2"/>

  
  <arg name="depth_cameras"       default="2" />

  <arg name="rgb0/image"          default="/camera2/rgb/image_rect_color" />
  <arg name="depth0/image"        default="/camera2/depth_registered/image_raw" />
  <arg name="rgb0/camera_info"    default="/camera2/rgb/camera_info" />

  <arg name="rgb1/image"          default="/camera1/rgb/image_rect_color" />
  <arg name="depth1/image"        default="/camera1/depth_registered/image_raw" />
  <arg name="rgb1/camera_info"    default="/camera1/rgb/camera_info" />   
  
  <arg name="rgbd_namespace"          default="/camera"/>
  <arg name="rgb_topic"               default="$(arg rgbd_namespace)/rgb/image_rect_color" />
  <arg name="depth_topic"             default="$(arg rgbd_namespace)/depth_registered/image_raw" />
  <arg name="camera_info_topic"       default="$(arg rgbd_namespace)/rgb/camera_info" />
     
  
  <group ns="rtabmap">

        <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="$(arg args)">
                <param name="database_path"                 type="string"   value="$(arg database_path)"/>
                <param name="frame_id"                      type="string"   value="base_link"/>
                <param name="odom_frame_id"                 type="string"   value="odom"/>
                <param name="wait_for_transform_duration"   type="double"   value="$(arg wait_for_transform_duration)"/>
                <param name="subscribe_depth"               type="bool"     value="true"/>
                <param name="subscribe_scan"                type="bool"     value="false"/>
                <param name="gen_scan"                      type="bool"     value="true"/>
                <param name="depth_cameras"                 type="int"      value="$(arg depth_cameras)"/>
                
                <remap from="rgb0/image"                    to="$(arg rgb0/image)"/>
                <remap from="depth0/image"                  to="$(arg depth0/image)"/>
                <remap from="rgb0/camera_info"              to="$(arg rgb0/camera_info)"/>

                <remap from="rgb1/image"                    to="$(arg rgb1/image)"/>
                <remap from="depth1/image"                  to="$(arg depth1/image)"/>
                <remap from="rgb1/camera_info"              to="$(arg rgb1/camera_info)"/>

                
                <remap from="proj_map" to="/map"/>

                
                <param name="Kp/MaxDepth"                  type="string"    value="4.0"/>      
                <param name="RGBD/AngularUpdate"           type="string"    value="0.1"/>    
                <param name="RGBD/LinearUpdate"            type="string"    value="0.1"/>                              
                <param name="RGBD/ProximityBySpace"        type="string"    value="true"/>   
                <param name="RGBD/OptimizeFromGraphEnd"    type="string"    value="false"/>  
                <param name="Reg/Strategy"                 type="string"    value="2"/>      
                <param name="Reg/Force3DoF"                type="string"    value="true"/>                   
                <param name="Icp/CoprrespondenceRatio"     type="string"    value="0.3"/>
                <param name="Vis/MinInliers"               type="string"    value="5"/>      
                <param name="Vis/InlierDistance"           type="string"    value="0.1"/>    
                <param name="Rtabmap/TimeThr"              type="string"    value="700"/>
                <param name="Optimizer/Slam2D"             type="string"    value="true"/>

                <param name="Grid/MinClusterSize"          type="int"       value="20"/>
                <param name="Grid/ClusterRadius"           type="double"    value="0.05"/>
                <param name="Grid/MaxObstacleHeight"       type="double"    value="0.0"/>

[... 49 lines cut ...]
  
        <node if="$(arg rviz)" pkg="rviz" type="rviz" name="rviz" args="$(arg rviz_cfg)"/>
        <node if="$(arg rviz)" pkg="nodelet" type="nodelet" name="points_xyzrgb" args="standalone rtabmap_ros/point_cloud_xyzrgb">
                <remap from="rgb/image"         to="$(arg rgb_topic)"/>
                <remap from="depth/image"       to="$(arg depth_topic)"/>
                <remap from="rgb/camera_info"   to="$(arg camera_info_topic)"/>
                <remap from="cloud"             to="voxel_cloud" />

                <param name="decimation"  type="double" value="2"/>
                <param name="voxel_size"  type="double" value="0.02"/>
                <param name="approx_sync" type="bool"   value="$(arg approx_sync)"/>
        </node> 

        <node pkg="nodelet" type="nodelet" name="obstacles_detection" args="standalone rtabmap_ros/obstacles_detection">
                
                <remap from="cloud"             to="voxel_cloud" />                
                
                <remap from="obstacles"         to="voxel_obstacles" />
                <remap from="ground"            to="voxel_ground" />
                
                <param name="frame_id"                type="string" value="base_link"/>     
                <param name="wait_for_transform"      type="bool"   value="true"/>

        </node>  
</launch>

matlabbe, 2016-11-28:
Try [updating](https://github.com/introlab/rtabmap_ros#update-to-new-version) the code to latest version (0.11.12). I cannot reproduce your error with the [current](https://github.com/introlab/rtabmap_ros/blob/19eb9cb836c029f3319b48310cd04ac119d58903/launch/demo/demo_two_kinects.launch) demo launch file:
[image]

Severn-Vergil, 2016-11-29:
I'm using ROS indigo with Ubuntu 14.04, is it matter if I update the rtabmap code as the rtabmap_ros is the indigo version? Thank you for your advice.
I'm trying to study about the usage of the forum, Sorry to have troubled you!
http://official-rtab-map-forum.67519.x6.nabble.com/demo-two-kinects-launch-dumped-td2382.html

matlabbe, 2016-11-29:
Hi,

You will have to update both, rtabmap and rtabmap_ros packages. See https://github.com/introlab/rtabmap_ros#build-from-source, you can skip the optional dependencies step and go forward on rtabmap and rtabmap_ros. Make sure you uninstall the indigo binaries.

cheers,
Mathieu

Severn-Vergil, 2016-11-30:
Hi, Mathieu
Is it mean that I can not use ROS indigo version? The newest rtabmap_ros package is in kinetic branch.
Another question, is your newest dual Kinect mapping demo use both camera info in a single signature, instead of generate two separate signatures and binding them by their static camera transformation? 
It seems that you use Kinect 1414 and 1473 as one of the image data has more lightly red than another, isn't it?

Thank's a lot!
Severn

matlabbe, 2016-11-30:
Hi Severn,

No, the master branch can be built on Indigo/Jade/Kinetic. The [rosdistro]-devel branches are all 0.11.8 (indigo/jade/kinetic, 0.8.12 for hydro). kinetic_devel on rtabmap_ros has a small [patch](https://github.com/introlab/rtabmap_ros/commit/40439707fd7c09bc89595c1b432ff62f285ef2ba) on ros dependencies, but the code is the same as indigo-devel and jade-devel.

Only one signature is created from the two kinects. Their images/camera_info are synchronized into a single signature.

They are two Kinect for XBOX 360 1414. Not sure what is causing the red tint on the left camera, maybe related to auto exposure.

cheers

Severn-Vergil, 2016-12-01:
Thank you Mathieu, I'll modify it a.s.a.p.!

closed 2016-12-01 by Severn-Vergil