introlab/rtabmap_ros #135: TX1 rtabmap JacobiSVD Error

brude-93, 2016-12-02:
I'm currently running L4T 2.1 R23.2 (Ubuntu 14.04) on an NVIDIA Jetson TX1 with ROS Indigo. I'm trying to run rtabmap on this platform (all installed from the default ros_indigo_rtabmap_ros package) but I am receiving this error at runtime:

tx1: /usr/local/include/eigen3/Eigen/src/SVD/JacobiSVD.h:632: const MatrixVType& Eigen::JacobiSVD<MatrixType, QRPreconditioner>::matrixV() const [with _MatrixType = Eigen::Matrix<double, 3, 3>; int QRPreconditioner = 2; Eigen::JacobiSVD<MatrixType, QRPreconditioner>::MatrixVType = Eigen::Matrix<double, 3, 3>; typename _MatrixType::Scalar = double]: Assertion `computeV() && "This JacobiSVD decomposition didn't compute V. Did you ask for it?"' failed.

I've tried updating to the newest L4T 2.3 (Ubuntu 16.04), I've tried downloading Eigen3 from source and PCL 1.8 from source. None of these solutions have worked so far. Has anyone else tried running rtabmap on a TX1? Are there any other things I can try or another solution for this error?

Thanks!

matlabbe, 2016-12-02:
Hi,

It seems similar to this closed issue: https://github.com/introlab/rtabmap/issues/112

brude-93, 2016-12-02:
I've tried this as well. I do have the file in that location and I've tried installing Eigen from source. Still no luck :( I think it has something to do with compiling with C++11 enabled or with the architecture of the TX1.

matlabbe, 2016-12-02:
You can try without C++11. To do that you must [disable g2o, GTSAM, ZED and realsense support](https://github.com/introlab/rtabmap/blob/9276607920c8d593929a7e4c813e9ffc426b525b/CMakeLists.txt#L331):
```bash
$ cmake -DWITH_G2O=OFF -DWITH_ZED=OFF -DWITH_GTSAM=OFF -DWITH_REALSENSE=OFF ..
```
(At the of cmake status, `CMAKE_CXX_FLAGS`  should not have `-std=c++11`)

cheers,
Mathieu

brude-93, 2016-12-08:
Hello again! I found the solution for me. Again, running an NVIDIA TX1 with L4T 23.2 Ubuntu 14.04, ROS Indigo, OpenCV4Tegra 2.4, ZED SDK 1.0.0. To get this to work I had to do the following:

1. Install Eigen3.2.10 from source
2. Install PCL-1.7.1 from source
3. Install RTABMAP 0.11.12 from source (without ROS, just standalone source install)

After I installed all these packages from source, I re-installed the ZED SDK and I now am able to run everything. The free() issue I was dealing with was an issue with freeing a pointer to a buffer where I was gathering point cloud data. Thanks again for the help!

closed 2016-12-08 by brude-93