introlab/rtabmap_ros #136: exporting large map

owner01, 2016-12-09:
Hi,

I have a database of around 700 MB in size. 
When trying to export the 3D scan the application keeps crashing.
I can see it takes more then 6 GB rams and then crashes with segmentation fault.
My specs are i5 6200U with 8 gb Ram on ubuntu.
I can send you the file if needed.

matlabbe, 2016-12-09:
Hi,

What is the size of each scan? How many? What are the steps you do to export? Using the Export Scans dialog?
[image]

cheers,
Mathieu

owner01, 2016-12-09:
Hi Mathieu,

I am using the export 3D clouds dialog. There are around 3000 scans made. TBH I dont really know the size of each scan? Its the default settings using Kinect.

matlabbe, 2016-12-09:
Set the voxel size to 1 cm for example (or more), this will help to reduce a lot the size of the point clouds to assemble.

owner01, 2016-12-09:
Unfortunately when using the voxel size it will continue for longer but still crash in the end. I think the generated map is somewhat useable though.

matlabbe, 2016-12-09:
For very large map, you could go for 5 cm voxel size (similar size used for occupancy grids)