introlab/rtabmap_ros #138: rtabmap confused on orientation, drifts with map

natejgardner, 2016-12-20:
In localization mode, after turning the camera around once or twice, rtabmap totally loses orientation as if it's not localizing against a map at all. When I rotate, it's obvious to me rtabmap's frame is totally wrong, as if there's no relationship between `odom` and `map`. It looks like this in the rtabmap viewer. Examining in RViz, it's clear that the odometry drifts significantly in position and orientation. It's also as if there is no transform between the live scan and the map, even though rtabmap claims to be seeing features from the map and a few objects that were not in my original scan show as black, indicting rtabmap seems to be correctly localizing against known features. Yet everything is sideways.

How can I get rtabmap to consistently localize against a fixed map?

[image]
[image]

natejgardner, 2016-12-20:
With additional testing, rtabmap seems not to have any global idea of the map's coordinates. Doesn't make much sense. It loop closes, but it has no idea which direction it's facing.

matlabbe, 2016-12-22:
Are you using the same launch file as in this [tutorial](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping) (with command lines used)? Which I could use to try to reproduce the problem.

At first view, it seems related to `RGBD/OptimizeFromGraphEnd` parameter. If true, the map is transformed in current odometry frame, rotating the map if the camera was started tilted differently than when the map was first created (e.g., TF /map -> /odom will be always null). If false, the current odometry frame is transformed in map frame, so the map should look at the same pose as before, but the camera is moved/rotated accordingly to localization (e.g., TF /map -> /odom would be not null).

cheers,
Mathieu

tekomoro, 2020-07-16:
> Are you using the same launch file as in this [tutorial](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping) (with command lines used)? Which I could use to try to reproduce the problem.
>
> At [...]

i use zed camera can i change RGBD/OptimizeFromGraphEnd parameter with my synchronized camera or what can i do to improve lozalizing ?

closed 2020-07-29 by matlabbe