introlab/rtabmap_ros #139: Database error

QinZiwen, 2017-01-05:
$rtabmap-databaseViewer ~/.ros/rtabmap.db
Terminal show errors are as follow:
```
[ WARN] (2017-01-05 16:46:46.718) Parameters.cpp:728::readINI() Section "Core" in /home/turtlebot/.rtabmap/rtabmap.ini doesn't exist... Ignore this warning if the ini file does not exist yet. The ini file will be automatically created when this node will close.
[ INFO] (2017-01-05 16:46:46.791) DBDriverSqlite3.cpp:339::connectDatabaseQuery() Using database "/home/turtlebot/.ros/rtabmap.db" from the hard drive.
[ INFO] (2017-01-05 16:46:46.826) DBDriverSqlite3.cpp:374::connectDatabaseQuery() Database version = 0.11.13
[ERROR] (2017-01-05 16:46:46.826) DBDriverSqlite3.cpp:380::connectDatabaseQuery() Opened database version (0.11.13) is more recent than rtabmap installed version (0.11.8). Please update rtabmap to new version!
[ INFO] (2017-01-05 16:46:46.826) DBDriverSqlite3.cpp:421::disconnectDatabaseQuery() Disconnecting database /home/turtlebot/.ros/rtabmap.db...
[ INFO] (2017-01-05 16:46:51.468) main.cpp:54::main() All done! closing...
*** Error in `rtabmap-databaseViewer': free(): invalid pointer: 0x00007f4c0a78c978 ***
Aborted (core dumped)
```

How to update rtabmap to new version?

matlabbe, 2017-01-05:
I assume you are using the ROS binaries (0.11.8). You have to build from source to get 0.11.13. Follow instructions [here](https://github.com/introlab/rtabmap_ros#build-from-source) (you can skip optional dependencies section as you already installed once rtabmap).

QinZiwen, 2017-01-06:
Thanks @matlabbe .
The error reasons maybe is "rtabmap's apt-get source" being not update, that version still is 0.11.8. When I build this from source, database can open. But database viewer tell me that "The database is using 1 different parameter than those currently set in Core parameters panel. Do you want to use database's parameters?"

matlabbe, 2017-01-06:
That question is normal. After mapping, rtabmap saves in the database the latest parameters used. When you open an old database and that your current settings (GUI) are different, you will be asked to update them to match those in the database. Pressing No you keep the current settings and pressing Yes you update the settings accordingly to database (so you don't need to remember what were the parameters for a particular database). Generally I press Yes. You would see in the terminal which parameters have been updated.

QinZiwen, 2017-01-06:
Yes, thank you! @matlabbe

closed 2017-01-06 by matlabbe