introlab/rtabmap_ros #144: Installation error from the source using ubuntu 14.04

PeterPanxb, 2017-01-10:
Hi,
**When i installed rtabmap from the source,followed all the instructions for the rtabmap_ros install,i got something wrong,just as follows,**

```
`peter@peter:~/index_download/rtabmap/build$cmake ..
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- checking for module 'eigen3'
--   found eigen3, version 3.3.1
-- Found eigen: /usr/local/include/eigen3  
-- checking for module 'flann'
--   found flann, version 1.8.4
-- Found Flann: /usr/lib/x86_64-linux-gnu/libflann_cpp_s.a  
-- Found qhull: /usr/lib/x86_64-linux-gnu/libqhull.so  
-- checking for module 'libopenni'
--   found libopenni, version 1.5.4.0
-- Found openni: /usr/lib/libOpenNI.so  
-- checking for module 'libopenni2'
--   found libopenni2, version 2.2.0.3
-- Found OpenNI2: /usr/lib/libOpenNI2.so  
-- Found libusb-1.0: /usr/include  
-- Found PCL_COMMON: /usr/lib/libpcl_common.so  
-- Found PCL_KDTREE: /usr/lib/libpcl_kdtree.so  
-- Found PCL_OCTREE: /usr/lib/libpcl_octree.so  
-- Found PCL_SEARCH: /usr/lib/libpcl_search.so  
-- Found PCL_SURFACE: /usr/lib/libpcl_surface.so  
-- Found PCL_SAMPLE_CONSENSUS: /usr/lib/libpcl_sample_consensus.so  
-- Found PCL_IO: /usr/lib/libpcl_io.so  
-- Found PCL_FILTERS: /usr/lib/libpcl_filters.so  
-- Found PCL_FEATURES: /usr/lib/libpcl_features.so  
-- Found PCL_GEOMETRY: /usr/include/pcl-1.7  
-- Found PCL_KEYPOINTS: /usr/lib/libpcl_keypoints.so  
-- Found PCL_REGISTRATION: /usr/lib/libpcl_registration.so  
-- Found PCL_SEGMENTATION: /usr/lib/libpcl_segmentation.so  
-- Found PCL_RECOGNITION: /usr/lib/libpcl_recognition.so  
-- Found PCL_VISUALIZATION: /usr/lib/libpcl_visualization.so  
-- Found PCL_PEOPLE: /usr/lib/libpcl_people.so  
-- Found PCL_OUTOFCORE: /usr/lib/libpcl_outofcore.so  
-- Found PCL_TRACKING: /usr/lib/libpcl_tracking.so  
-- Found PCL_APPS: /usr/lib/libpcl_apps.so  
-- Try OpenMP C flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Try OpenMP CXX flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Found OpenMP: -fopenmp  
-- Found OpenMP
-- Found OpenCV: /usr/local/include/opencv;/usr/local/include
-- Found PCL: /usr/include/pcl-1.7;/usr/local/include/eigen3;/usr/include;/usr/include/ni;/usr/include/openni2;/usr/include/vtk-5.8
-- Found ZLIB: /usr/include
-- Looking for Q_WS_X11
-- Looking for Q_WS_X11 - found
-- Looking for Q_WS_WIN
-- Looking for Q_WS_WIN - not found
-- Looking for Q_WS_QWS
-- Looking for Q_WS_QWS - not found
-- Looking for Q_WS_MAC
-- Looking for Q_WS_MAC - not found
-- Found Qt4: /usr/bin/qmake (found version "4.8.6") 
-- Found QVTK: /usr/lib/libQVTK.so  
-- Found Freenect: /usr/include
[... 101 lines cut ...]
  TARGET_LINK_LIBRARIES(csparse_extension csparse)
ELSE()
  TARGET_LINK_LIBRARIES(csparse_extension ${CSPARSE_LIBRARY})
ENDIF()

ADD_LIBRARY(solver_csparse${G2O_LIB_TYPE}
  solver_csparse.cpp
  linear_solver_csparse.h
  g2o_csparse_api.h
)
SET_TARGET_PROPERTIES(solver_csparse PROPERTIES OUTPUT_NAME ${LIB_PREFIX}solver_csparse)

TARGET_LINK_LIBRARIES(solver_csparse csparse_extension core)

INSTALL(TARGETS solver_csparse csparse_extension
  RUNTIME DESTINATION ${CMAKE_INSTALL_PREFIX}/bin
  LIBRARY DESTINATION ${CMAKE_INSTALL_PREFIX}/lib
  ARCHIVE DESTINATION ${CMAKE_INSTALL_PREFIX}/lib
)
FILE(GLOB headers "${CMAKE_CURRENT_SOURCE_DIR}/*.h" "${CMAKE_CURRENT_SOURCE_DIR}/*.hpp")
INSTALL(FILES ${headers} DESTINATION ${CMAKE_INSTALL_PREFIX}/include/g2o/solvers/csparse)
```

**So, i think the library file "libcsparse_extension.so" should contain the functions. However  the library file "libcsparse_extension.so" is found when make the program(we can see the message showed from "make VERBOSE=1"),but there still has error message .I also use the latest g2o from https://github.com/RainerKuemmerle/g2o.  So it really fused me .I'm not sure what wrong here,wish your help.**
Thanks a lot.

matlabbe, 2017-01-10:
Hi,

On my computer, csparse seems already installed, so g2o doesn't use its own version:
```bash
(rtabmap cmake log)
-- Found CSPARSE: /usr/include/suitesparse  
-- Found g2o: /usr/local/include;/usr/include/suitesparse;/usr/include/suitesparse
```
Try install `libsuitesparse-dev` *before* building g2o:
```bash
sudo apt-get install libsuitesparse-dev
```

Other info:
```bash
$ ldd rtabmap | grep sparse
	libg2o_csparse_extension.so => /usr/local/lib/libg2o_csparse_extension.so (0x00007fa2c49d2000)
	libcxsparse.so.3.1.2 => /usr/lib/x86_64-linux-gnu/libcxsparse.so.3.1.2 (0x00007fa2c47a8000)
```

cheers,
Mathieu

PeterPanxb, 2017-01-11:
Hi, 
 Thanks very much for your reply. I tried with the method just like you said.But there are still have the problem. Here are the steps which i tried.
First, i try to unload the g2o library, but i can't get any informations about it.
```
peter@peter:~$ dpkg -l | grep g2o
peter@peter:~$ 

```
I tried "sudo apt-get install libsuitesparse-dev",the terminal shows 
```
peter@peter:~$ sudo apt-get install libsuitesparse-dev
[sudo] password for peter: 
正在读取软件包列表... 完成
正在分析软件包的依赖关系树       
正在读取状态信息... 完成       
libsuitesparse-dev 已经是最新的版本。
下列软件包是自动安装的并且现在不需要了：
  beignet libntdb1 python-ntdb ros-indigo-costmap-2d
  ros-indigo-create-dashboard ros-indigo-create-driver ros-indigo-create-node
  ros-indigo-move-base-msgs ros-indigo-voxel-grid
Use 'apt-get autoremove' to remove them.
升级了 0 个软件包，新安装了 0 个软件包，要卸载 0 个软件包，有 89 个软件包未被升级。

```
And then ,i reinstalled the g2o,here is the steps:
```
peter@peter:~/g2o/build$ cmake ../
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Compiling on Unix
-- Found CHOLMOD: /usr/include/suitesparse  
-- Looking for sgemm_
-- Looking for sgemm_ - found
-- A library with BLAS API found.
-- Looking for cheev_
-- Looking for cheev_ - found
-- A library with LAPACK API found.
-- Found CHOLMOD and its dependencies
-- Found CSPARSE: /usr/include/suitesparse  
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/x86_64-linux-gnu/libX11.so
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Compiling with OpenGL support
-- Looking for Q_WS_X11
-- Looking for Q_WS_X11 - found
-- Looking for Q_WS_WIN
-- Looking for Q_WS_WIN - not found
-- Looking for Q_WS_QWS
-- Looking for Q_WS_QWS - not found
-- Looking for Q_WS_MAC
-- Looking for Q_WS_MAC - not found
[... 115 lines cut ...]
--   With g2o                  = YES (License: BSD)
--   With GTSAM                = YES (License: BSD)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = YES (License: GPLv2)
--   With ZED                  = NO (ZED sdk not found)
--   With RealSense            = NO (librealsense not found)
--   With OCTOMAP              = YES (License: BSD)
--   With Qt4                  = YES (License: Open Source or Commercial)
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/peter/index_download/rtabmap/build
```
So, i get the same error like before.
In your reply, i know there already have csparse installed.When i run "sudo apt-get install libsuitesparse-dev",it already installed.Even if i rebuild "g2o",there seems like nothing changed.
Is there anything i did wrong or i forgot.
Beside,when i run "ldd rtabmap | grep sparse",there shows
```
peter@peter:~/index_download/rtabmap/build$ ldd rtabmap | grep sparse
ldd: ./rtabmap: 没有那个文件或目录

```
Should i remove g2o before i rebuild g2o.Is it  possible this reason caused  this problem?And how to remove "g2o" from my computer.

Thanks

matlabbe, 2017-01-12:
Did you try building with the g2o version from ROS? http://wiki.ros.org/libg2o
```
$ sudo apt-get install ros-indigo-libg2o
```

PeterPanxb, 2017-01-13:
Thanks very much for your reply. I have tried building with the g2o version just like you said and it installed a package.However when i try  building  rtabmap ,the wrong message appear again.
```
eter@peter:~$ sudo apt-get install ros-indigo-libg2o
[sudo] password for peter: 
正在读取软件包列表... 完成
正在分析软件包的依赖关系树       
正在读取状态信息... 完成       
下列软件包是自动安装的并且现在不需要了：
  beignet libntdb1 python-ntdb ros-indigo-costmap-2d
  ros-indigo-create-dashboard ros-indigo-create-driver ros-indigo-create-node
  ros-indigo-move-base-msgs ros-indigo-voxel-grid
Use 'apt-get autoremove' to remove them.
下列【新】软件包将被安装：
  ros-indigo-libg2o
升级了 0 个软件包，新安装了 1 个软件包，要卸载 0 个软件包，有 99 个软件包未被升级。
需要下载 879 kB 的软件包。
解压缩后会消耗掉 5,564 kB 的额外空间。
获取：1 http://packages.ros.org/ros/ubuntu/ trusty/main ros-indigo-libg2o amd64 2014.2.18-0trusty-20160321-175501-0700 [879 kB]
下载 879 kB，耗时 2秒 (330 kB/s)            
正在选中未选择的软件包 ros-indigo-libg2o。
(正在读取数据库 ... 系统当前共安装有 272269 个文件和目录。)
正准备解包 .../ros-indigo-libg2o_2014.2.18-0trusty-20160321-175501-0700_amd64.deb  ...
正在解包 ros-indigo-libg2o (2014.2.18-0trusty-20160321-175501-0700) ...
正在设置 ros-indigo-libg2o (2014.2.18-0trusty-20160321-175501-0700) ...

```
However when i try  building  rtabmap ,the wrong message appear again.

```
peter@peter:~/index_download/rtabmap/build$ make -j4
Scanning dependencies of target rtabmap_utilite
[  0%] [  1%] [  1%] [  2%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UEventsManager.cpp.o
Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UEventsHandler.cpp.o
Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UEventsSender.cpp.o
Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UFile.cpp.o
[  2%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UDirectory.cpp.o
[  3%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UConversion.cpp.o
[  3%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/ULogger.cpp.o
[  4%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UThread.cpp.o
[  4%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UTimer.cpp.o
[  5%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UProcessInfo.cpp.o
[  5%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UVariant.cpp.o
Linking CXX shared library ../../../bin/librtabmap_utilite.so
[  5%] Built target rtabmap_utilite
Scanning dependencies of target extractObject
Scanning dependencies of target res_tool
Scanning dependencies of target imagesJoiner
[  5%] Building CXX object utilite/resource_generator/CMakeFiles/res_tool.dir/main.cpp.o
[  6%] Building CXX object tools/ImagesJoiner/CMakeFiles/imagesJoiner.dir/main.cpp.o
[  6%] Building CXX object tools/ExtractObject/CMakeFiles/extractObject.dir/main.cpp.o
Linking CXX executable ../../../bin/rtabmap-res_tool
[  6%] Built target res_tool
[  7%] [Creating resources]
 Using namespace=rtabmap
 Using target directory=/home/peter/index_download/rtabmap/build/corelib/src
Input file "/home/peter/index_download/rtabmap/corelib/src/resources/DatabaseSchema.sql" size = 4208 bytes
Output file "/home/peter/index_download/rtabmap/build/corelib/src/DatabaseSchema_sql.h" size = 8637 bytes
Scanning dependencies of target rtabmap_core
Linking CXX executable ../../../bin/rtabmap-imagesJoiner
[  7%] Built target imagesJoiner
[  7%] [  8%] [  8%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Rtabmap.cpp.o
Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/RtabmapThread.cpp.o
Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Statistics.cpp.o
[  9%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Memory.cpp.o
[  9%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/DBDriver.cpp.o
[ 10%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/DBDriverSqlite3.cpp.o
[ 10%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/DBReader.cpp.o
[ 11%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Camera.cpp.o
[ 11%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/CameraThread.cpp.o
[ 12%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/CameraRGB.cpp.o
[... 175 lines cut ...]
[ 81%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/utilite/moc_UPlot.cxx.o
[ 82%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/utilite/moc_UImageView.cxx.o
[ 82%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_CloudViewer.cxx.o
[ 83%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_OdometryViewer.cxx.o
[ 83%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_LoopClosureViewer.cxx.o
[ 84%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_DataRecorder.cxx.o
[ 84%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_CameraViewer.cxx.o
[ 85%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_CalibrationDialog.cxx.o
[ 85%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/moc_ExportDialog.cxx.o
[ 86%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/moc_PostProcessingDialog.cxx.o
[ 86%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/moc_ExportCloudsDialog.cxx.o
[ 87%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/moc_ExportScansDialog.cxx.o
[... cut ...]
What's more,if i change the order of  " cxsparse "  " g2o_ext_csparse" in the fourth line of "FindG2O.cmake" (the sentence is "FIND_LIBRARY(CSPARSE_LIBRARY NAMES cxsparse g2o_ext_csparse"),they will show different wrong message. If put "cxsparse" in front of  " g2o_ext_csparse" , the error i get is the same as  https://github.com/introlab/rtabmap/issues/154 .But if i let  "cxsparse" behind " g2o_ext_csparse",the error shows just like above.So,is it possible the wrong path lead to this problem? I try to check up the path of g2o library, but i get nothing for there are too many storage path.What should i do ?Thanks a lot.

matlabbe, 2017-01-13:
Cleanup the build directory (at least the CMake Cache) before remaking rtabmap:
```bash
$ cd rtabmap/build/
$ rm CMakeCache.txt
$ cmake ..
```

PeterPanxb, 2017-01-13:
Everytime i try build rtabmap, i usually delete the rtabmap files. And redownload from the github. So this maybe not work.

matlabbe, 2017-01-13:
What does CMake tells about which g2o/csparse found? The "Found CSPARSE" and "Found g2o" lines. Make sure you don't have remaining g2o installation in /usr/local.

PeterPanxb, 2017-01-14:
Hi,
  Good news, i found the way to solve this question according to your suggestion.It is the wrong path caused this question.Many thanks for your help.Thank you.