introlab/rtabmap_ros #145: catkin_make error

QinZiwen, 2017-01-16:
I am use "ros-indigo", when I run "catkin_make",some error show are as follow:
```
/home/turtlebot/catkin_ws/src/rtabmap_ros/src/nodelets/rgbd_sync.cpp: In member function ‘void rtabmap_ros::RGBDSync::callback(const ImageConstPtr&, const ImageConstPtr&, const CameraInfoConstPtr&)’:
/home/turtlebot/catkin_ws/src/rtabmap_ros/src/nodelets/rgbd_sync.cpp:128:15: error: ‘const class cv_bridge::CvImage’ has no member named ‘toCompressedImageMsg’
     imagePtr->toCompressedImageMsg(msgCompressed.rgbCompressed, cv_bridge::JPG);
               ^
/home/turtlebot/catkin_ws/src/rtabmap_ros/src/nodelets/rgbd_sync.cpp:128:65: error: ‘JPG’ is not a member of ‘cv_bridge’
     imagePtr->toCompressedImageMsg(msgCompressed.rgbCompressed, cv_bridge::JPG);
                                                                 ^
make[2]: *** [rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/nodelets/rgbd_sync.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/turtlebot/catkin_ws/src/rtabmap_ros/src/MsgConversion.cpp: In function ‘void rtabmap_ros::toCvCopy(const RGBDImage&, cv_bridge::CvImagePtr&, cv_bridge::CvImagePtr&)’:
/home/turtlebot/catkin_ws/src/rtabmap_ros/src/MsgConversion.cpp:129:48: error: no matching function for call to ‘toCvCopy(const _rgbCompressed_type&)’
   rgb = cv_bridge::toCvCopy(image.rgbCompressed);
                                                ^
/home/turtlebot/catkin_ws/src/rtabmap_ros/src/MsgConversion.cpp:129:48: note: candidates are:
In file included from /home/turtlebot/catkin_ws/src/rtabmap_ros/include/rtabmap_ros/MsgConversion.h:41:0,
                 from /home/turtlebot/catkin_ws/src/rtabmap_ros/src/MsgConversion.cpp:28:
/opt/ros/indigo/include/cv_bridge/cv_bridge.h:126:12: note: cv_bridge::CvImagePtr cv_bridge::toCvCopy(const ImageConstPtr&, const string&)
 CvImagePtr toCvCopy(const sensor_msgs::ImageConstPtr& source,
            ^
/opt/ros/indigo/include/cv_bridge/cv_bridge.h:126:12: note:   no known conversion for argument 1 from ‘const _rgbCompressed_type {aka const sensor_msgs::CompressedImage_<std::allocator<void> >}’ to ‘const ImageConstPtr& {aka const boost::shared_ptr<const sensor_msgs::Image_<std::allocator<void> > >&}’
/opt/ros/indigo/include/cv_bridge/cv_bridge.h:148:12: note: cv_bridge::CvImagePtr cv_bridge::toCvCopy(const Image&, const string&)
 CvImagePtr toCvCopy(const sensor_msgs::Image& source,
            ^
/opt/ros/indigo/include/cv_bridge/cv_bridge.h:148:12: note:   no known conversion for argument 1 from ‘const _rgbCompressed_type {aka const sensor_msgs::CompressedImage_<std::allocator<void> >}’ to ‘const Image& {aka const sensor_msgs::Image_<std::allocator<void> >&}’
/home/turtlebot/catkin_ws/src/rtabmap_ros/src/MsgConversion.cpp: In function ‘void rtabmap_ros::toCvShare(const RGBDImageConstPtr&, cv_bridge::CvImageConstPtr&, cv_bridge::CvImageConstPtr&)’:
/home/turtlebot/catkin_ws/src/rtabmap_ros/src/MsgConversion.cpp:165:49: error: no matching function for call to ‘toCvCopy(const _rgbCompressed_type&)’
   rgb = cv_bridge::toCvCopy(image->rgbCompressed);
                                                 ^
/home/turtlebot/catkin_ws/src/rtabmap_ros/src/MsgConversion.cpp:165:49: note: candidates are:
In file included from /home/turtlebot/catkin_ws/src/rtabmap_ros/include/rtabmap_ros/MsgConversion.h:41:0,
                 from /home/turtlebot/catkin_ws/src/rtabmap_ros/src/MsgConversion.cpp:28:
/opt/ros/indigo/include/cv_bridge/cv_bridge.h:126:12: note: cv_bridge::CvImagePtr cv_bridge::toCvCopy(const ImageConstPtr&, const string&)
 CvImagePtr toCvCopy(const sensor_msgs::ImageConstPtr& source,
            ^
/opt/ros/indigo/include/cv_bridge/cv_bridge.h:126:12: note:   no known conversion for argument 1 from ‘const _rgbCompressed_type {aka const sensor_msgs::CompressedImage_<std::allocator<void> >}’ to ‘const ImageConstPtr& {aka const boost::shared_ptr<const sensor_msgs::Image_<std::allocator<void> > >&}’
/opt/ros/indigo/include/cv_bridge/cv_bridge.h:148:12: note: cv_bridge::CvImagePtr cv_bridge::toCvCopy(const Image&, const string&)
 CvImagePtr toCvCopy(const sensor_msgs::Image& source,
            ^
/opt/ros/indigo/include/cv_bridge/cv_bridge.h:148:12: note:   no known conversion for argument 1 from ‘const _rgbCompressed_type {aka const sensor_msgs::CompressedImage_<std::allocator<void> >}’ to ‘const Image& {aka const sensor_msgs::Image_<std::allocator<void> >&}’
make[2]: *** [rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/MsgConversion.cpp.o] Error 1
make[1]: *** [rtabmap_ros/CMakeFiles/rtabmap_ros.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
```
I don't know how to do?

matlabbe, 2017-01-16:
Hi,

CvImage::toCompressedImageMsg is defined in latest cv_bridge indigo version: https://github.com/ros-perception/vision_opencv/blob/indigo/cv_bridge/include/cv_bridge/cv_bridge.h#L111. Make sure cv_bridge is updated on your computer, or rebuild cv_bridge from source to have the missing function.

cheers

yalan2017, 2017-03-03:
Hi
after catkin_make command i have this error reinstalled my ros but dose not work yet ! i use ubuntu 14-indigo.
thank you

  
Built target kinect2_registration
Built target geometry_msgs_generate_messages_lisp
[  2%] Built target sensor_msgs_generate_messages_lisp
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_MapGraph
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_Point3f
[  2%] [  2%] Built target _rtabmap_ros_generate_messages_check_deps_SetGoal
Built target _rtabmap_ros_generate_messages_check_deps_ListLabels
[  2%] [  2%] [  2%] Built target _rtabmap_ros_generate_messages_check_deps_Point2f
Built target _rtabmap_ros_generate_messages_check_deps_GetMap
Built target _rtabmap_ros_generate_messages_check_deps_Goal
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_Info
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_PublishMap
[  2%] [  2%] Built target _rtabmap_ros_generate_messages_check_deps_Link
Built target _rtabmap_ros_generate_messages_check_deps_MapData
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_SetLabel
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_ResetPose
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_NodeData
[  2%] [  3%] Built target _rtabmap_ros_generate_messages_check_deps_OdomInfo
Built target rtabmap_ros_gencfg
[  3%] Built target std_msgs_generate_messages_py
[  3%] [  3%] Built target _rtabmap_ros_generate_messages_check_deps_UserData
Built target geometry_msgs_generate_messages_py
[  3%] [  3%] Built target sensor_msgs_generate_messages_py
Built target geometry_msgs_generate_messages_cpp
[  3%] [  3%] [  3%] [  3%] Built target std_msgs_generate_messages_cpp
Built target _rtabmap_ros_generate_messages_check_deps_KeyPoint
Built target _rtabmap_ros_generate_messages_check_deps_RGBDImage
Built target sensor_msgs_generate_messages_cpp
[  4%] Built target icp_odometry
[  5%] Built target rgbd_odometry
[  6%] Built target rgbdicp_odometry
[  7%] Built target kinect2_bridge
[  8%] Built target rtabmap
[  9%] Built target stereo_odometry
[ 10%] [ 10%] Built target kinect2_bridge_nodelet
Built target kinect2_calibration
[ 38%] [ 42%] [ 58%] Built target rtabmap_ros_generate_messages_py
Built target rtabmap_ros_generate_messages_lisp
Built target rtabmap_ros_generate_messages_cpp
[ 58%] Built target rtabmap_ros_generate_messages
[ 59%] Built target kinect2_viewer
[ 59%] Built target camera
[ 60%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/MapsManager.cpp.o
/home/cyber/catkin_ws/src/rtabmap_ros/src/MapsManager.cpp: In member function ‘void MapsManager::backwardCompatibilityParameters(ros::NodeHandle&, rtabmap::ParametersMap&) const’:
/home/cyber/catkin_ws/src/rtabmap_ros/src/MapsManager.cpp:280:52: error: ‘kGrid3DGroundIsObstacle’ is not a member of ‘rtabmap::Parameters’
  parameterMoved(pnh, "octomap_ground_is_obstacle", Parameters::kGrid3DGroundIsObstacle(), parameters);
                                                    ^
make[2]: *** [rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/MapsManager.cpp.o] Error 1
make[1]: *** [rtabmap_ros/CMakeFiles/rtabmap_ros.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

matlabbe, 2017-03-03:
This error was introduced this week after a this commit https://github.com/introlab/rtabmap/commit/17121d5861301fd2e4e885dcc37c4b29997a9417 on the rtabmap standalone library. I synchronized rtabmap and rtabmap_ros to version 0.12.2. This should be fixed by https://github.com/introlab/rtabmap_ros/commit/a264275404210acd175937912fda9c10f47d882c. 

cheers,
Mathieu

yalan2017, 2017-03-10:
thank you, its working now ,
cheers

closed 2017-03-10 by matlabbe