introlab/rtabmap_ros #146: rtabmap_ros fails after start due to 'UException' error in CameraModel.cpp

miscott, 2017-01-21:
Hello,

I use the MORSE simulator with ros to help with my robotics research. I would like to use rtabmap to create maps on a robot which has stereo camera equipped for navigation purposes. 
I created a launch file similar to the outdoor_mapping example:

```
<launch>
  
   
   <!-- Choose visualization -->
   <arg name="rviz" default="true" />
   <arg name="rtabmapviz" default="false" />
    
   <param name="use_sim_time" type="bool" value="true"/>
   <param name="approx_sync" type="bool" value="true"/>
   
   <!-- Just to uncompress images for stereo_image_rect -->
   <node name="republish_left"  type="republish" pkg="image_transport" args="raw in:=/stereo_camera/left/image_raw_throttle/image raw out:=/stereo_camera/left/image_raw_throttle_relay" />
   <node name="republish_right" type="republish" pkg="image_transport" args="raw in:=/stereo_camera/right/image_raw_throttle/image raw out:=/stereo_camera/right/image_raw_throttle_relay" />

   <!-- Run the ROS package stereo_image_proc for image rectification -->
   <group ns="/stereo_camera" >
      <node pkg="nodelet" type="nodelet" name="stereo_nodelet"  args="manager"/>
   
      <node pkg="stereo_image_proc" type="stereo_image_proc" name="stereo_image_proc">
         <remap from="left/image_raw"    to="left/image_raw_throttle_relay"/>
         <remap from="left/camera_info"  to="left/image_raw_throttle/camera_info"/>
         <remap from="right/image_raw"   to="right/image_raw_throttle_relay"/>
         <remap from="right/camera_info" to="right/image_raw_throttle/camera_info"/>
         <param name="disparity_range" value="128"/>
      </node>
   </group>
   <node name="republish_left2"  type="republish" pkg="image_transport" args="raw in:=/left_rgb raw out:=/left_rgb_trans" />
   <node name="republish_right2" type="republish" pkg="image_transport" args="raw in:=/right_rgb raw out:=/right_rgb_trans" />

   <group ns="rtabmap">   
      <!-- Visual SLAM: args: "delete_db_on_start" and "udebug" -->
      <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="--delete_db_on_start">
         <!--param name="wait_for_transform_duration" type="double" value="15" /-->
         <param name="frame_id" type="string" value="base_footprint"/>
         <param name="subscribe_stereo" type="bool" value="true"/>
         <param name="subscribe_depth" type="bool" value="false"/>
         <param name="approx_sync"   type="bool"   value="true"/>

         <remap from="left/image_rect" to="/left_rgb"/>
         <remap from="right/image_rect" to="/right_rgb"/>
         <remap from="left/camera_info" to="/stereo_camera/left/image_raw_throttle/camera_info"/>
         <remap from="right/camera_info" to="/stereo_camera/right/image_raw_throttle/camera_info"/>

         <remap from="odom" to="/odom"/>

         <param name="queue_size" type="int" value="100"/>

         <!-- RTAB-Map's parameters -->
         <param name="Rtabmap/TimeThr" type="string" value="700"/>
         <param name="Rtabmap/DetectionRate" type="string" value="1"/>
         
         <param name="Kp/MaxFeatures" type="string" value="200"/>
         <param name="Kp/RoiRatios" type="string" value="0.03 0.03 0.04 0.04"/>
         <param name="Kp/DetectorStrategy" type="string" value="0"/>   <!-- use SURF -->
         <param name="Kp/NNStrategy" type="string" value="1"/>         <!-- kdTree -->

         <param name="SURF/HessianThreshold" type="string" value="1000"/>

         <param name="Vis/MinInliers" type="string" value="10"/>
         <param name="Vis/EstimationType" type="string" value="1"/> <!-- 3D->2D (PnP) -->

         <param name="RGBD/LoopClosureReextractFeatures" type="string" value="true"/>
         <param name="Vis/MaxFeatures" type="string" value="500"/>
         <param name="Vis/MaxDepth" type="string" value="10"/> 
      </node>

      
      <!-- Visualisation RTAB-Map -->
      <node if="$(arg rtabmapviz)" pkg="rtabmap_ros" type="rtabmapviz" name="rtabmapviz" args="-d $(find rtabmap_ros)/launch/config/rgbd_gui.ini" output="screen">
         <param name="approx_sync"   type="bool"   value="true"/>
         <param name="subscribe_stereo" type="bool" value="true"/>
         <param name="subscribe_odom_info" type="bool" value="false"/>
         <param name="queue_size" type="int" value="10"/>
         <param name="frame_id" type="string" value="base_footprint"/>
         <remap from="left/image_rect" to="/left_rgb"/>
         <remap from="right/image_rect" to="/right_rgb"/>
         <remap from="left/camera_info" to="/stereo_camera/left/camera_info_throttle"/>
         <remap from="right/camera_info" to="/stereo_camera/right/camera_info_throttle"/>
         <!--remap from="odom_info" to="/odom_info"/-->
         <remap from="odom" to="/odom"/>
         <remap from="mapData" to="mapData"/>
      </node>
         
   </group>
  
   <!-- Visualisation RVIZ --> 
   <node if="$(arg rviz)" pkg="rviz" type="rviz" name="rviz" args="-d $(find rtabmap_ros)/launch/config/demo_stereo_outdoor.rviz"/>

</launch>
```
I then start the simulation along with the necessary transformations and hope that the node will then work successfully. The problem is that this error keeps occurring:

[image]

I am wondering about the meaning of this error? All of my topics are publishing correctly.
My theory is that my camera_info topics may be off because I had to transform the image from an rgba encoding to a rgd using cv_bridge... but my attempts to solve that issue have all failed thus far.

Thanks in advance!

matlabbe, 2017-01-27:
Hi,

Can you show the camera_info topic?
```
$ rostopic echo /stereo_camera/left/camera_info 
$ rostopic echo /stereo_camera/right/camera_info
(well, adapt with your actual camera_info topics)
```
The error says that the Distortion matrix in camera_info is not in the format used in OpenCV. Note that you don't need republish nodes on the robot, you can connect your topics directly.

cheers

miscott, 2017-01-27:
Thanks for getting back to me. 
I actually just solved the initial issue earlier today, but it wasn't a total fix. The camera info topics were not publishing proper data like I had thought, so I wrote a new publisher to publish the corrected information format and the initial condition was taken care of.

But! Due to that I have a new error, also surrounding the info topics. I keep getting this warning when I run rtabmap_ros:
[image]
Once again, the calibration parameters must be off but I do not know how to update them in my Blender virtual world. This warning is leading to problems in my mapping. For example, here is simple Blender scene:
[image]
And here is the RVIZ view:
[image]
For some reason, rtabmap thinks all the obstacles are above the UAS when in reality they are parallel. 
So I guess my question now is: is there any advice for getting the proper calibration parameters for rtabmap_ros in a virtual world where ROS calibration packages won't work?

Here are my camera information topics:
right camera:
[image]

left camera:
[image]

Thanks again, and if this is question is now beyond the scope of rtabmap_ros I can search elsewhere.

Mitchell

matlabbe, 2017-01-28:
The "a large number of stereo correspondences are rejected" warning can be ignored, though ~50 features is very low. You may need to add more visual textures to your simulated environment.

For the map orientation, it should be a TF problem. Normally for cameras, we have an optical rotation to transform projected points from the camera frame to robot frame.  An example of rotation is shown on the [stereo tutorial here](http://wiki.ros.org/rtabmap_ros/Tutorials/StereoHandHeldMapping) (see the fourth step in **2.Bring-up the camera** section).

cheers,
Mathieu

miscott, 2017-01-28:
That sounds great I'll double check all that, thanks so much!

closed 2017-01-28 by miscott