introlab/rtabmap_ros #147: CameraModel.cpp:77::CameraModel() Condition (fx > 0.0) not met! [fx=0.000000] Error

robotic-enthusiast, 2017-02-06:
Hello,

I am using an Astra Pro Camera. Since RGB is not supported for OpenNi for the Pro version, I am using the Astra SDK for the Depth, and the usb_cam_node for the RGB camera:
https://github.com/orbbec/ros_astra_camera
https://github.com/bosch-ros-pkg/usb_cam

I made sure to change the topics for the rtabmap launch, but I am getting the following error:

```
/rtabmap/rgbd_odometry subscribed to (approx sync):
   /usb_cam/image_raw,
   /camera/depth_registered/image_raw,
   /usb_cam/camera_info
terminate called after throwing an instance of 'UException'
  what():  [FATAL] (2017-02-06 16:37:20.365) CameraModel.cpp:77::CameraModel() Condition (fx > 0.0) not met! [fx=0.000000]
```

Followed by the following:
`[ WARN] [1486366640.733577972]: odometry: Could not get transform from camera_link to head_camera (stamp=1486366640.510289) after 0.200000 seconds ("wait_for_transform_duration"=0.200000)! Error=". canTransform returned after 0.201911 timeout was 0.2."`

I tried creating a calibration file, but it seems that if I go to Window->Preferences->Source it doesn't allow me to change any of the parameters (everything is dark).

My launch commands are as follows (none of them work):
```
#rviz
roslaunch rtabmap_ros rgbd_mapping.launch rtabmap_args:="--delete_db_on_start" rviz:=true rtabmapviz:=false rgb_topic:=/usb_cam/image_raw depth_topic:=/camera/depth_registered/image_raw camera_info_topic:=/usb_cam/camera_info

#rtabmapviz
roslaunch rtabmap_ros rgbd_mapping.launch rtabmap_args:="--delete_db_on_start" rgb_topic:=/usb_cam/image_raw depth_topic:=/camera/depth_registered/image_raw camera_info_topic:=/usb_cam/camera_info
```

Does anyone have any idea what might be happening?

Thank you!

matlabbe, 2017-02-06:
Hi,

What is inside "/usb_cam/camera_info" ? 
```
$ rostopic echo /usb_cam/camera_info
```
Normally it should have the intrinsic parameters set (your fx>0.0 assert error). 

What are the topics published by ros_astra_camera node?

cheers

robotic-enthusiast, 2017-02-07:
Dear matlabbe,

Thank you for your answer!
The topics from the ros astra camera node are:

`/camera/camera_nodelet_manager/bond
/camera/depth/camera_info
/camera/depth/image
/camera/depth/image/compressed
/camera/depth/image/compressed/parameter_descriptions
/camera/depth/image/compressed/parameter_updates
/camera/depth/image/compressedDepth
/camera/depth/image/compressedDepth/parameter_descriptions
/camera/depth/image/compressedDepth/parameter_updates
/camera/depth/image/theora
/camera/depth/image/theora/parameter_descriptions
/camera/depth/image/theora/parameter_updates
/camera/depth/image_raw
/camera/depth/image_raw/compressed
/camera/depth/image_raw/compressed/parameter_descriptions
/camera/depth/image_raw/compressed/parameter_updates
/camera/depth/image_raw/compressedDepth
/camera/depth/image_raw/compressedDepth/parameter_descriptions
/camera/depth/image_raw/compressedDepth/parameter_updates
/camera/depth/image_raw/theora
/camera/depth/image_raw/theora/parameter_descriptions
/camera/depth/image_raw/theora/parameter_updates
/camera/depth/image_rect
/camera/depth/image_rect/compressed
/camera/depth/image_rect/compressed/parameter_descriptions
/camera/depth/image_rect/compressed/parameter_updates
/camera/depth/image_rect/compressedDepth
/camera/depth/image_rect/compressedDepth/parameter_descriptions
/camera/depth/image_rect/compressedDepth/parameter_updates
/camera/depth/image_rect/theora
/camera/depth/image_rect/theora/parameter_descriptions
/camera/depth/image_rect/theora/parameter_updates
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth/image_rect_raw/theora
/camera/depth/image_rect_raw/theora/parameter_descriptions
/camera/depth/image_rect_raw/theora/parameter_updates
/camera/depth/points
/camera/depth_rectify_depth/parameter_descriptions
/camera/depth_rectify_depth/parameter_updates
/camera/depth_registered/camera_info
/camera/depth_registered/image_raw
/camera/depth_registered/image_raw/compressed
/camera/depth_registered/image_raw/compressed/parameter_descriptions
/camera/depth_registered/image_raw/compressed/parameter_updates
/camera/depth_registered/image_raw/compressedDepth
/camera/depth_registered/image_raw/compressedDepth/parameter_descriptions
/camera/depth_registered/image_raw/compressedDepth/parameter_updates
/camera/depth_registered/image_raw/theora
/camera/depth_registered/image_raw/theora/parameter_descriptions
/camera/depth_registered/image_raw/theora/parameter_updates
/camera/depth_registered/points
/camera/depth_registered/sw_registered/camera_info
/camera/depth_registered/sw_registered/image_rect
/camera/depth_registered/sw_registered/image_rect/compressed
/camera/depth_registered/sw_registered/image_rect/compressed/parameter_descriptions
/camera/depth_registered/sw_registered/image_rect/compressed/parameter_updates
/camera/depth_registered/sw_registered/image_rect/compressedDepth
/camera/depth_registered/sw_registered/image_rect/compressedDepth/parameter_descriptions
/camera/depth_registered/sw_registered/image_rect/compressedDepth/parameter_updates
/camera/depth_registered/sw_registered/image_rect/theora
/camera/depth_registered/sw_registered/image_rect/theora/parameter_descriptions
/camera/depth_registered/sw_registered/image_rect/theora/parameter_updates
/camera/depth_registered/sw_registered/image_rect_raw
/camera/depth_registered/sw_registered/image_rect_raw/compressed
/camera/depth_registered/sw_registered/image_rect_raw/compressed/parameter_descriptions
/camera/depth_registered/sw_registered/image_rect_raw/compressed/parameter_updates
/camera/depth_registered/sw_registered/image_rect_raw/compressedDepth
/camera/depth_registered/sw_registered/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth_registered/sw_registered/image_rect_raw/compressedDepth/parameter_updates
/camera/depth_registered/sw_registered/image_rect_raw/theora
/camera/depth_registered/sw_registered/image_rect_raw/theora/parameter_descriptions
/camera/depth_registered/sw_registered/image_rect_raw/theora/parameter_updates
/camera/driver/parameter_descriptions
/camera/driver/parameter_updates
/camera/ir/camera_info
/camera/ir/image
/camera/ir/image/compressed
/camera/ir/image/compressed/parameter_descriptions
/camera/ir/image/compressed/parameter_updates
/camera/ir/image/compressedDepth
/camera/ir/image/compressedDepth/parameter_descriptions
/camera/ir/image/compressedDepth/parameter_updates
/camera/ir/image/theora
/camera/ir/image/theora/parameter_descriptions
/camera/ir/image/theora/parameter_updates
/rosout
/rosout_agg
/tf
/tf_static`

It seems like the /usb_cam/camera_info has all the transformations set to zero for some reason...
`header: 
  seq: 528
  stamp: 
    secs: 1486431518
    nsecs: 934969955
  frame_id: head_camera
height: 480
width: 640
distortion_model: ''
D: []
K: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
R: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
P: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
`
Thank you for your time! I'll look into this, if you have any advice please let me know.

matlabbe, 2017-02-07:
It is strange that ros_astra_camera package doesn't provide a RGB image topic with camera_info by default.  You may need to calibrate the RGB camera to fill at least K and P: http://wiki.ros.org/camera_calibration

robotic-enthusiast, 2017-02-07:
Dear matlabbe,

Thank you for your help!

I finished calibrating the RGB camera, and the topic is no longer zero. 

However, I am still getting the same warning:
`[ WARN] [1486439355.464980014]: odometry: Could not get transform from camera_link to head_camera (stamp=1486439355.235581) after 0.200000 seconds ("wait_for_transform_duration"=0.200000)! Error=". canTransform returned after 0.201314 timeout was 0.2."`

And no map is shown neither in rviz nor in rtabmapviz. Is this warning the reason why rtabmap is not working?

Cheers,

matlabbe, 2017-02-07:
Hi,

you may change the frame_id in image and camera_info topic to /camera_rgb_optical_frame as /head_camera link seems not published on TF. You can also do this to show the TF tree:
```
$ rosrun tf view_frames
$ evince frames.pdf
```

robotic-enthusiast, 2017-02-07:
Perfect! It works!
I will post the entire solution now and close the issue. Thank you very much for your help!

There were two problems when setting the Astra Pro, Ubuntu 16.04, ROS Kinetic with RTABmap:
1. There was no default RGB calibration file. Simply run the monocular calibration procedure (http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration ; requires a checkerboard) and obtain a .yaml file. Then edit this .yaml file to match with the camera_name(head_camera by default), and place it inside the $ROS_HOME/camera_info folder. 

2. Since Astra Pro did not seem to have direct access to the RGB camera on the Astra SDK, I used the usb_cam package to access the data: 
https://github.com/bosch-ros-pkg/usb_cam
However, usb_cam uses by default the name "head_camera" for the frame_id and camera_name.
Simply change the frame_id to "camera_rgb_optical_frame", and remake the package.

Just remember to use the right arguments for the topic names for rtabmap!

closed 2017-02-07 by robotic-enthusiast