introlab/rtabmap_ros #149: Issues when running on clean TX1: cannot find -lopencv_dep_cudart

neweirik, 2017-02-22:
Im running a clean flash of jetpack L4T on my Jetson TX1. I cloned rtabmap_ros into src and got << cannot find -lopencv_dep_cudart  >> from catkin_make. I remember having no problems running rtabmap on tx1 a while back, but now I got this error. Do you think its the Jetpack version or something from your end? Do you have a clear guide to install rtabmap on TX1 or TK1? As a lot of people are using it with this new hardware

neweirik, 2017-02-22:
additional information, I get this warning from cmake: 

`CMake Warning at rtabmap_ros/CMakeLists.txt:315 (add_executable):
  Cannot generate a safe runtime search path for target odom_msg_to_tf
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libfreenect.so.0.5] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /opt/ros/kinetic/lib
      /opt/ros/kinetic/share/rviz/cmake/../../../lib
    runtime library [libfreenect_sync.so.0.5] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /opt/ros/kinetic/lib
      /opt/ros/kinetic/share/rviz/cmake/../../../lib

  Some of these libraries may not be found correctly.
`

matlabbe, 2017-02-23:
I don't own TX1 or TK1 development kits, so I cannot really help on a guide to build on these platforms. For the opencv_dep_cudart error, it seems related to installed opencv: in this case opencv tells that there should be a opencv_dep_cudart installed somewhere, seems related to CUDA. Make sure you have only one libfreenect installed. 

Not sure if the binary packages are available on this platform, but can you install rtabmap/rtabmap_ros kinectic binary release (`sudo apt-get install ros-kinectic-rtabmap-ros`)?

neweirik, 2017-02-25:
I managed to get it working on TX1 running kinetic and ubuntu 16.04. Some problems along the way: 

-  If you install rtabmap with apt-get it installs perfectly, however some conflicts showed up later. Best thing is to just build from source. 

 - Building from source works well, however TX1 does not have enough swap memory to run  make -j4. You need to set aside 2Gb in a seperate swap file.  

 - Catkin will compile rtabmap_ros if you dont have the zed ros wrapper installed. If you need to have the zed_wrapper installed use sudo apt-get install ros-kinetic-rtabmap-ros. this didnt lead to any conflicts. 

I might make a complete .sh file that does everything on tx1 with the newest jetpack version. Would have saved me two days of work.