introlab/rtabmap_ros #151: rgbd_odometry won't run with F2M and g2o

capstayn, 2017-02-23:
On ubuntu 14.04 LTS Trusty,
ros indigo, 
rtabmap_ros 1.11.20, 
rtabmap 0.11.14

I run the following launch file: 

    <launch>
    <param name="use_sim_time" value="true" />

    
    <arg name="vo_frame_id" value="base_link"/>
    <arg name="camera_frame_id" value="/openni_rgb_optical_frame"/>

    <arg name="pi_2" value="1.57079633" />
    <arg name="pi" value="3.14159265" />

    
    <node pkg="tf" type="static_transform_publisher" name="camera_to_base_tf" 
        args="0 0 0 $(arg pi_2) $(arg pi) $(arg pi_2) $(arg vo_frame_id) $(arg camera_frame_id) 100" />

    
    <node pkg="rtabmap_ros" type="rgbd_odometry" name="rgbd_odometry" output="screen" required="true" ns="rtabmap" args="--logconsole --udebug" >

    <param name="frame_id" value="$(arg vo_frame_id)"/>
    <param name="OdomF2M/BundleAdjustment" value="1"/>
    <param name="Odom/Strategy" type="string" value="0"/>

    <remap from="rgb/image"       to="/rgb/image/down"/>
    <remap from="depth/image"     to="/depth/image/down"/>
    <remap from="rgb/camera_info" to="/depth/image/camera_info"/>
    </node>
    </launch>

And it crashes at start with the following error: 

> terminate called after throwing an instance of 'UException'
  what():  [FATAL] (2017-02-23 15:46:10.248) OdometryF2M.cpp:808::computeTransform() Condition (lastFrame_->id() > 0) not met! [Input data should have ID greater than 0 when odometry bundle adjustment is enabled!]

This happens with all the rosbags I have tried it with; if I set either the strategy to F2F or the bundle adjustment to disabled, it runs normally. I've been browsing the code and it seems that the variable `lastFrame_->id` is not updated when a new frame is received, but I'm surely missing something. 

Is it possible to run the odometry with strategy F2M with bundle adjustment optimization?

matlabbe, 2017-02-23:
Yes, it should work. The problem is that the first frame sent by the rosbag may have ID=0, but rtabmap generally starts frame sequence at 1. That assert should not be there, as it is referring code that doesn't exist anymore. I removed the assert in this commit https://github.com/introlab/rtabmap/commit/45ca3500c653436f3b559ba502c38439775719ed, so it can work with id >=0.

Thx for reporting!

capstayn, 2017-02-24:
Thanks for your prompt answer. I've run the new code and, after a few frames without being able to refine with bundle adjustment, it crashes with this message: 

> terminate called after throwing an instance of 'UException'
  what():  [FATAL] (2017-02-24 13:25:13.421) OdometryF2M.cpp:238::computeTransform() Condition (bundlePoses_.size()-1 == bundleLinks_.size()) not met! [poses=1 links=1]

If I comment this assert, it works, but the bundle adjustment keeps failing and issuing warnings about the transform not being refined. 

Let me know any test I could perform. Glad to help!

matlabbe, 2017-02-24:
Hi,

that assert is legitimate, I think your input frames have all same sequence ID (e.g., 0) as I can reproduce your error/warning by forcing all generated frames with ID=0. I updated the code to be independent of the ID set on the frames (see https://github.com/introlab/rtabmap/commit/b0170d1f6f49ead6b1c6423ae2928f0c64660552).

cheers

capstayn, 2017-02-27:
Yes, you were right, my input frames have all sequence equal to 0. Your last commit fixed the issue. 
However, shouldn't  `rtabmap_ros` set the correct ID to each new frame received from the rosbag? How should I set the ID's to the frames in the rosbag? In the `camera_info` topic?

In any case, thanks for the fixing. Feel free to close the thread as it's been solved.

matlabbe, 2017-02-27:
`rtabmap` node generates new IDs by default (`Mem/GenerateIds=true`), but not `rgbd_odometry`. The fix above makes `rgbd_odometry` generate its own IDs, so ignoring completly ROS message IDs. 

cheers,
Mathieu

closed 2017-02-27 by matlabbe