introlab/rtabmap_ros #154: compiler error

HoangGiang88, 2017-02-27:
Hi,
I have tried on desktop and it worked well,
but install on ubuntu(Tegra K1 kit)
-  git clone --branch 0.11.13 https://github.com/introlab/rtabmap.git rtabmap
-  cd rtabmap/build
-  cmake ..
-  make -j4
-  sudo make install
then got compiler error as follow:

[ 10%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/DBDriver.cpp.o
c++: internal compiler error: Killed (program cc1plus)

Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
make[2]: *** [tools/ExtractObject/CMakeFiles/extractObject.dir/main.cpp.o] Error 4
make[1]: *** [tools/ExtractObject/CMakeFiles/extractObject.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

[ 10%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/DBDriverSqlite3.cpp.o
c++: internal compiler error: Killed (program cc1plus)

Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/Rtabmap.cpp.o] Error 4
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
make: *** [all] Error 2
ubuntu@tegra-ubuntu:~/rtabmap/build$ vim ///usr/share/doc/gcc-4.8/README.Bugs

HoangGiang88, 2017-02-27:
use that branch is compiled 
git clone --branch kinetic-devel https://github.com/introlab/rtabmap.git rtabmap

matlabbe, 2017-02-27:
`The error c++: internal compiler error: Killed (program cc1plus)` seems like an out of memory compiler error. Try "`make -j1`" instead of "`make -j4`"  or increase your swap memory.

cheers,
Mathieu

HoangGiang88, 2017-02-28:
Also close all running task for save memory.
thanks y :)

closed 2017-02-28 by HoangGiang88