introlab/rtabmap_ros #155: compiler error 0.11.13 rtabmap_ros

HoangGiang88, 2017-02-28:
Hi,

The error got when linking exe file,
I have success compile one in tegraK1 but fail for another.

Scanning dependencies of target data_player
[100%] Building CXX object rtabmap_ros/CMakeFiles/data_player.dir/src/DbPlayerNode.cpp.o
Linking CXX executable /home/ubuntu/aos/system/PERSONAL_ROBOT/ros_ws/devel/lib/rtabmap_ros/data_player

/home/ubuntu/ros_ws/devel/lib/librtabmap_ros.so: undefined reference to 'rtabmap::Signature::setWords(std::multimap<int, cv::KeyPoint, std::less<int>, std::allocator<std::pair<int const, cv::KeyPoint> > > const&)'
collect2: error: ld returned 1 exit status

make[2]: *** [/home/ubuntu/ros_ws/devel/lib/rtabmap_ros/data_player] Error 1
make[1]: *** [rtabmap_ros/CMakeFiles/data_player.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j1 -l1" failed
ubuntu@tegra-ubuntu:~/ros_ws$

matlabbe, 2017-03-01:
That function is now [defined](https://github.com/introlab/rtabmap/blob/af1cdadabd97e179d30421666b2974f973b6e898/corelib/src/Signature.cpp#L231) in cpp instead of directly in the header. Make sure rtabmap_ros is linking on the latest version of rtabmap libraries.

cheers,
Mathieu

HoangGiang88, 2017-03-02:
I used wrong corelib version 0.11.14 with ros 0.11.13 
after compiled core-lib
replaced librtabmap_ros.so  then error gone.

Thanks y :)

closed 2017-03-02 by HoangGiang88