introlab/rtabmap_ros #157: Rtabmapviz and Visual Odometry process keeps dying on start-up

JoshuaSiy, 2017-03-16:
Hello, i hope i am not making a replicated issue or any disturbances. However i am having problems trying to start up rtabmap_ros . Rtabmapviz and visual odometry keeps dying immediately without even bringing me to the gui of rtabmap_ros. I'm not sure if any errors will be newly produced after this one is resolved. I tried my best to search the issues forums and i am not pretty sure what to do since many of the cases doesn't seem to correspond with what's happening in my case.

Level of intelligence: Newbie at Linux and ROS (including cmake, C++ and python)

things i installed : Rtabmap, ROS-kinetic,rtabmap_ros, iai_kinect2  ubuntu 16.04, opencv 2.4.11 , cuda 
Software: Everything in the optional parts and needed parts and dependencies inside the instructions of iai kienct2 and rtabmap/rtabmap_ros except for beignet from the instructions(incompatible according to clinfo) which i built beignet1.3 from source.

Extraordinary things i did aside form the instruction: edited CMakelists [cvsba, rtabmap, rtabmapros, and iai kinect2] on lines regarding find_package(OPENCV REQUIRED) To find_package(OPENCV 2.4.11 REQUIRED)

Hardware: Intel i7  and Nvidia GEFORCE GTX1050

Things i did without understanding. Recompiled everything by deleting build folders, and devel repeatedly and using cmake and catkin_make

Details that might be helpful:
standalone rtabmap works fine
kinect2_bridge.launch works with rosrun kinect2_viewer kinect2_viewer sd cloud 
./Protonect cl,gl,cuda, and cpu works fine 

**Errors produced:**
[ INFO] [1489643274.416961031]: Setting RTAB-Map parameter "Vis/MinInliers"="20"
[rtabmap/rtabmapviz-4] process has died [pid 30553, exit code -11, cmd /home/nerga/catkin_ws/devel/lib/rtabmap_ros/rtabmapviz -d /home/nerga/catkin_ws/src/rtabmap_ros/launch/config/rgbd_gui.ini rgb/image:=/kinect2/qhd/image_color_rect depth/image:=/kinect2/qhd/image_depth_rect rgb/camera_info:=/kinect2/qhd/camera_info __name:=rtabmapviz __log:=/home/nerga/.ros/log/50ee4348-0a0b-11e7-a66d-c8215827d99f/rtabmap-rtabmapviz-4.log].
log file: /home/nerga/.ros/log/50ee4348-0a0b-11e7-a66d-c8215827d99f/rtabmap-rtabmapviz-4*.log
[ INFO] [1489643274.588235580]: RTAB-Map detection rate = 1.000000 Hz
[ INFO] [1489643274.588321344]: rtabmap: Deleted database "/home/nerga/.ros/rtabmap.db" (--delete_db_on_start is set).
[ INFO] [1489643274.588336165]: rtabmap: Using database from "/home/nerga/.ros/rtabmap.db".
[rtabmap/visual_odometry-2] process has died [pid 30541, exit code -11, cmd /home/nerga/catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry rgb/image:=/kinect2/qhd/image_color_rect depth/image:=/kinect2/qhd/image_depth_rect rgb/camera_info:=/kinect2/qhd/camera_info __name:=visual_odometry __log:=/home/nerga/.ros/log/50ee4348-0a0b-11e7-a66d-c8215827d99f/rtabmap-visual_odometry-2.log].
log file: /home/nerga/.ros/log/50ee4348-0a0b-11e7-a66d-c8215827d99f/rtabmap-visual_odometry-2*.log

JoshuaSiy, 2017-03-17:
Okay i managed to fix rtabmap ROS, all i did was to make another workspace. and compiled rtabmap_ros along with vision opencv [DESPITE RTABMAP is linked to opencv2.4  rtabmapviz keeps on linking to ros-kinetic-opencv3.2, so you have to compile both rtabmap_ros with vision_opencv [i don't know why but this is how it worked on my side]  and build iai_kinect2 make another catkin workspace folder and compile it with given instruction to find libfreenect2. I don't know why did this happen, but some details and understanding would be nice. Feel free to close the issue, but i still want to know why. Thank you.

matlabbe, 2017-03-17:
rtabmapviz uses cv_bridge, which would be linked to OpenCV3 on kinetic. So if rtabmapviz is linked to rtabmap library (which is on OpenCV2) and cv_bridge (which is on OpenCV3), it is likely that the app crashes on start. When rebuilding vision_opencv, cv_bridge will be now linked to same OpenCV version than rtabmap library (OpenCV2). Hope this explanation satisfy you!

cheers,
Mathieu

closed 2017-03-17 by matlabbe