introlab/rtabmap_ros #159: Map not updating the moving object

Annaiyan, 2017-03-23:
Hi,
I'm using the stereo camera for building the rtabmap. But when i introduce moving object, it's not updated in the **/rtabmap/octomap_full** . I'm using the latest version of rtabmap_ros with ros-indigo -ubuntu14.04

How to update the moving object inside the map?
thanks,

matlabbe, 2017-03-23:
Hi,

I did some tests and it should clear the obstacle only if the sensor can see structure behind the obstacle. First example here is with a Kinect, note that the full wall behind can be seen by the Kinect:
[image]

With a stereo camera, the wall behind cannot be seen (textureless), so after the obstacle is gone, there is no ray tracing possible to clear the obstacle. We can see the legs are cleared (because of the obstacles behind on the floor), but not the upper body cannot be cleared:
[image]

cheers,
Mathieu

Annaiyan, 2017-03-27:
Thanks the details.
yes, it is working if the moving object is around the static features. if it is only the moving object, then it is considered to be static (what is observed).

Annaiyan, 2017-11-16:
In rtabmap ros , is there any way to update the moving object without having static structure behind? Because it is updating only if it is mixed with static objects around.

matlabbe, 2017-11-16:
With cameras, it is difficult to know if it is seeing an empty volume or that it cannot see an object (because it is reflective, black). It is then not safe to remove the moving object if we cannot see behind. What you may want to do is to filter the dynamic object off the depth image before feeding it to rtabmap, so the map is built only from static structures. As dynamic obstacles would not show in the map, another module will have to track them for obstacle avoidance (like the local costmap in move_base).