introlab/rtabmap_ros #160: map_assembler crash after update

masterhui, 2017-04-03:
After updating rtabmap/rtabmap_ros to the latest binary releases today (0.11.13-0xenial-20170315-031732-0700 and 0.11.13-0xenial-20170315-042452-0700), the map_assembler node crashes as soon as a subscriber requests topic /rtabmap/map_assembler/cloud_map.

The error message is:

```
terminate called after throwing an instance of 'UException'
  what():  [FATAL] (2017-04-04 01:13:15.488) util3d.cpp:980::cloudRGBFromSensorData() Condition (!sensorData.imageRaw().empty()) not met!
[rtabmap/map_assembler-27] process has died [pid 22661, exit code -6, cmd /opt/ros/kinetic/lib/rtabmap_ros/map_assembler __name:=map_assembler __log:=/home/ros/.ros/log/0342a84a-18bc-11e7-b78f-f3a2f552dd8
e/rtabmap-map_assembler-27.log].
log file: /home/ros/.ros/log/0342a84a-18bc-11e7-b78f-f3a2f552dd8e/rtabmap-map_assembler-27*.log
```
The log file contains:

```
os@autonomy:~$ cat /home/ros/.ros/log/0342a84a-18bc-11e7-b78f-f3a2f552dd8e/rtabmap-map_assembler-27*.log
[ INFO] [1491260406.115398960]: /rtabmap/map_assembler(maps): grid_cell_size             = 0.050000
[ INFO] [1491260406.115479506]: /rtabmap/map_assembler(maps): grid_incremental           = false
[ INFO] [1491260406.115499454]: /rtabmap/map_assembler(maps): grid_size                  = 0.000000
[ INFO] [1491260406.115515146]: /rtabmap/map_assembler(maps): grid_eroded                = false
[ INFO] [1491260406.115529152]: /rtabmap/map_assembler(maps): grid_footprint_radius      = 0.000000
[ INFO] [1491260406.115542998]: /rtabmap/map_assembler(maps): map_filter_radius          = 0.000000
[ INFO] [1491260406.115562734]: /rtabmap/map_assembler(maps): map_filter_angle           = 30.000000
[ INFO] [1491260406.115576297]: /rtabmap/map_assembler(maps): map_cleanup                = true
[ INFO] [1491260406.115589196]: /rtabmap/map_assembler(maps): map_negative_poses_ignored = false
[ INFO] [1491260406.115601665]: /rtabmap/map_assembler(maps): cloud_output_voxelized     = true
[ INFO] [1491260406.115615328]: /rtabmap/map_assembler(maps): cloud_subtract_filtering   = false
[ INFO] [1491260406.115628558]: /rtabmap/map_assembler(maps): cloud_subtract_filtering_min_neighbors = 2
[ INFO] [1491260406.125814893]: /rtabmap/map_assembler(maps): octomap_tree_depth         = 16
[ INFO] [1491260406.125853058]: /rtabmap/map_assembler(maps): octomap_occupancy_thr      = 0.500000
[ INFO] [1491260413.286485871]: map_assembler: Publishing data = 0.394065s
[ INFO] [1491260413.540310893]: map_assembler: Publishing data = 0.004196s
[ INFO] [1491260414.444269796]: map_assembler: Publishing data = 0.006281s
---
[ WARN] [1491261195.488030598]: Many occupancy grids should be loaded (~83), this may take a while to update the map(s)...
```
Any help is appreciated.

Regards

matlabbe, 2017-04-03:
The commit above should fix the problem. I also took the time to update the parameters interface to match the one used by `rtabmap` node:
```bash
$ rosrun rtabmap_ros map_assembler --params
Param: Grid/3D = "true"                                    [A 3D occupancy grid is required if you want an OctoMap (3D ray tracing). Set to false if you want only a 2D map, the cloud will be projected on xy plane. A 2D map can be still generated if checked, but it requires more memory and time to generate it. Ignored if laser scan is 2D and "Grid/FromDepth" is false.]
Param: Grid/CellSize = "0.05"                              [Resolution of the occupancy grid.]
Param: Grid/ClusterRadius = "0.1"                          [[Grid/NormalsSegmentation=true] Cluster maximum radius.]
Param: Grid/DepthDecimation = "4"                          [[Grid/DepthDecimation=true] Decimation of the depth image before creating cloud. Negative decimation is done from RGB size instead of depth size (if depth is smaller than RGB, it may be interpolated depending of the decimation value).]
Param: Grid/DepthMax = "4.0"                               [[Grid/FromDepth=true] Maximum cloud's depth from sensor. 0=inf.]
Param: Grid/DepthMin = "0.0"                               [[Grid/FromDepth=true] Minimum cloud's depth from sensor.]
Param: Grid/DepthRoiRatios = "0.0 0.0 0.0 0.0"             [[Grid/FromDepth=true] Region of interest ratios [left, right, top, bottom].]
Param: Grid/FlatObstacleDetected = "true"                  [[Grid/NormalsSegmentation=true] Flat obstacles detected.]
Param: Grid/FootprintHeight = "0.0"                        [Footprint height used to filter points over the footprint of the robot. Footprint length and width should be set.]
Param: Grid/FootprintLength = "0.0"                        [Footprint length used to filter points over the footprint of the robot.]
Param: Grid/FootprintWidth = "0.0"                         [Footprint width used to filter points over the footprint of the robot. Footprint length should be set.]
Param: Grid/FromDepth = "true"                             [Create occupancy grid from depth image(s), otherwise it is created from laser scan.]
Param: Grid/GroundIsObstacle = "false"                     [[Grid/3D=true] Ground segmentation (Grid/NormalsSegmentation) is ignored, all points are obstacles. Use this only if you want an OctoMap with ground identified as an obstacle (e.g., with an UAV).]
Param: Grid/MapFrameProjection = "false"                   [Projection in map frame. On a 3D terrain and a fixed local camera transform (the cloud is created relative to ground), you may want to disable this to do the projection in robot frame instead.]
Param: Grid/MaxGroundAngle = "45"                          [[Grid/NormalsSegmentation=true] Maximum angle (degrees) between point's normal to ground's normal to label it as ground. Points with higher angle difference are considered as obstacles.]
Param: Grid/MaxGroundHeight = "0.0"                        [Maximum ground height (0=disabled). Should be set if "Grid/NormalsSegmentation" is true.]
Param: Grid/MaxObstacleHeight = "0.0"                      [Maximum obstacles height (0=disabled).]
Param: Grid/MinClusterSize = "10"                          [[Grid/NormalsSegmentation=true] Minimum cluster size to project the points.]
Param: Grid/MinGroundHeight = "0.0"                        [Minimum ground height (0=disabled).]
Param: Grid/NoiseFilteringMinNeighbors = "5"               [Noise filtering minimum neighbors.]
Param: Grid/NoiseFilteringRadius = "0.0"                   [Noise filtering radius (0=disabled). Done after segmentation.]
Param: Grid/NormalK = "10"                                 [[Grid/NormalsSegmentation=true] K neighbors to compute normals.]
Param: Grid/NormalsSegmentation = "true"                   [Segment ground from obstacles using point normals, otherwise a fast passthrough is used.]
Param: Grid/ProjRayTracing = "false"                       [[Grid/3D=false] 2D ray tracing is done for each projected obstacle, filling unknown space between the sensor and obstacles.]
Param: Grid/Scan2dMaxFilledRange = "4.0"                   [Unknown space filled maximum range. If 0, the laser scan maximum range is used.]
Param: Grid/Scan2dUnknownSpaceFilled = "false"             [Unknown space filled. Only used with 2D laser scans.]
Param: Grid/ScanDecimation = "1"                           [[Grid/FromDepth=false] Decimation of the laser scan before creating cloud.]
Param: StereoBM/BlockSize = "15"                           [See cv::StereoBM]
Param: StereoBM/MinDisparity = "0"                         [See cv::StereoBM]
Param: StereoBM/NumDisparities = "128"                     [See cv::StereoBM]
Param: StereoBM/PreFilterCap = "31"                        [See cv::StereoBM]
Param: StereoBM/PreFilterSize = "9"                        [See cv::StereoBM]
Param: StereoBM/SpeckleRange = "4"                         [See cv::StereoBM]
Param: StereoBM/SpeckleWindowSize = "100"                  [See cv::StereoBM]
Param: StereoBM/TextureThreshold = "10"                    [See cv::StereoBM]
Param: StereoBM/UniquenessRatio = "15"                     [See cv::StereoBM]
[ WARN] [1491238592.186505815]: Node will now exit after showing default parameters because argument "--params" is detected!
```

Thx for reporting the bug,
Mathieu

masterhui, 2017-04-04:
Hello Mathieu,
The commit 388e8f5 fixes the problem of map_assembler crashing. However, now that I can subscribe to /rtabmap/map_assembler/cloud_map, I noticed another problem: The PointCloud2 data of the cloud_map is not a cloud_map but a scan_map. Further testing reveals that the rtabmap clouds also don't look the way they used to:

/rtabmap/map_assembler/cloud_map
/rtabmap/map_assembler/cloud_ground
/rtabmap/map_assembler/scan_map

and

/rtabmap/cloud_map
/rtabmap/cloud_ground
/rtabmap/scan_map

... all show the same scan_map type data (see attached images). Did I miss something? 

Regards

[image]
[image]

matlabbe, 2017-04-04:
Hi,

This is the correct behavior. `/scan_map` is deprecated, it is a copy of `/cloud_map` until it is removed. For memory/time efficiency, `/cloud_map` is now created from empty and obstacle cells saved in the nodes of the graph (instead of being regenerated from raw images). The empty/obstacle cells are created following the `Grid/*` parameters. If `Grid/FromDepth` is true, you will get the original `/cloud_map` created from the camera. If `Grid/FromDepth` is false, you will get `/cloud_map` the cloud of the scans assembled like in your pictures above. By default, `Grid/FromDepth` is false if rtabmap node is subscribed to a laser scan topic. Try this to see the difference:
```xml
<node pkg="rtabmap_ros" type="rtabmap" name="rtabmap">
   ...
   <param name="Grid/FromDepth" type="string" value="false"/>
</node>
```

Here are the differences between the topics:

- `/cloud_map` : empty + obstacle cells
- `/cloud_ground` : empty cells
- `/cloud_obstacles`: obstacle cells
- `/grid_map` : 2D occupancy grid (match `/cloud_map` projected in 2D)
- (deprecated) `/proj_map` : same as `/grid_map` 
- (deprecated) `/scan_map` : same as `/cloud_map`

cheers,
Mathieu

masterhui, 2017-04-05:
Thanks Mat, works as described. Cheers

closed 2017-04-05 by masterhui