introlab/rtabmap_ros #161: Cannot set map_assembler params

masterhui, 2017-04-10:
Hello Mat,
Following the recent commit b0ee4b63983eb32b61da9a7b1516a2e4f349e7be, I have noticed that the params for map_assembler cannot be set. I will be using parameter CellSize as an example, which exists for both nodes rtabmap and map_assembler

If the CellSize parameter is not specified, the default value is used:
```
ros@autonomy:~$ rosparam list | grep Cell
/Grid/CellSize
ros@autonomy:~$ rosparam get /Grid/CellSize
'0.05'
```
Setting CellSize for node rtabmap:
```
<node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" >
  <param name="Grid/CellSize" type="double" value="0.01"/>
</node>
```
results in:
```
ros@autonomy:~$ rosparam list | grep Cell
/Grid/CellSize
/rtabmap/Grid/CellSize
ros@autonomy:~$ rosparam get /Grid/CellSize
'0.01'
ros@autonomy:~$ rosparam get /rtabmap/Grid/CellSize
0.01
```
However, this does not affect the CellSize of node map_assembler. If CellSize is set separately for node map_assembler:
```
<node pkg="rtabmap_ros" type="map_assembler" name="map_assembler"  output="log" >
  <param name="Grid/CellSize" type="double" value="0.02"/>    
</node>
```
The result is:
```
ros@autonomy:~$ rosparam list | grep Cell
/Grid/CellSize
/map_assembler/Grid/CellSize
/rtabmap/Grid/CellSize
ros@autonomy:~$ rosparam get /Grid/CellSize
'0.01'
ros@autonomy:~$ rosparam get /rtabmap/Grid/CellSize
0.01
ros@autonomy:~$ rosparam get /map_assembler/Grid/CellSize
0.02
```
But map_assembler does not use the new value for CellSize. Instead, the default value of 0.05 seems to be used (/map_assembler/cloud_map is still as low-res as before).

Regards

matlabbe, 2017-04-10:
Hi,

Thx to point out, there are some repetitive parameters about the grid size that I will try to fix today. Until it is fixed, you should use "grid_cell_size" ros parameter for map_assembler. For rtabmap node, you would need to set both "grid_cell_size" and "Grid/CellSize". 

cheers,
Mathieu

matlabbe, 2017-04-10:
I removed `grid_cell_size`, only `Grid/CellSize` is now used (from commit above). Normally with 
```
<node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" >
  <param name="Grid/CellSize" type="double" value="0.01"/>
</node>
<node pkg="rtabmap_ros" type="map_assembler" name="map_assembler"  >
  <param name="Grid/CellSize" type="double" value="0.02"/>    
</node>
```
You should have a `/rtabmap/cloud_map` at 1 cm voxel and `/map_assembler/cloud_map` at 2 cm voxel. Be aware that if `Grid/CellSize` of rtabmap is higher (e.g., 5 cm) than in `map_assembler` (e.g., 1 cm), the output of `map_assembler` will be at the resolution of rtabmap (e.g. 5 cm). In contrast if `Grid/CellSize` of rtabmap is lower (e.g. 1 cm) than in map_assembler (e.g. 5cm), the output of `map_assembler` will be at resolution set in map_assembler (e.g. 5 cm).

cheers,
Mathieu

masterhui, 2017-04-12:
That's it, thanks.

closed 2017-04-12 by masterhui