introlab/rtabmap_ros #166: tf problem on ZED SDK 2.0?

ehong-tl, 2017-05-04:
Hi,

I've updated ZED SDK to 2.0 and tried running rtabmap_ros as per instruction [http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping](url)

But I keep getting the error,

[ WARN] [1493868976.086698133]: Could not get transform from camera_link to /zed_depth_frame after 0.200000 seconds (for stamp=1493868975.823453)! Error=". canTransform returned after 0.20122 timeout was 0.2.".
[ERROR] [1493868976.086743701]: TF of received depth image 0 at time 1493868975.823453s is not set!
[ERROR] [1493868976.086773001]: Could not convert rgb/depth msgs! Aborting rtabmapviz update...

It worked before when using SDK 1.2, but after updated to 2.0, it is unusable.

Need help to fix this.

Thanks.

matlabbe, 2017-05-05:
Hi,

I reverified with latest `zed_ros_wrapper` (latest [commit](https://github.com/stereolabs/zed-ros-wrapper/commit/15e1b1a8fe27926b7030468c9e75bce97c27af1d)) and it should be okay (follow exactly instructions for ZED). The only thing I noticed is that odometry with ZED is less reliable on ROS than on rtabmap standalone: you have to set confidence to 100 (instead of 80 when launching zed camera) to get same results than with standalone:
```bash
$ rosrun zed_wrapper zed_wrapper_node _confidence:=100
```
I updated the tutorial.

cheers,
Mathieu

ehong-tl, 2017-05-08:
Hi Mathieu,

Thanks for the tips.
Eventually I have to build the latest rtabmap_ros from source in order to make it working. The one in ROS repositories gave the same error even after setting confidence to 100.

closed 2017-05-08 by ehong-tl

txs, 2017-05-18:
Do you mean that I cannot apt-get install using the kinectic package? I have to build from source my own in order to get it working? I have this issue too!

ehong-tl, 2017-05-19:
I believe it's because the apt-get kinetic rtabmap-ros package is an older version of rtabmap (0.11.3 if I were right).